Koenkk/zigbee2mqtt #11110: Raspbee 2 (Deconz) crashes Zigbee2MQTT

DennisJohnsen, 2022-01-30:
### What happened?

I was trying to setup some colors via, the frontend on a group of bulbs, but suddenly the device (all devices) stopped responding for a minute or two. I suspect that something was overwhelmed with zigbee or mqtt requests.

### What did you expect to happen?

That using the frontend wouldn't crash Zigbee2MQTT.

### How to reproduce it (minimal and precise)

Go to a group.
Change color by moving the cursor around in the color picker.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26690700

### Adapter

RaspBee2

### Debug log

```
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor', payload '{"brightness":254,"color":{"hue":239,"saturation":87,"x":0.1875,"y":0.125},"color_mode":"xy","color_temp":271,"state":"ON"}'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3', payload '{"brightness":254,"color":{"hue":239,"saturation":87,"x":0.1875,"y":0.125},"color_mode":"xy","color_temp":271,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":"idle"},"update_available":false}'
error 2022-01-30 02:07:42: Publish 'set' 'color' to 'GU10 Color @kontor @1' failed: 'Error: Command 0x0c4314fffe90cbcf/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":17873,"colory":10923}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
debug 2022-01-30 02:07:42: Error: Command 0x0c4314fffe90cbcf/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":17873,"colory":10923}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:641:23)
    at Endpoint.sendRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:291:20)
    at Endpoint.command (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:722:28)
    at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1036:13)
    at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1061:32)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:246:36)
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"Publish 'set' 'color' to 'GU10 Color @kontor @1' failed: 'Error: Command 0x0c4314fffe90cbcf/1 lightingColorCtrl.moveToColor({\"transtime\":0,\"colorx\":17873,\"colory\":10923}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"GU10 Color @kontor @1"},"type":"zigbee_publish_error"}'
error 2022-01-30 02:07:42: Publish 'set' 'color' to 'GU10 Color @kontor @1' failed: 'Error: Command 0x0c4314fffe90cbcf/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":17672,"colory":11045}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
debug 2022-01-30 02:07:42: Error: Command 0x0c4314fffe90cbcf/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":17672,"colory":11045}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:641:23)
    at Endpoint.sendRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:291:20)
    at Endpoint.command (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:722:28)
    at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1036:13)
    at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1061:32)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:246:36)
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/bridge/log', payload '{"message":"Publish 'set' 'color' to 'GU10 Color @kontor @1' failed: 'Error: Command 0x0c4314fffe90cbcf/1 lightingColorCtrl.moveToColor({\"transtime\":0,\"colorx\":17672,\"colory\":11045}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"GU10 Color @kontor @1"},"type":"zigbee_publish_error"}'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/state', payload 'ON'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/color_mode', payload 'xy'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/brightness', payload '254'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/brightness', payload '254'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/color_temp', payload '271'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/state', payload 'ON'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/color_mode', payload 'xy'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/color_temp', payload '271'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/color-x', payload '0.1875'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/color-x', payload '0.1875'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/color-y', payload '0.125'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/color-y', payload '0.125'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/color-hue', payload '239'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/color-hue', payload '239'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/Kontor/color-saturation', payload '87'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/color-saturation', payload '87'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/update-state', payload 'idle'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/update_available', payload 'false'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/color_temp_startup', payload '65535'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/power_on_behavior', payload 'previous'
info  2022-01-30 02:07:42: MQTT publish: topic 'z2m/GU10 Color @kontor @3/linkquality', payload '255'
```
Last line of log file

[log-after.txt](https://github.com/Koenkk/zigbee2mqtt/files/7965119/log-after.txt)
[log-before.txt](https://github.com/Koenkk/zigbee2mqtt/files/7965120/log-before.txt)

Koenkk, 2022-01-30:
Looks like some congestion indeed. This can also happen due to interference, make sure the conbee is connected via a usb extension cable and use a zigbee channel that doesn't conflict with wifi: https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html#usb-based-adapter

DennisJohnsen, 2022-02-02:
I've located the issue for these vast amount of MQTT topics, and it turns out it's Homebridge Zigbee2MQTT plugin. I have +80 devices in Zigbee2Mqtt, which results in +200 accessories in Homebridge. I can replicate the crash by restarting Homebridge (and thus the zigbee2mqtt plugin), and sure enough after a minute (the keep alive function) it will start to fail with the above error. Uninstalling or disabling the homebridge plugin and my zigbee2mqtt network is stable.

Is there any way to limit the amount of mqtt topics that Zigbee2Mqtt will handle pr. second? It seems like it could be a nice trade off to have for the few times a day that Zigbee2Mqtt Homebridge plugin sends out a get topic for all the devices.

DennisJohnsen, 2022-02-02:
> @ChrisHae it seems the Conbee is crashing when it sends a lot of commands, could you add some rate limiting to the adapter code?
> 
> EDIT: Just saw [#10542 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/10542#issuecomment-1027945775) , can you try with an adapter delay of 200? Add to your `configuration.yaml`.
> 
> ```yaml
> advanced:
>   adapter_delay: 200
> ```

Will try later tonight and see how it performs! Nice find, fingers crossed!

DennisJohnsen, 2022-02-03:
Tried at 100 and it crashed upon starting homebridge.
Tried at 250 and also crashed.

Took some logs directly from the terminal this time, and this part looks odd:
```
Feb 02 23:51:19 homebridge npm[4198]: UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "undefined".
Feb 02 23:51:20 homebridge systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Feb 02 23:51:20 homebridge systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

The entire thing with the latest device before crash. Also note that just this single accessory is about 15 mqtt topics alone. Times that with at least 80, and there is the problem.  I don't know exactly how adapter delay works, but it seems like it lets as many requests through that it can in intervals of the given MS. Maybe you can explain how exactly it works, the feature looks fairly new.
```
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:debug 2022-02-02 23:51:17: Received Zigbee message from 'GU10 Color @victor @3', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":1,"colorTemperature":370,"currentSaturation":201,"currentX":30015,"currentY":26870,"enhancedCurrentHue":0}' from endpoint 1 with groupID null
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3', payload '{"brightness":254,"color":{"hue":32,"saturation":81,"x":0.458,"y":0.41},"color_mode":"xy","color_temp":368,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"state":"idle"},"update_available":false}'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/color_mode', payload 'xy'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/brightness', payload '254'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/state', payload 'OFF'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/color_temp', payload '368'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/color-x', payload '0.458'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/color-y', payload '0.41'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/color-hue', payload '32'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/color-saturation', payload '81'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/color_temp_startup', payload '65535'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/power_on_behavior', payload 'previous'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/update-state', payload 'idle'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/update_available', payload 'false'
Feb 02 23:51:17 homebridge npm[4198]: Zigbee2MQTT:info  2022-02-02 23:51:17: MQTT publish: topic 'z2m/GU10 Color @victor @3/linkquality', payload '255'
Feb 02 23:51:19 homebridge npm[4198]: UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "undefined".
Feb 02 23:51:20 homebridge systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Feb 02 23:51:20 homebridge systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Feb 02 23:51:20 homebridge systemd[1]: zigbee2mqtt.service: Consumed 14.823s CPU time.
Feb 02 23:51:20 homebridge systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 7.
Feb 02 23:51:20 homebridge systemd[1]: Stopped zigbee2mqtt.
```

Koenkk, 2022-02-03:
What if you try with a value of 1000?

DennisJohnsen, 2022-02-03:
I have it on 350 atm, at it seems to have og running more stable overall. It still crashed on initial load from Zigbee2MQTT Homebridge plug-in, but after that i havnt seen a crash since. Which im pretty sure i would have Seen by now before.

i Will try and see if i Can lower interference even more by looking into placement of my adapter, and also exclude some devices I don’t need in HomeKit.

How does the adapter delay feature even work? By looking at log it appears not to put a delay between all request, but more like it has a small window which allows requests then a small break inbtween of that number in ms.. but I could be wrong?

Koenkk, 2022-02-03:
> How does the adapter delay feature even work?

@chrishae could you explain this?

ChrisHae, 2022-02-04:
The idea is when sending many DATA_REQUESTS they sould be sent one after another and only when the sending of the previous one got confirmed. The confirm usually only takes very little time (less than 10ns). The  default value of 0 means that the sendig queue does not wait for a confirm and send out all requests very fast. All other values wait for that confirm but there is a timout if no confirm is received. adapter-delay of 500 means after 500ms and no confirm the next request is sent anyways.
Also the processing time of the internal queues is slightly higher so there is also a little delay between all commands (but not as high as the adapter_delay value. The value is capped at 60ms).

ChrisHae, 2022-02-04:
I havent tested the RaspBee2 with mqtt and so many devices. I should try to get a test environment. Perhaps the performance could be optimised.

DennisJohnsen, 2022-02-04:
Sounds like it's not exactly what could save RaspBee2 from crashing when sending +80 device mqtt topics at once (it's like +10 topics pr. device it looks like.) It does however seems to  have made Zigbee2MQTT more stable with the Homebridge plugin running as it haven't crashed once yet in over 36 hours (it would have at least twice before)

> (The value is capped at 60ms).

What defines the value? Is it always 60ms?

> could you provide an extensive log with enabled zigbee-herdsman debug logging?

I would love to, but I have no idea how to start Zigbee2MQTT with zigbee-herdsman debug logging. How would the command look like? And where would the logs be located then? *(sorry, i'm a CSS wizard, not normally working in a terminal)*

ChrisHae, 2022-02-07:
Look at this link it should be described there how to activate debug logging:
https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

two things:
1. You should update the RaspBee2 FW to the newest version 26720700
You will need the GCFFlasher tool which is shipped with the deCONZ software (https://phoscon.de/en/conbee/install#raspbian)
or can be compiled and installed as stand alone (https://github.com/dresden-elektronik/gcfflasher)
The FW update process is describted here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

2. You could install newest zigbee2mqtt dev branch which should be available soon. 
There I have activated APS acknowledgements for adapter_delay values >= 200. That should improve stability of larger networks but I don't know if it fix the crashes that you experience. Would be nice to know how this affects your network.

antoinelibert, 2022-02-07:
Hello,

I'm might experiencing an almost similar issue with my setup of 28 devices.

I was testing some new automations in HA and one was "shutdown every light bulb". I have noticed that if I called that particular automation, it would result of _some_ commands send to bulb to fail (`... failed (no response received)`).

ChrisHae, 2022-02-08:
This is the IKEA remote? This is another story. I'm not sure if it is possible to unbind the default group from it:
https://github.com/Koenkk/zigbee2mqtt/issues/3424

antoinelibert, 2022-02-08:
> This is the IKEA remote? This is another story. I'm not sure if it is possible to unbind the default group from it:
> #3424

Actually I have 3 of these IKEA E1743 remotes and I could unbind them from the default group before (green ok alert in the front end). I’m still on the old firmware allowing to bind to a group though.

DennisJohnsen, 2022-02-10:
Here is the log I could copy from the terminal before Zigbee2MQTTcrashed. No 
https://pastebin.com/AYHegSz1

I had adapter delay at 200, on Zigbee2MQTT version 1.23.0. I restarted Zigbee2MQTT Homebridge plugin a few times before it crashed, so adapter delay at 200 did do SOMETHING. I had Zigbee2MQTT running for +5 days without any crashes at all which was new to me, and that was at adapter delay 350.

Doesn't look like the new dev branch you're talking about is out yet. Let me know when I can test that.

closed 2022-03-22 by github-actions[bot]

Mistic92, 2022-06-10:
Is there any update for it? I think I might have similar issue. I had to reboot my home assistant and now when I connect new device to network zigbee crashes. 
I reflashed conbee II and it did not helped

```
Zigbee2MQTT:info  2022-06-10 23:52:59: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-06-10 23:53:20: Configuring 'aqara_sypialnia'
Zigbee2MQTT:info  2022-06-10 23:53:20: Successfully configured 'aqara_sypialnia'
UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "TIMEOUT".
[23:53:29] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.2 start
> node index.js
Zigbee2MQTT:info  2022-06-10 23:53:38: Logging to console and directory: '/config/zigbee2mqtt/log/2022-06-10.23-53-38' filename: log.txt
Zigbee2MQTT:info  2022-06-10 23:53:38: Starting Zigbee2MQTT version 1.25.2 (commit #unknown)
Zigbee2MQTT:info  2022-06-10 23:53:38: Starting zigbee-herdsman (0.14.34)
CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2022-06-10 23:53:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-06-10 23:53:48: Failed to start zigbee
Zigbee2MQTT:error 2022-06-10 23:53:48: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-06-10 23:53:48: Exiting...
Zigbee2MQTT:error 2022-06-10 23:53:48: undefined
```