Koenkk/zigbee2mqtt #11113: Alarm Keypad not arming

bravesfan171, 2022-01-30:
### What happened?

I have the Universal Electronics Inc XHK1-UE Alarm Keypad. Device is successfully paired and interviewed, however all of the action exposes remain "Null". When I send the arm code to zigbee2mqtt/Alarm Keypad/set...
 
```
{
    "arm_mode": {
        "mode": "arm_all_zones"
    }
}
```

nothing happens on the keypad either. This is true for all arm modes. I am on the Dev Branch and before I switched to this one, commands from the keypad would also not work, however that was resolved after switching.

Here is the log when the correct payload is sent:
```
Debug Received MQTT message on 'zigbee2mqtt/Alarm Keypad/set' with data '{ "arm_mode": { "mode": "arm_all_zones" } }'
Debug Publishing 'set' 'arm_mode' to 'Alarm Keypad'
```

### What did you expect to happen?

I expected that my Alarm Keypad would arm to respective modes

### How to reproduce it (minimal and precise)

Send the MQTT payload to arm the keypad

### Zigbee2MQTT version

1.22.2-dev commit: 93921c6

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0

Koenkk, 2022-01-30:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/10504

Fixed in the dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and in the next release on 1 February.

closed 2022-01-30 by Koenkk

bravesfan171, 2022-01-30:
Got it. Looking forward to that. Thanks!

bravesfan171, 2022-02-02:
@Koenkk The issue from #10504 is now resolved, but it looks like mine still persists. The keypad is able to send arm and disarm messages properly to the server, but the MQTT message in my issue above still does not arm the physical keypad. The Debug log still shows the same as above when I send the payload from the documentation [here](https://www.zigbee2mqtt.io/devices/XHK1-UE.html) and the keypad does not beep and light up the currently armed mode as it should.

Let me know if there's any further info I can provide. Thanks!

Koenkk, 2022-02-02:
@Rockman18  could you maybe help here, what is the correct way to arm the keypad?

bravesfan171, 2022-02-05:
@Koenkk @Rockman18 any help would be greatly appreciated. I've been following the documentation I linked to but it is not arming the keypad at all. I had it working on ZHA but this has been the only thing I haven't gotten to work properly after switching.

Rockman18, 2022-02-05:
Linking is not enough for the keypad. You have to create scenarios to respect the workflow.

Read the documentation correctly. The keypad requires an acknoledgement after an input. Home Assistant has to check the input code and accept or reject the command. In my case, i use node-red for that.

[image]

1.  I read the mqtt keypad topic.
```json
{
    "action": "arm_all_zones", // This is the example
    "action_code": "123", // The code being entered
    "action_zone": 0, // The zone being armed (always 0)
    "action_transaction": 99 // The transaction number
}
```

2. My scenario check the code and the current state of the alarm and send back the acknoledgement to the keypad mqtt topic.
```json
{
    "arm_mode": {
        "transaction": 99, // Transaction number (this must be the same as the keypad request `action_transaction`)
        "mode": "arm_all_zones" // Mode (this must be the same as the keypad request `action`)
    }
}
```
Valid mode values are **disarm**, **arm_day_zones**, **arm_night_zones**, **arm_all_zones**, **invalid_code**, **not_ready**, **already_disarmed**

Be careful, this is not enough to change the state of the keypad. At this moment the keypad only knows that home assistant has accepted or rejected the command and the keypad is still waiting for a message from the alarm system, telling him that the alarm is really activated.

3. Then I modify the state of the Alarm (cf. "Manual" hassio integration) in home assistant.

4. When "Manual" has changed the alarm system state, I send back a new command to the keypad to tell him to definitively change the keypad state.

[image]

```json
{
    "arm_mode": {
        "mode": "arm_all_zones"
    }
}
```
Valid mode values as per ZCL specifications are **disarm**, **arm_day_zones**, **arm_night_zones**, **arm_all_zones**, **exit_delay**, **entry_delay**, **not_ready**, **in_alarm**, **arming_stay**, **arming_night**, **arming_away**

The automation server must follow the notification with an actual change to the correct arm mode. For the example above, the server should respond with exit_delay, count the elapsed time (e.g 30 secs), then change mode again to arm_all_zones

bravesfan171, 2022-02-06:
@Rockman18 I appreciate the reply. I have read all of that documentation thoroughly and I have your steps 1-3 working perfectly. Codes from the keypad are able to arm and disarm the Manual Alarm in Home Assistant.

It is your step 4 that is not working properly for me. The simple arm mode and mode MQTT message from HA is not arming my keypad. Do you have the same model as me? I can see in my logs Zigbee2MQTT is receiving the message and passing it to the keypad, but somewhere in the process between it is not working. The keypad does not chime or illuminate the currently armed mode at all.

Any ideas?

Rockman18, 2022-02-08:
I use the exact same keypad (XHK1-UE). Try to update Zigbee2MQTT. Or share your logs.

bravesfan171, 2022-02-10:
@Rockman18 @Koenkk  I’m on the latest dev build of Zigbee2MQTT. Not sure why mine won’t work properly. All the logs that show up are the ones I’ve shared already. Anything else I can find to share that might help diagnose?

golijdam, 2022-02-19:
(not trying to hijack this topic, just a quick question)
> Valid mode values as per ZCL specifications are **disarm**, **arm_day_zones**, **arm_night_zones**, **arm_all_zones**, **exit_delay**, **entry_delay**, **not_ready**, **in_alarm**, **arming_stay**, **arming_night**, **arming_away**

@Rockman18 Does the mode **exit_delay** and **entry_delay** work for you on the XHK1? I got all other modes working with my XHK1 keypad but those modes seem to not changing anything on the keypad, no beeping, would like to have exit/entry beeps.

thecobra666, 2022-04-25:
> (not trying to hijack this topic, just a quick question)
> 
> > Valid mode values as per ZCL specifications are **disarm**, **arm_day_zones**, **arm_night_zones**, **arm_all_zones**, **exit_delay**, **entry_delay**, **not_ready**, **in_alarm**, **arming_stay**, **arming_night**, **arming_away**
> 
> @Rockman18 Does the mode **exit_delay** and **entry_delay** work for you on the XHK1 (TC or EU)? I got all other modes working with my XHK1-TC keypad but those modes do not change anything (noticeable) on the keypad while I was expecting a beeping sound, would like to have exit/entry beeps.

It works on the EU model but not on my TC. I can't figure out why (yet). I use the same automation as for the EU but, it's not responding on it. Instead, it's putting the device into arm day zones... it also stays there, really weird. 

`Debug Received MQTT message on 'zigbee2mqtt/keypadboven/set' with data '{ "arm_mode": { "mode": "entry_delay" } }'
Debug Publishing 'set' 'arm_mode' to 'keypadboven'
Debug Received Zigbee message from 'keypadboven', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/keypadboven', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"last_seen":"2022-04-25T08:37:55.243Z","linkquality":255,"occupancy":true,"presence":null,"tamper":true,"temperature":21,"voltage":6900}'
Debug Received Zigbee message from 'keypadboven', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/keypadboven', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"last_seen":"2022-04-25T08:38:00.916Z","linkquality":255,"occupancy":true,"presence":null,"tamper":true,"temperature":21,"voltage":6900}'`