Koenkk/zigbee2mqtt #11115: Erroneous reporting of 0 kWh energy meter on TuYa TS0121_plug

RubenKelevra, 2022-01-30:
### What happened?

That's quite a minor issue, but it throws off the whole whole energy consumption calculation from HA:

I got four TuYa TS0121_plug and at least 3 of them show (what looks like) random drops to 0 kWh on the energy meter. The graph is hiding a bit how low the value drops (I think).

[image]
[image]
[image]

Not sure if a configuration setting is responsible for that:

```
"availability_timeout": 3600,
"cache_state": true,
"cache_state_persistent": false,
"cache_state_send_on_startup": false,
"homeassistant_legacy_entity_attributes": true,
"homeassistant_legacy_triggers": true,
"last_seen": "disable",
"legacy_api": false,
```

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20201026

### Adapter

Slaesh's CC2652RB

RubenKelevra, 2022-01-30:
Here's such an event more detailed. Looks not like a drop to 0 but some old value to be pushed to HA before the device has send its own new value.

Could there be an issue with turned off persistence, that still the some values are stored and then restored? 🤔

[image]

RubenKelevra, 2022-01-30:
I think if that's the case it would also explain this behaviour:

[image]

The switch turns "on" for HA from unavailable and 20 seconds later it turns "off". After a restart of Z2M.

While the switch remained in the off position all the time.

That's kind of messing with my automations, since they react to on/off changes of the switch which then turns the smart light bulb on/off (which has power permanently).

This switch is a ZB Mini.

RubenKelevra, 2022-02-06:
I can confirm this bug for [1.23.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.23.0) as well:

[image]

closed 2022-03-16 by github-actions[bot]

maxbec, 2023-12-17:
Hey guys. Any fix on this?