Koenkk/zigbee2mqtt #11117: Trv02 température local

mcike59295, 2022-01-30:
### What happened?

Bonjour
local temperature not updating
TZE200_hue3yfsn
TS0601

### What did you expect to happen?

temperature update

### How to reproduce it (minimal and precise)

the temperature updates when I change the calibration

### Zigbee2MQTT version

1.22.2-dev

### Adapter firmware version

20190608

### Adapter

CC2531

Koenkk, 2022-01-30:
@vladi1234 could you check this?

vladi1234, 2022-01-30:
@Koenkk Ok, I'll do it.
thank you

vladi1234, 2022-01-30:
Here you can read when and how often you can get local_temperature.

https://github.com/Koenkk/zigbee-herdsman-converters/issues/3142#issuecomment-976885156
https://github.com/Koenkk/zigbee2mqtt/issues/8522#issuecomment-990070405
https://github.com/Koenkk/zigbee2mqtt/issues/8522#issuecomment-997474913

mcike59295, 2022-01-31:
thank you I'm a newbie! how to do on home assistant cordially

vladi1234, 2022-01-31:
@pvyleta explained everything very well here. 

https://github.com/Koenkk/zigbee2mqtt/issues/8522#issuecomment-997474913

Personally I don't use HA, best to ask @pvyleta.

mcike59295, 2022-01-31:
thank you very much problem solved 🤗

closed 2022-02-06 by mcike59295