Koenkk/zigbee2mqtt #11119: [New device support]: Smoke sensor TS0601 _TZE200_yh7aoahi

jocamane, 2022-01-30:
### Link

https://pt.aliexpress.com/item/1005002284013316.html?gatewayAdapt=glo2bra&spm=a2g0s.12269583.0.0.14d66f73TAcBoa

### Database entry

{"id":36,"type":"EndDevice","ieeeAddr":"0x84fd27fffeca5ada","nwkAddr":32700,"manufId":4098,"manufName":"_TZE200_yh7aoahi","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_yh7aoahi","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1643548709601,"useImplicitCheckin":true}

### Comments

Warning Device '0x84fd27fffeca5ada' with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yh7aoahi'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

jocamane, 2022-02-03:
hi, would it be possible for someone to help me.
thanks

jocamane, 2022-02-06:
Hum....

Itried to do an External by myself, but something is not right.

After adding the device to the z2m it gives the following errors in the log: 

Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Error Failed to configure '0x84fd27fffeca5ada', attempt 2 (Error: Bind 0x84fd27fffeca5ada/1 genBasic from '0x00212effff07afaa/1' failed (waiting for response TIMEOUT) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19) at runMicrotasks (<anonymous>) at runNextTicks (internal/process/task_queues.js:60:5) at processTimers (internal/timers.js:497:9) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:522:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) at Object.configure (/app/dist/util/externally-loaded.js:25:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))

MY External:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =[
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t5p1vj8r'}, 
            {modelID: 'TS0601', manufacturerName: '_TZE200_uebojraa'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_yh7aoahi'}],
        model: 'TS0601_smoke',
        vendor: 'TuYa',
        description: 'Smoke sensor',
        whiteLabel: [{vendor: 'TuYa ', model: 'TS0601_smoke'}],
        fromZigbee: [fz.tuya_smoke],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },
        exposes: [e.smoke(), e.battery_low(), e.tamper(), e.battery()],
    },
]

module.exports = definition;

what is not ok ?

jocamane, 2022-02-06:
[image]

jocamane, 2022-02-21:
can anyone help me?

jocamane, 2022-03-18:
I asked for help but one at a time please !!!

Koenkk, 2022-03-18:
Seems this device uses different data points compared to already supported TS0601 smoke sensors. You can find out which via https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

jocamane, 2022-03-18:
Thanks Koenkk,
After i find the "codes" what do i need to do ?

In my case:
silence = Code 16
smoke detection state = code 1
Fault Alarm = Code 11
battery level state = code 14
battery level = Code 15

[image]

Koenkk, 2022-03-19:
That looks like the correct info, if 1 is indeed the smoke state it should already work with the converter from https://github.com/Koenkk/zigbee2mqtt/issues/11119#issuecomment-1030918708 , what is logged when you trigger the sensor with smoke?

jocamane, 2022-03-19:
Hi,
When exist smoke the sensor work ok and indicate true, but the battery don't work and errors continue to appear in the log:

Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}

jocamane, 2022-03-19:
and after errors :

Error Failed to configure '0x84fd27fffeca5ada', attempt 1 (Error: Bind 0x84fd27fffeca5ada/1 genBasic from '0x00212effff07afaa/1' failed (waiting for response TIMEOUT) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:761:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:34:9) at Object.configure (/app/dist/util/externally-loaded.js:23:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))

jocamane, 2022-03-19:
and after each detection:

Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #14 with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[2]}}
Warning zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}

Koenkk, 2022-03-19:
Could you try with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 14:
                    // battery state, ignore
                    break
                case 15:
                    result.battery = value;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['5p1vj8r'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t5p1vj8r'}, {modelID: 'TS0601', manufacturerName: '_TZE200_uebojraa'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_yh7aoahi'}],
    model: 'TS0601_smoke',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [],
    exposes: [e.smoke(), e.battery_low()],
};

module.exports = definition;

```

jocamane, 2022-03-19:
It's better. No errors, but...
The battery apear in "state" but not in "Exposes",

[image]
[image]
[image]

strange

also in HA devices:
[image]

Koenkk, 2022-03-19:
try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 14:
                    // battery state, ignore
                    break
                case 15:
                    result.battery = value;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['5p1vj8r'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t5p1vj8r'}, {modelID: 'TS0601', manufacturerName: '_TZE200_uebojraa'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_yh7aoahi'}],
    model: 'TS0601_smoke',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [],
    exposes: [e.smoke(), e.battery()],
};

module.exports = definition;

```

jocamane, 2022-03-19:
spectacular !! Thank you

PS: Can you now solve the others that have been pending for some time? 
ie. https://github.com/Koenkk/zigbee2mqtt/issues/11392 and the #10903

jocamane, 2022-03-19:
Please include more this  TS0601_TZE200_vzekyi4c
is another smoke sensor that i have.
I will try add in external, but in future versions include twice

thanks

jocamane, 2022-03-19:
The new version:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 14:
                    // battery state, ignore
                    break
                case 15:
                    result.battery = value;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['5p1vj8r'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t5p1vj8r'}, {modelID: 'TS0601', manufacturerName: '_TZE200_uebojraa'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_yh7aoahi'}, {modelID: 'TS0601', manufacturerName: '_TZE200_vzekyi4c'}],
    model: 'TS0601_smoke',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [],
    exposes: [e.smoke(), e.battery()],
};

module.exports = definition;

closed 2022-03-20 by jocamane

Koenkk, 2022-03-20:
great, pushed the fix

jocamane, 2022-03-20:
ok
Z2M getting better 

you are the boss :)

viko-repo, 2022-04-28:
Hi, can you help please
Hey
I have it appears exactly as it says here but there is still an error

https://www.aliexpress.com/item/1005002214198266.html

Manufacturer: _TZE200_dq1mfjug

[image]
[image]
[image]

jmarfer, 2022-06-19:
> The new version:
>
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = [...]

Good afternoon, where do I have to put this code for the smoke sensor to work?

Thank you