Koenkk/zigbee2mqtt #11123: Xaiomi RTCGQ01LM occupancy_timeout can't be configured

edulechuga, 2022-01-30:
### What happened?

Hi All, how are you?

I have configured the devices.yaml file to make a Xiaomi's RTCGQ01LM motion sensor to be able to "make another reading" after 60 seconds as described on documentation (https://www.zigbee2mqtt.io/devices/RTCGQ01LM.html#notes) but I keep getting an error message like this one:

MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Xiaomi Motion Sensor"},"error":"Device 'Xiaomi Motion Sensor' cannot be configured","status":"error","transaction":"gurj5-3"}'

Here is my devices file:

'0x00158d000457e764':
  friendly_name: Xiaomi Motion Sensor
  no_occupancy_since:
    - 10
    - 60
  occupancy_timeout: 60

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20220103

### Adapter

SONOFF Zigbee 3.0 USB Dongle

Koenkk, 2022-01-31:
There is  no need to click the configure button, the `occupancy_timeout` should already be used.

edulechuga, 2022-01-31:
Hi @Koenkk I'm sorry but I didn't get you. I didn't press the configure button. What I did was to go into the devices.yaml file and include that information on the first topic. 

But the effect was exactly the same from the standard 90 seconds, even configured to be only 60 seconds.

Koenkk, 2022-01-31:
Did you restart z2m after setting this?