Koenkk/zigbee2mqtt #11126: Aqara Opple switches reports non-existing "state" property

uncle-fed, 2022-01-30:
### What happened?

The 1 and 2 button Aquara Opple switches WXCJKG11LM / WXCJKG12LM (quite possibly 3 button switch WXCJKG13LM as well) report property `{"state": OFF}` from time to time (or if the "pair / wake up" button on the back is pressed).

### What did you expect to happen?

This property does not belong to a non-wired switch, as it is a "stateless click event generator" only.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20220103

### Adapter

CC1352P2

Koenkk, 2022-01-31:
Should be fixed now, it will not be published anymore however the cached property needs to be removed manually from `data/state.json`.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

uncle-fed, 2022-01-31:
I can confirm, this has been now fixed, indeed. Thanks a lot!

leroyloren, 2022-02-06:
the temperature is still provided

`{
    "battery": 80,
    "last_seen": "2022-02-06T12:05:36+01:00",
    "linkquality": 72,
    "temperature": 25,
    "voltage": 2900
}`

state.json
` "xx": {
        "voltage": 2900,
        "battery": 80,
        "temperature": 25
    }
`

uncle-fed, 2022-02-06:
I am not a 100% certain about the `temperature`, because the official documentation page mentions `temperature_precision`.
But then again, the same document does not mention that device is supposed to expose the `temperature`.
It also mentions `illuminance_precision` which definitely does not apply to these switches.
I know that some Xiaomi/Aqara switches do have temperature sensors, so I tried to hide this switch in the fridge. :-)
Seems like the temperature reading did not change.

@Koenkk I guess we need more things removed from these switches and from the accompanying documentation.
Also, I detected that a single button switch still reports the "state" propert (previously I only tested 2-button switches).

So we need to remove:

- "state" from 1-button switch (maybe 3-button as well, but I don't have it to test it)
- "temperature" from all 3 switches
- update doc pages for all 3 switches and remove `temperature_precision`, `temperature_calibration`, `illuminance_precision` and `illuminance_calibration`.

Koenkk, 2022-02-06:
> "state" from 1-button switch (maybe 3-button as well, but I don't have it to test it)

this is already done for the WXCJKG11LM, WXCJKG12LM and WXCJKG13LM. But after updating to z2m 1.23.0 you need to remove it manually from data/state.json

> "temperature" from all 3 switches

done

> update doc pages for all 3 switches and remove temperature_precision, temperature_calibration, illuminance_precision and illuminance_calibration.

already fixed in the dev branch (docs will be updated with the next release on 1 march

uncle-fed, 2022-02-06:
> this is already done for the WXCJKG11LM, WXCJKG12LM and WXCJKG13LM. But after updating to z2m 1.23.0 you need to remove it manually from data/state.json

@Koenkk could you please double check it. I have all caching disabled since forever (you might remember, I was the one responsible for introducing new  'no caching' parameters to the config). Running the latest dev and state is being reported for 1 button switch but not for 2 button. Thanks!

Koenkk, 2022-02-06:
@uncle-fed what value is reported for the `state`? I cannot find any path in https://github.com/Koenkk/zigbee-herdsman-converters/blob/e8126f5910345ebd110a10df2a233fee3521bc99/converters/fromZigbee.js#L5252 which adds the state property.

uncle-fed, 2022-02-06:
> what value is reported for the `state`? I cannot find any path in

OFF

I'm going to double check the state caching on my side, just to be sure.

uncle-fed, 2022-02-06:
@Koenkk: I've checked my setup, I don't even have `state.json` file at all.

If this is of any help, the state is still being reported when you press the "wake up / pair" button on the back of the switch, not when you press the switch button ordinarily. 

And btw, I've pulled latest dev and I can still see temperature being reported, together with the state on the 1 button switch.

🤷

leroyloren, 2022-02-07:
With yesterday Edge it is ok for me. (WXCJKG13LM)

`{
    "battery": 100,
    "last_seen": "2022-02-07T16:38:45+01:00",
    "linkquality": 76,
    "voltage": 3113
}`

uncle-fed, 2022-02-08:
@Koenkk: here it is, with today's pull of the dev branch:

```
Received Zigbee message from 'switch/wall', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.remote.b286opcn01"}' from endpoint 1 with groupID 0
Received Zigbee message from 'switch/wall', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,115,11,3,40,25,4,33,168,67,5,33,106,0,6,36,4,0,0,0,0,8,33,17,1,10,33,243,120,12,32,1,100,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
lumi.remote.b286opcn01: unknown index 4 with value 17320
lumi.remote.b286opcn01: unknown index 6 with value 4
lumi.remote.b286opcn01: unknown index 8 with value 273
lumi.remote.b286opcn01: unknown index 10 with value 30963
lumi.remote.b286opcn01: unknown index 12 with value 1
MQTT publish: topic 'z2m/switch/wall', payload '{"battery":60,"linkquality":72,"state":"OFF","temperature":25,"voltage":2931}'
```

Can it be that endpoint ID(s) are different on the 1-button switch (compared to 2 and 3 button switches) ?

Koenkk, 2022-02-08:
Damm, turned out this was a [trailing space](https://github.com/Koenkk/zigbee-herdsman-converters/commit/8b54505d7ba875c81f8edf3140bc9664255c7fbd) 😆 

Fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

uncle-fed, 2022-02-09:
@Koenkk that's curious indeed. The switch does not report the `state` anymore, but it still reports the `temperature` :-)
Can there be [more whitespace](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L5365) to erase? ;-)

Koenkk, 2022-02-09:
Sigh 😄 Done now, thanks for your alertness 😄

uncle-fed, 2022-02-09:
Thanks a lot for your quick reaction / feedback!

closed 2022-02-09 by uncle-fed