Koenkk/zigbee2mqtt #11132: [New device support]: CASAIA CSAC-451-DC Dry Contact relay

KKox-26, 2022-01-30:
### Link

https://casaia.fr/product/module-relais-contact-sec-zigbee/

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0x3c6a2cfffed0ab8b","nwkAddr":1951,"manufId":4412,"manufName":"CASAIA","powerSource":"Mains (single phase)","modelId":"CSAC451","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":50257,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"CSAC451","manufacturerName":"CASAIA","powerSource":1,"zclVersion":1,"dateCode":"20201228"}},"genOnOff":{"attributes":{"onOff":1}},"genOnOffSwitchCfg":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad8b4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"dateCode":"20201228","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1643562499119,"useImplicitCheckin":true}

### Comments

I tried the following external converter, the switch works but i have to reset manually after an activation by clicking on it. 

[image]

Is there a better way to integrate to get only a one action button ?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = { zigbeeModel: ['CSAC451'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
model: 'CSAC451', // Vendor model number, look on the device for a model number
vendor: 'CASAIA', // Vendor of the device (only used for documentation and startup logging)
description: 'CASAIA contact sec', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
fromZigbee: [fz.on_off], // We will add this later
toZigbee: [tz.on_off],
exposes: [e.switch()],
configure: async (device, coordinatorEndpoint, logger) => {
		const endpoint = device.getEndpoint(1);
		await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
		await reporting.onOff(endpoint);},
};

module.exports = definition;
```

closed 2022-03-09 by github-actions[bot]

KKox-26, 2022-06-05:
this device is still not well supported can i have some heklp to improve it ?

Koenkk, 2022-06-06:
> Is there a better way to integrate to get only a one action button ?

To me it looks just like an on/off device, so you should turn it off manually.

KKox-26, 2022-06-06:
Thanks for your awnser !
The device timeout is 2 seconds, i was wondering if it is not possible to automatically turn off after 2 seconds on ON "state".

Koenkk, 2022-06-07:
I don't see how to do this, this device advertises itself as a regular on/off switch.