Koenkk/zigbee2mqtt #11135: CENTRALITE 3400 ( device announced and no code sent)

kevincaradant, 2022-01-30:
### What happened?

Everytime I enter a code ( 6 digits ) , I got device announced 
It should send the code. I got many issue like this since the version 1.20.0. But now 1.22.2 it seems fully broken because I got `device anounced` and `code_action: null` but I wrote a code, so it should display it.

### What did you expect to happen?

Centralite understands the code that I entered and send a message wiht the code.

### How to reproduce it (minimal and precise)

Click on one of three main button.
Enter a code
Hoping that it works and Centralite send to Z2M the code.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211210

### Adapter

Launchpad CC1352-P2

### Debug log

```
Debug Received Zigbee message from '0x000d6f000d8a5646', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x000d6f000d8a5646', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"linkquality":87,"occupancy":true,"temperature":24.71,"voltage":3100}'
Debug Received Zigbee message from '0x000d6f000d8a5646', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x000d6f000d8a5646', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"linkquality":87,"occupancy":true,"temperature":24.71,"voltage":3100}'
Info MQTT publish: topic 'zigbee2mqtt/router_pc_desktop', payload '{"energy":487.36,"linkquality":87,"power":184.5,"state":"ON"}'
Debug Received Zigbee message from '0x000d6f000d8a5646', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x000d6f000d8a5646', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"linkquality":91,"occupancy":true,"temperature":24.71,"voltage":3100}'
Debug Received Zigbee message from '0x000d6f000d8a5646', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x000d6f000d8a5646', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"linkquality":91,"occupancy":true,"temperature":24.71,"voltage":3100}'
Debug Received Zigbee message from 'power_lavelinge_bathroom', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,17827],"currentSummReceived":[0,63],"instantaneousDemand":6}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x000d6f000d8a5646', type 'commandArm', cluster 'ssIasAce', data '{"armmode":3,"code":"260894","zoneid":1}' from endpoint 1 with groupID 0
Debug Device '0x000d6f000d8a5646' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f000d8a5646","ieee_address":"0x000d6f000d8a5646"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x000d6f000d8a5646"},"type":"device_announced"}'
Debug Retrieving state of '0x000d6f000d8a5646' after reconnect
Debug Received Zigbee message from '0x000d6f000d8a5646', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x000d6f000d8a5646', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"linkquality":98,"occupancy":true,"temperature":24.71,"voltage":3100}'
Debug Received Zigbee message from '0x000d6f000d8a5646', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x000d6f000d8a5646', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"linkquality":102,"occupancy":true,"temperature":24.71,"voltage":3100}'
```

Koenkk, 2022-01-31:
Could you check with the latest dev? Looks like #10504 
 (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kevincaradant, 2022-01-31:
Arf oki sorry I didn't know the other name for the centralite.

Perfect it works :)

Thanks

closed 2022-01-31 by kevincaradant

GLehnhoff, 2023-06-26:
@kevincaradant , did you get 3500-D running? I don't. It is paired by I have lots of problems. Not sure, whether this thread here is the right place to publish my problems.