Koenkk/zigbee2mqtt #11145: TRADFRI ON/OFF (E1743) & Hue Dimmer switch 324131092621 sending two events when pressed once

UhtredTheBold, 2022-01-31:
### What happened?

Two "on" events are sent when pressed once, which make automations based on number of presses fail

This happens with any button pressed; on, off, up-press and down-press

This looks very similar to https://github.com/Koenkk/zigbee2mqtt/issues/11056 (closed)

### What did you expect to happen?

When pressed once, only one event is sent.

### How to reproduce it (minimal and precise)

Setup a E1743 wireless switch, and press the on button

### Zigbee2MQTT version

1.22.2-dev

### Adapter firmware version

20211217

### Adapter

zzh!

### Debug log

`debug 2022-01-31 09:16:34: Received Zigbee message from '0xbc33acfffeb5bb1f', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2022-01-31 09:16:34: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffeb5bb1f', payload '{"action":"on","battery":87,"click":"on","linkquality":61,"update":{"state":"idle"},"update_available":false}'
info  2022-01-31 09:16:34: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffeb5bb1f', payload '{"action":"","battery":87,"click":null,"linkquality":61,"update":{"state":"idle"},"update_available":false}'
info  2022-01-31 09:16:34: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffeb5bb1f', payload '{"action":null,"battery":87,"click":"","linkquality":61,"update":{"state":"idle"},"update_available":false}'
info  2022-01-31 09:16:34: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffeb5bb1f/action', payload 'on'
info  2022-01-31 09:16:34: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffeb5bb1f/click', payload 'on'`

Koenkk, 2022-01-31:
I don't see two `on` events being send in the log to the same topic, the debug log you provided looks as expected.

UhtredTheBold, 2022-02-02:
Yeah I have confirmed that by viewing mqtt directly. I don't understand where the extra actions are coming from in home assistant and nodered. I've filtered them out in my automations.

closed 2022-02-02 by UhtredTheBold