Koenkk/zigbee2mqtt #11149: Unable to pair TuYa Smart Knob ERS-10TZBVK-AA

wildekek, 2022-01-31:
### What happened?

I am unable to pair the device to Zigbee2MQTT. It does not show up in the device list and there is no information in the logs.

### What did you expect to happen?

The device is discovered and shows up in the device list.
Also, pairing instructions on the [device page](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html).

### How to reproduce it (minimal and precise)

- Set Zigbee2MQTT in `Permit join (all )` mode
- Reset the Smart Knob by pressing the reset button until the light starts blinking.
- Wait for the device to be discovered, which never happens.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Slaesh's CC2652RB stick

wildekek, 2022-01-31:
Same for me: https://github.com/Koenkk/zigbee2mqtt/issues/11149

toogooda, 2022-01-31:
Same problem, tried so many times with both CC2531 and SONOFF Zigbee 3.0 USB Dongle Plus.
Nothing in logs that I could find

kirovilya, 2022-01-31:
Are you using Dev/Edge version of z2m?

wildekek, 2022-01-31:
> Are you using Dev/Edge version of z2m?

No, latest release version ([1.22.2](https://github.com/Koenkk/zigbee2mqtt/tree/1.22.2))

TEF2one, 2022-01-31:
No issue on 1.22.2 using Conbee2

toogooda, 2022-01-31:
1.22.2-1 z2m SONOFF Zigbee 3.0 USB Dongle Plus

toogooda, 2022-01-31:
> No issue on 1.22.2 using Conbee2

Did you do anything other than setting z2m to permit join and holding the reset on the smart knob?

TEF2one, 2022-02-01:
> > No issue on 1.22.2 using Conbee2
> 
> @TEF2one did you do anything other than setting z2m to permit join and holding the reset on the smart knob?

@toogooda I didn't do anything extra

kirovilya, 2022-02-01:
everything works. but perhaps this is because I first used this knob with a native tuya-gateway.
yesterday to make a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3789) by analogy with other TS004F devices - perhaps it need a special initialization procedure https://github.com/Koenkk/zigbee2mqtt/discussions/7158

kirovilya, 2022-02-01:
> @kirovilya mine were never paired with anything else prior.

maybe that's the reason why it works for me.
wait for my changes to be accepted, you will need to re-pair the device

r-vit, 2022-02-02:
I was able to pair this device with z2m without previously pairing it to the Tuya gateway. While doing that I found out that the device needed a fresh battery to pair, and I had to allow pairing on the nearest router only.

wildekek, 2022-02-04:
Thanks @r-vit, making sure the battery was full made all the difference. I hooked up the device to a power supply and below 3v, the device turns on and responds to presses, but it refuses to go into pairing mode. Above 3v it pairs just fine. I'll update the docs for this device with pairing instructions and then I'll close this one.

closed 2022-02-04 by wildekek

koenhendriks, 2023-04-16:
@wildekek I'm running into this issue. I used a fresh battery and the green led blinks for about 30 seconds. But still I'm not able to pair it. 

Can you explain how you solved it? Did you connect it to an Arduino for 3v power supply?

wildekek, 2023-04-16:
> Did you connect it to an Arduino for 3v power supply?
@koenhendriks In my case I used a bench power supply (RD6006), but any supply running at AT LEAST 3v should work, so an Arduino (3.3v) should be fine if that is all you have. The power supply is useful for ruling out a low voltage issue during pairing. After pairing, the device is stable and keeps working fine on a battery for me.

If the pairing does not work, even though you are positive it has enough power, I'd recommend pairing it next to your zigbee coordinator. Tuya devices are known to not always play nicely with all zigbee routers. _Veel succes man!_

timm0e, 2024-08-27:
I also was unable to pair the smart knobs, but was able to solve it:

My coordinator is a CC2531, and I had it on Firmware 20201128. 
Turns out updating to 20211115 solved the issue.