Koenkk/zigbee2mqtt #11152: WHD02 paired as TS0001

wileu, 2022-01-31:
### What happened?

I have that switch:
https://www.zigbee2mqtt.io/devices/WHD02.html

but is paired as:
https://www.zigbee2mqtt.io/devices/TS0001.html#tuya-ts0001

My switch have an option "Power_on_behavior", ts0001  doesn't.
There https://github.com/Koenkk/zigbee2mqtt/issues/8380#issue-971092601 is the same problem.

### What did you expect to happen?

Get paired as WHD02.

### How to reproduce it (minimal and precise)

Just pair.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Egony v4 CC2652P

Koenkk, 2022-01-31:
could you provide the data/database.db entry of your device?

wileu, 2022-01-31:
`{"id":27,"type":"Router","ieeeAddr":"0xa4c1384ded6063dc","nwkAddr":47532,"manufId":4417,"manufName":"_TZ3000_npzfdcof","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_npzfdcof","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0ed5c","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1643653836208,"useImplicitCheckin":true}`

Koenkk, 2022-02-01:
fixed

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-01 by Koenkk

pupsnet, 2022-12-18:
Hello, same issue, could it be fixed?

Zigbee2MQTT version [1.28.4](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.4) commit: [unknown]
Coordinator type zStack3x0
Coordinator revision 20220219
Frontend version 0.6.119

two relays WHD02 paired as TS0001

==1st==
{"id":6,"type":"Router","ieeeAddr":"0x847127fffe0a9b29","nwkAddr":49228,"manufId":4098,"manufName":"_TZ3000_hktqahrq","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_hktqahrq","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b4950b","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1671390666466,"defaultSendRequestWhen":"immediate"}

==2nd==

{"id":11,"type":"Router","ieeeAddr":"0x847127fffe0aa7fa","nwkAddr":37690,"manufId":4098,"manufName":"_TZ3000_hktqahrq","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_hktqahrq","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b4950b","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1671390910279,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-01-17:
@GuidoKJ already fixed for this case which will be included in the next release on 1 February (https://github.com/Koenkk/zigbee2mqtt/issues/15957)