Koenkk/zigbee2mqtt #11155: Frontend not starting on >- 1.22.0

cradcore, 2022-01-31:
### What happened?

Brand new install, using a Conbee II, below is my configuration.yaml:
```
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://<ip>'
  user: <user>
  password: <pass>
  reject_unauthorized: false
  include_device_information: true
serial:
  adapter: deconz
  port: /dev/ttyACM1
frontend:
  port: 8080
advanced:
  log_level: debug
  network_key: GENERATE
```
The frontend never starts. Updated logging to debug, but it gives no useful information. It connects to the adapter just fine, but never starts the frontend, and never connects the MQTT broker (in my case, Mosquitto) Here are the verbose logs:

What is interesting is, if I go back to 1.21.2, it starts up just fine. I am then able to upgrade to 1.22.0 and it does start the frontend. So there is a workaround. But if I start with an empty config folder with version >= 1.22.0, it never starts the frontend, connects to Mosquitto, or gives me any information on why.

### What did you expect to happen?

The frontend to start on a clean install

### How to reproduce it (minimal and precise)

Run container as described by docs with an empty config (clean install)

### Zigbee2MQTT version

1.22.0

### Adapter firmware version

0x26580700

### Adapter

    ConBee2/RaspBee2

### Debug log

```
Using '/app/data' as data directory
Zigbee2MQTT:debug 2022-01-31 17:10:34: Can't load state from file /app/data/state.json (doesn't exist)
Zigbee2MQTT:info  2022-01-31 17:10:34: Logging to console and directory: '/app/data/log/2022-01-31.17-10-32' filename: log.txt
Zigbee2MQTT:info  2022-01-31 17:10:34: Starting Zigbee2MQTT version 1.22.2 (commit #414c51f)
Zigbee2MQTT:info  2022-01-31 17:10:34: Starting zigbee-herdsman (0.13.188)
Zigbee2MQTT:debug 2022-01-31 17:10:34: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM1"}}'
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-01-31 17:10:34: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-01-31 17:10:34: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:debug 2022-01-31 17:10:37: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2022-01-31 17:10:37: Currently 0 devices are joined:
Zigbee2MQTT:warn  2022-01-31 17:10:37: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-01-31 17:10:37: Allowing new devices to join.
Zigbee2MQTT:warn  2022-01-31 17:10:37: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-01-31 17:10:37: Zigbee: allowing new devices to join.
```

IanAdd, 2022-02-01:
Identical issue with 1.22.2
HA 2021.12.10, mqtt on a separate machine (as is implied above) also using conbee2 with adapter: deconz

Unfortunately I am installing for the first time and hence don't have access to 1.19 (which @cradcore says works).
I did try edge but no difference.

A strange observation is that changing the frontend port does not make the links change, they still show failing to connect to 8099.
System log always shows:
```
ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.3:8099 ssl:default [Connect call failed ('172.30.33.3', 8099)]
```

My debug log....
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[14:40:29] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[14:40:30] INFO: MQTT available, fetching server detail ...
[14:40:31] INFO: Previous config file found, checking backup
[14:40:31] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[14:40:31] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[14:40:32] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.2 start
> node index.js
Zigbee2MQTT:debug 2022-02-01 14:40:44: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-02-01 14:40:44: Logging to console and directory: '/config/zigbee2mqtt/log/2022-02-01.14-40-36' filename: log.txt
Zigbee2MQTT:debug 2022-02-01 14:40:44: Removing old log directory '/config/zigbee2mqtt/log/2022-01-31.11-59-46'
Zigbee2MQTT:info  2022-02-01 14:40:44: Starting Zigbee2MQTT version 1.22.2 (commit #unknown)
Zigbee2MQTT:info  2022-02-01 14:40:44: Starting zigbee-herdsman (0.13.188)
Zigbee2MQTT:debug 2022-02-01 14:40:44: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-02-01 14:40:44: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-02-01 14:40:44: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:debug 2022-02-01 14:40:44: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2022-02-01 14:40:44: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-02-01 14:40:44: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-01 14:40:45: Connecting to MQTT server at 192.168.0.50:1883
Zigbee2MQTT:debug 2022-02-01 14:40:45: Using MQTT login with username: mqtt
Zigbee2MQTT:debug 2022-02-01 14:45:44: Saving state to file /config/zigbee2mqtt/state.json
```

cradcore, 2022-02-01:
> Unfortunately I am installing for the first time and hence don't have access to 1.19 (which @cradcore says works).
> I did try edge but no difference.

@IanAdd  are you using docker? You can either specify the image version: (here I have 1.20.0)
[image]

Or checkout that version from git:
`git checkout 1.19.2`

and then build the docker container locally and use that

IanAdd, 2022-02-02:
@cradcore I am just using vanilla HA installation on pi4 with SSD. No docker access.

IanAdd, 2022-02-02:
1.23.0 is no different :-(

rchrd, 2022-02-02:
For me all versions up to 1.23.0 start the frontend, but 1.23.0 does not.

For example the logs from 1.22.2:
```
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-02 11:14:08: Logging to console and directory: '/app/data/log/2022-02-02.11-14-08' filename: log.txt
Zigbee2MQTT:info  2022-02-02 11:14:08: Starting Zigbee2MQTT version 1.22.2 (commit #414c51f)
Zigbee2MQTT:info  2022-02-02 11:14:08: Starting zigbee-herdsman (0.13.188)
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-02-02 11:14:08: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-02-02 11:14:08: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2022-02-02 11:14:08: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-02-02 11:14:08: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-02 11:14:09: Connecting to MQTT server at mqtt://mqtt.smarthome.svc
Zigbee2MQTT:info  2022-02-02 11:14:09: Connected to MQTT server
Zigbee2MQTT:info  2022-02-02 11:14:09: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-02-02 11:14:09: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2022-02-02 11:14:09: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"414c51f","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.22.2"}'
```

and from 1.23.0
```
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-02 11:19:14: Logging to console and directory: '/app/data/log/2022-02-02.11-19-14' filename: log.txt
Zigbee2MQTT:info  2022-02-02 11:19:14: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-02 11:19:14: Starting zigbee-herdsman (0.14.10)
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-02-02 11:19:14: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-02-02 11:19:14: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2022-02-02 11:19:14: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-02-02 11:19:14: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-02 11:19:15: Connecting to MQTT server at mqtt://mqtt.smarthome.svc
Zigbee2MQTT:info  2022-02-02 11:19:16: Connected to MQTT server
Zigbee2MQTT:info  2022-02-02 11:19:16: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-02-02 11:19:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"6ea61ba","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.23.0"}'
```

I'm running zigbee2mqtt on Kubernetes (k3s).

rchrd, 2022-03-05:
It works again on version 1.24.0 👍🏻

birarda, 2022-03-16:
I still have this problem on 1.24.X, running via unraid. Manually specifying 1.20.0 works.

closed 2022-04-24 by github-actions[bot]

pleaseremove, 2022-04-24:
Hi Guys. Does anyone have an answer for this? I tried starting out with an old version, which seems to start the ui, but only just (perhaps loading my 1.22 config in is there issue there. Then I upgrade as per the upgrade instructions to latest version and now I get no UI. I have gone back to basics with a blank config and started from the begging and im basically at the point of giving up. I have now rebuilt in docker maybe 20 times and now in a VM and over and over, no UI starts! This is driving me crazy.

kyleaffolder, 2022-05-09:
@pleaseremove Hey Craig, did you ever get this figured out then? If so, I could use some help.

My logs look good and I have Zigbee2MQTT talking to my Mosquitto MQTT broker. I am running two Docker containers and they share the same network.

```
# docker-compose.yml

version: "3"
services:
  mqtt:
    image: "eclipse-mosquitto:latest"
    container_name: mosquitto
    volumes:
      - "/opt/mosquitto/config/mosquitto.conf:/mosquitto/config/mosquitto.conf:ro"
      - "mosquitto_data:/mosquitto/data"
    ports:
      - "1883:1883"

  zigbee2mqtt:
    image: "ghcr.io/koenkk/zigbee2mqtt:latest"  # appears to be version 1.25.1
    container_name: zigbee2mqtt
    environment:
      - "TZ=Europe/Zurich"
    volumes:
      - "/opt/zigbee2mqtt/data:/app/data"
      - "/run/udev:/run/udev:ro"
    ports:
      - "8088:8080/tcp"  # Frontend port
    devices:
      - "/dev/ttyACM0:/dev/ttyACM0"
    depends_on:
      - mqtt

volumes:
  mosquitto_data:
```

```
# Zigbee2MQTT configuration.yaml

mqtt:
  server: mqtt://mosquitto:1883
  base_topic: zigbee2mqtt
  user: '!secret user'
  password: '!secret password'
serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend:
  port: 8088
```

I have opened port 8088 in my firewall (see above as I'm mapping internal port `8080` of Zigbee2MQTT container to port `8088` on my host. If I'm understanding correctly, theoretically, I should be able to then navigate to `192.168.x.x:8088` and see the Zigbee2MQTT front-end, however, all I'm getting is a broken page. :(

I can supply complete logs if anyone would like, but in short, I am seeing the following lines in the logs... *"Connected to MQTT server"* and *"Started frontend on port 0.0.0.0:8088"*. So, it would seem that my connection to my MQTT broker (mosquitto) is working and also that the frontend is starting on port 8088 with no failure.

May need to open a new issue either here or via https://github.com/nurikk/zigbee2mqtt-frontend/issues.

lukepighetti, 2022-06-07:
My frontend wasn't loading. No errors. Apparently zigbee2mqtt daemon wasn't connecting to MQTT server. Once I got that working, the frontend started working.

pleaseremove, 2022-06-08:
FYI, forgot to come back here and say what my issue was. Turned out my USB dongle was not getting enough power, so was not there to connect to. As such the rest of the process wasn't completing. The logs weren't that helpful, ended up going to raw log output and reading the code to work out where it failed. As such the solution was to use a different USB port... The things you learn... Thanks all

lukepighetti, 2022-06-08:
I think the big takeaway from this issue is that z2m would benefit from more robust logging

heeen, 2022-11-15:
I have this issue and going back to 1.20.0 makes the frontend start with otherwise identical configs

The-Game-Breaker, 2022-12-27:
> I think the big takeaway from this issue is that z2m would benefit from more robust logging

This gave me the idea to plug it in directly to cut out the voltage drop over the extension and this worked. Looks like I am shortening my extention cable.

The-Game-Breaker, 2022-12-27:
> FYI, forgot to come back here and say what my issue was. Turned out my USB dongle was not getting enough power, so was not there to connect to. As such the rest of the process wasn't completing. The [...]

This gave me the idea to plug it in directly to cut out the voltage drop over the extension and this worked. Looks like I am shortening my extention cable.

seventhsite, 2023-02-17:
Same issue on version 1.30.1 (commit #eb878d3). 
Config simply 
`frontend: true`
No logs, no open port

```
debug 2023-02-18 02:41:46: Loaded state from file /app/data/state.json
info  2023-02-18 02:41:46: Logging to console and directory: 'data/log/' filename: z2m_2023-02-18.02-41-46.log
info  2023-02-18 02:41:46: Starting Zigbee2MQTT version 1.30.1 (commit #eb878d3)
info  2023-02-18 02:41:46: Starting zigbee-herdsman (0.14.89)
debug 2023-02-18 02:41:46: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app
/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[23],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"baudRat
e":115200,"path":"/dev/ttyAML2","rtscts":false}}'
info  2023-02-18 02:41:55: zigbee-herdsman started (resumed)
info  2023-02-18 02:41:55: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210218,"transportrev":2},"type":"zStack3x0"}'
debug 2023-02-18 02:41:55: Zigbee network parameters: {"channel":23,"extendedPanID":"0x00124b00215bac80","panID":6754}
info  2023-02-18 02:41:55: Currently 0 devices are joined:
info  2023-02-18 02:41:55: Zigbee: disabling joining new devices.
info  2023-02-18 02:41:56: Connecting to MQTT server at mqtt://192.168.7.3:1883
debug 2023-02-18 02:41:56: Using MQTT keepalive: 60
debug 2023-02-18 02:41:56: Using MQTT login with username: homeassistant
debug 2023-02-18 02:41:56: Using MQTT client ID: 'homeassistant'
```

```
seventh@jethub-j80:~/z2m$ ss -tulpn
Netid               State                Recv-Q               Send-Q                               Local Address:Port                                Peer Address:Port               Process               
udp                 UNCONN               0                    0                                    127.0.0.53%lo:53                                       0.0.0.0:*                                        
udp                 UNCONN               0                    0                                        127.0.0.1:323                                      0.0.0.0:*                                        
udp                 UNCONN               0                    0                                            [::1]:323                                         [::]:*                                        
tcp                 LISTEN               0                    4096                                 127.0.0.53%lo:53                                       0.0.0.0:*                                        
tcp                 LISTEN               0                    128                                        0.0.0.0:22                                       0.0.0.0:*                                        
tcp                 LISTEN               0                    100                                        0.0.0.0:1883                                     0.0.0.0:*                                                                          
tcp                 LISTEN               0                    128                                           [::]:22                                          [::]:*                                        
tcp                 LISTEN               0                    100                                           [::]:1883                                        [::]:*                                        

```

Any idea how to run frontend or get a more detailed logs?

saiedt, 2023-02-21:
I am not able to get the frontend running, neither with 1.30.1 nor with 1.30.1.-dev. Everything else seems to work as expected, e.g. I am able to pair devices and I get the mqtt messages.

My configuration:
```
$ cat configuration.yaml
homeassistant: false
experimental:
  new_api: true
fronend:
  port: 8080
  host: localhost
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
serial:
  port: /dev/ttyACM0
advanced:
  channel: 25
  log_level: debug
devices: devices.yaml
```

The log output:
```
$ cat log/2023-02-21.11-39-32/log.txt
debug 2023-02-21 11:39:32: Can't load state from file state.json (doesn't exist)
info  2023-02-21 11:39:32: Logging to console and directory: 'log/2023-02-21.11-39-32' filename: log.txt
info  2023-02-21 11:39:32: Starting Zigbee2MQTT version 1.30.1-dev (commit #e934b0d)
info  2023-02-21 11:39:32: Starting zigbee-herdsman (0.14.93)
debug 2023-02-21 11:39:32: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"coordinator_backup.json","databaseBackupPath":"database.db.backup","databasePath":"database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2023-02-21 11:39:33: zigbee-herdsman started (resumed)
info  2023-02-21 11:39:33: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"}'
debug 2023-02-21 11:39:33: Zigbee network parameters: {"channel":25,"extendedPanID":"0x00124b0014d9d078","panID":6754}
info  2023-02-21 11:39:33: Currently 0 devices are joined:
warn  2023-02-21 11:39:33: `permit_join` set to  `true` in configuration.yaml.
warn  2023-02-21 11:39:33: Allowing new devices to join.
warn  2023-02-21 11:39:33: Set `permit_join` to `false` once you joined all devices.
info  2023-02-21 11:39:33: Zigbee: allowing new devices to join.
info  2023-02-21 11:39:33: Connecting to MQTT server at mqtt://localhost
debug 2023-02-21 11:39:33: Using MQTT anonymous login
info  2023-02-21 11:39:33: Connected to MQTT server
info  2023-02-21 11:39:33: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
debug 2023-02-21 11:39:34: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"eb878d3d","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":25,"extendedPanID":"0x00124b0014d9d078","panID":6754},"permit_join":true,"version":"1.30.1"}'
info  2023-02-21 11:39:34: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"e934b0d","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":25,"extendedPanID":"0x00124b0014d9d078","panID":6754},"permit_join":true,"version":"1.30.1-dev"}'
info  2023-02-21 11:39:34: Zigbee2MQTT started!
debug 2023-02-21 11:44:32: Saving state to file state.json
```

No frontend started:
```
$ sudo netstat -tupln | grep LISTEN
tcp        0      0 127.0.0.1:1883          0.0.0.0:*               LISTEN      581/mosquitto
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      557/cupsd
tcp6       0      0 ::1:1883                :::*                    LISTEN      581/mosquitto
tcp6       0      0 ::1:631                 :::*                    LISTEN      557/cupsd
```

I have tried different USB ports and tested with only `frontend: true`, also without `new_api` under `experimental`; nothing helps.

I am using a Pi-4B, freshly installed with the "recommended" Raspberry OS Desktop 32 bit; afterwards only installed Mosquitto and Z2M...

Can anyone help?

ampujl, 2023-03-20:
Same issue with  1.30.2-1, RPi3b hassio.
Nothing helps, frontend won't start but all devices works.

cnaslain, 2023-12-01:
Same for me with a fresh install with this docker version: `koenkk/zigbee2mqtt:1.33.2`
An upgrade with 25, 28, 29 and 33 works on another raspberry pi.

Nusfer, 2024-01-08:
My frontend not starting to

`Jan 08 14:22:59 homebridge systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 08 14:22:59 homebridge systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 08 14:22:59 homebridge systemd[1]: zigbee2mqtt.service: Consumed 3.372s CPU time.
Jan 08 14:22:59 homebridge systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 39.
Jan 08 14:22:59 homebridge systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Jan 08 14:22:59 homebridge systemd[1]: zigbee2mqtt.service: Consumed 3.372s CPU time.
Jan 08 14:22:59 homebridge systemd[1]: Started zigbee2mqtt.service - zigbee2mqtt.
Jan 08 14:23:00 homebridge npm[2762]: > zigbee2mqtt@1.35.1 start
Jan 08 14:23:00 homebridge npm[2762]: > node index.js
Jan 08 14:23:01 homebridge npm[2778]: /opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
Jan 08 14:23:01 homebridge npm[2778]:   return new YAMLException(message, mark);
Jan 08 14:23:01 homebridge npm[2778]:          ^
Jan 08 14:23:01 homebridge npm[2778]: YAMLException: duplicated mapping key (21:1)
Jan 08 14:23:01 homebridge npm[2778]:  18 | serial:
Jan 08 14:23:01 homebridge npm[2778]:  19 |   # Location of CC2531 USB sniffer
Jan 08 14:23:01 homebridge npm[2778]:  20 |   port: /dev/ttyUSB0
Jan 08 14:23:01 homebridge npm[2778]:  21 | frontend:
Jan 08 14:23:01 homebridge npm[2778]: ------^
Jan 08 14:23:01 homebridge npm[2778]:  22 | port: 8080
Jan 08 14:23:01 homebridge npm[2778]:  23 | host: 10.0.1.176
Jan 08 14:23:01 homebridge npm[2778]:     at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
Jan 08 14:23:01 homebridge npm[2778]:     at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
Jan 08 14:23:01 homebridge npm[2778]:     at storeMappingPair (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:358:7)
Jan 08 14:23:01 homebridge npm[2778]:     at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1173:9)
Jan 08 14:23:01 homebridge npm[2778]:     at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
Jan 08 14:23:01 homebridge npm[2778]:     at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
Jan 08 14:23:01 homebridge npm[2778]:     at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
Jan 08 14:23:01 homebridge npm[2778]:     at Object.load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1714:19)
Jan 08 14:23:01 homebridge npm[2778]:     at Object.read (/opt/zigbee2mqtt/lib/util/yaml.ts:7:29)
Jan 08 14:23:01 homebridge npm[2778]:     at read (/opt/zigbee2mqtt/lib/util/settings.ts:333:20)`

RPi4
homebridge

4val0v, 2024-03-16:
**Environment:**
- Zigbee2MQTT: v1.36.0
- node: v20.11.1
- npm: 10.2.4
- OS: Ubuntu 22.04.4

[image]

`sudo netstat -ntlp`:
```
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:1883          0.0.0.0:*               LISTEN      828/mosquitto       
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      787/systemd-resolve 
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      858/sshd: /usr/sbin 
tcp6       0      0 ::1:1883                :::*                    LISTEN      828/mosquitto       
tcp6       0      0 :::22                   :::*                    LISTEN      858/sshd: /usr/sbin 
```

4val0v, 2024-03-16:
@Koenkk, @ChrisHae I run it in debug mode
`DEBUG=* node index.js`

I suspect that it doesn't even get to the start of the frontend, it falls into a loop at some point and shows the following: 
```
zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +5ms
```

<details>
  <summary>Spoiler - output log (CLICKABLE)</summary>
  
  ```text
  ```
winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warn"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "http"
winston:create-logger: Define prototype method for "verbose"
winston:create-logger: Define prototype method for "debug"
winston:create-logger: Define prototype method for "silly"
winston:create-logger: Define prototype method for "emerg"
winston:create-logger: Define prototype method for "alert"
winston:create-logger: Define prototype method for "crit"
winston:create-logger: Define prototype method for "error"
winston:create-logger: Define prototype method for "warning"
winston:create-logger: Define prototype method for "notice"
winston:create-logger: Define prototype method for "info"
winston:create-logger: Define prototype method for "debug"
Zigbee2MQTT:debug 2024-03-16 20:54:37: Loaded state from file /opt/zigbee2mqtt/data/state.json
winston:file: written true false
Zigbee2MQTT:info  2024-03-16 20:54:37: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-03-16.20-54-37' filename: log.txt
winston:file: written true false
Zigbee2MQTT:debug 2024-03-16 20:54:37: Removing old log directory '/opt/zigbee2mqtt/data/log/2024-03-16.20-31-49'
winston:file: written true false
winston:file: logged 83 debug 2024-03-16 20:54:37: Loaded state from file /opt/zigbee2mqtt/data/state.json

winston:file: logged 210 info  2024-03-16 20:54:37: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-03-16.20-54-37' filename: log.txt

winston:file: logged 312 debug 2024-03-16 20:54:37: Removing old log directory '/opt/zigbee2mqtt/data/log/2024-03-16.20-31-49'

winston:file: ENOENT ok /opt/zigbee2mqtt/data/log/2024-03-16.20-54-37/log.txt
winston:file: stat done: log.txt { size: 0 }
winston:file: create stream start /opt/zigbee2mqtt/data/log/2024-03-16.20-54-37/log.txt { flags: 'a' }
winston:file: create stream ok /opt/zigbee2mqtt/data/log/2024-03-16.20-54-37/log.txt
winston:file: file open ok /opt/zigbee2mqtt/data/log/2024-03-16.20-54-37/log.txt
Zigbee2MQTT:info  2024-03-16 20:54:37: Starting Zigbee2MQTT version 1.36.0 (commit #2a53b8e)
winston:file: written true false
Zigbee2MQTT:info  2024-03-16 20:54:37: Starting zigbee-herdsman (0.35.1)
winston:file: written true false
Zigbee2MQTT:debug 2024-03-16 20:54:37: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[246,31,74,117,17,3,123,162],"networkKey":"HIDDEN","panID":1298},"serialPort":{"adapter":"deconz"}}'
winston:file: written true false
  serialport/bindings-cpp loading LinuxBinding +0ms
  zigbee-herdsman:adapter No path provided, auto detecting path +0ms
  serialport/bindings-cpp list +0ms
winston:file: logged 81 info  2024-03-16 20:54:37: Starting Zigbee2MQTT version 1.36.0 (commit #2a53b8e)

winston:file: logged 142 info  2024-03-16 20:54:37: Starting zigbee-herdsman (0.35.1)

winston:file: logged 593 debug 2024-03-16 20:54:37: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[246,31,74,117,17,3,123,162],"networkKey":"HIDDEN","panID":1298},"serialPort":{"adapter":"deconz"}}'

  zigbee-herdsman:adapter Auto detected path '/dev/ttyACM0' from adapter 'DeconzAdapter' +55ms
  zigbee-herdsman:deconz:driver Set delay to 0 +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[228,69,212,93,153,112,113,77,177,210,44,206,112,8,211,115],"panID":1298,"extendedPanID":[246,31,74,117,17,3,123,162],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0 +0ms
  serialport/stream opening path: /dev/ttyACM0 +0ms
  serialport/bindings-cpp open +57ms
  serialport/stream _read queueing _read for after open +0ms
  serialport/bindings-cpp/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +1ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +0ms
[... 2469 lines cut ...]
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +4ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +6ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +6ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +5ms
[... cut ...]
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +6ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +6ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +6ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +5ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +5ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending of APS Request +7ms
  zigbee-herdsman:deconz:driver no free slots. Delay sending