Koenkk/zigbee2mqtt #11164: TS0601 _TZE200_fzo2pocs (Zemismart ZM25TQ) - position and motor speed issue

Gimlir, 2022-02-01:
### What happened?

1) Motor doesn't respond (no physical movement of the motor) on changes of slider "position" on Exposes tab but "position" parameter changes in Status accordingly. Also parameter "running": become "true" in status at first and any next change of "position" but no real movement in fact. It keeps "running": "true" until any of STOP, OPEN or CLOSE state sent to device, only after sending of one of such states it become "running": "false" 
2) "motor speed" slider doesn't work and produces error messages (log attached) at any value to set.
3) In case of OPEN or CLOSE state sent to the motor - motor doesn't stop and continues to work until STOP state sent to the motor. Parameter Position doesn't change it's value in State during motor rotation and "running": remains "false" even during physical rotation of the motor i can see live.

### What did you expect to happen?

For 1: Motor should start it's rotation after position slider move, while it rotating - state parameters should change to "running": "true" and "position" should change from previous to new value as long as motor rotating, when motor reached desired position - state should be "running": "false"  and position should be equal to position sent at slider move.
For 2: at  "motor speed" slider change motor should change speed of rotation
For 3: At OPEN or CLOSE state sent motor should start rotation in direction to position 0 or 100 accordingly and stop automatically when desired position reached. During rotation state should change to "running": "true" and revert back to "running": "false" when desired position reached.

### How to reproduce it (minimal and precise)

For 1: Open device page, go to Exposes tab - slide position, look on motor - no rotation, switch to State tab - check running and position values.
For 2: Open device page, go to Exposes tab - slide motor speed, error will popup, switch to Logs tab - check error messages
For 3: Open device page, go to Exposes tab - click on OPEN or CLOSE, motor starts to rotate, position slider not changed, motor doesn't stop rotation until click on STOP. Switch to State tab while motor running - heck running and position values.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210708

### Adapter

zStack3x0 - SONOFF ZB Dongle Zigbee 3.0 (TI CC2652P + CP2102N)

### Debug log

For 1: 
Slider position moved from 88 to 29

> debug Received MQTT message on 'zigbee2mqtt/MotorZM25TQ/set' with data '{"position":29}'
> debug Publishing 'set' 'position' to 'MotorZM25TQ'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,71],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":157}' from endpoint 1 with groupID 0
> info MQTT publish: topic 'zigbee2mqtt/MotorZM25TQ', payload '{"linkquality":117,"motor_speed":false,"position":29,"running":true,"state":"OPEN"}'

Slider position moved from 29 to 100

> debug Received MQTT message on 'zigbee2mqtt/MotorZM25TQ/set' with data '{"position":100}'
> debug Publishing 'set' 'position' to 'MotorZM25TQ'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":160}' from endpoint 1 with groupID 0
> info MQTT publish: topic 'zigbee2mqtt/MotorZM25TQ', payload '{"linkquality":120,"motor_speed":false,"position":100,"running":true,"state":"OPEN"}'

For 2:
Motor speed set to 190

> 
> debug Received MQTT message on 'zigbee2mqtt/MotorZM25TQ/set' with data '{"options":{"motor_speed":190}}'
> debug Publishing 'set' 'options' to 'MotorZM25TQ'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":5,"dp":12,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":103,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":104,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":105,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> info MQTT publish: topic 'zigbee2mqtt/MotorZM25TQ', payload '{"linkquality":120,"motor_speed":false,"position":100,"running":true,"state":"OPEN"}'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":106,"fn":0,"seq":256}' from endpoint 1 with groupID 0

Motor speed set from 190 to 25

> debug Received MQTT message on 'zigbee2mqtt/MotorZM25TQ/set' with data '{"options":{"motor_speed":25}}'
> debug Publishing 'set' 'options' to 'MotorZM25TQ'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":5,"dp":12,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":103,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":104,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":105,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> info MQTT publish: topic 'zigbee2mqtt/MotorZM25TQ', payload '{"linkquality":117,"motor_speed":false,"position":100,"running":true,"state":"OPEN"}'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":106,"fn":0,"seq":256}' from endpoint 1 with groupID 0
[... 2 lines cut ...]
Click on Open - after few seconds on Stop

> debug Received MQTT message on 'zigbee2mqtt/MotorZM25TQ/set' with data '{"state":"OPEN"}'
> debug Publishing 'set' 'state' to 'MotorZM25TQ'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":165}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> info MQTT publish: topic 'zigbee2mqtt/MotorZM25TQ', payload '{"linkquality":123,"motor_speed":false,"position":100,"running":false,"state":"OPEN"}'
> debug Received MQTT message on 'zigbee2mqtt/MotorZM25TQ/set' with data '{"state":"STOP"}'
> debug Publishing 'set' 'state' to 'MotorZM25TQ'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":166}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,12],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> info MQTT publish: topic 'zigbee2mqtt/MotorZM25TQ', payload '{"linkquality":114,"motor_speed":false,"position":88,"running":false,"state":"OPEN"}' 

Click on Close - after few sec on Stop

> debug Received MQTT message on 'zigbee2mqtt/MotorZM25TQ/set' with data '{"state":"CLOSE"}'
> debug Publishing 'set' 'state' to 'MotorZM25TQ'
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":167}' from endpoint 1 with groupID 0
[... cut ...]
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":168}' from endpoint 1 with groupID 0
> debug Received Zigbee message from 'MotorZM25TQ', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,12],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"seq":256}' from endpoint 1 with groupID 0
> info MQTT publish: topic 'zigbee2mqtt/MotorZM25TQ', payload '{"linkquality":108,"motor_speed":false,"position":88,"running":false,"state":"OPEN"}'

Gimlir, 2022-02-01:
Device record from database.db
`{"id":11,"type":"Router","ieeeAddr":"0x0c4314fffe325174","nwkAddr":5435,"manufId":4098,"manufName":"_TZE200_fzo2pocs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_fzo2pocs","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1643727177880,"useImplicitCheckin":true}`

EddieDSuza, 2022-02-02:
I have this Zemismart model working perfectly with Z2M. For the motor to open and close to a desired position - you have to set it via the supplied RF remote - once done - you can have Z2M to work normally. I used this [video](https://www.youtube.com/watch?v=qJzoIED5xVc&t=613s) to set It via the remote

vheneraliuk, 2022-02-05:
Have the same issues with this motor also. After setting limits with Tuya App (you need to pair motor with Tuya Gateway first) and pairing with Zigbee2mqtt USB stick, position slider starting to work properly. Also, the motor automatically stops on those limits. But if we try to set motor speed after that we are getting a ton of errors and limits are resetting. So we definitely need the ability to set limits with zigbee2mqtt and remove the motor speed option because the motor does not support it.

Gimlir, 2022-02-15:
Unfortunately I don't have RF remote for this motor as my decision to buy this exact motor was based on support for this motor via ZigBee, so didn't ordered remote... Anyway recommendation for RF method from @EddieDSuza looks like workaround for me, there should be ZigBee controlled way to set limits and reverse as well as speed control. So ready to help with fix and provide any debug info needed for it.
Will wait for @Koenkk reply, please let me know what can I grab from debug and device traffic in order to help with fix

homeseer666, 2022-02-18:
Same here, ZM25TQ; fortunately, I ordered the RF remote so was able to set limits, but would be great to have the ability to set limits without the remote; also I confirm the motor_speed issue IE: doesn't work and as a bonus limits are reset... available to troubleshoot and help add missing feature. Thanks!

nilvanis, 2022-02-25:
Same here. Unfortunately I don't have the RF remote, nor TuYa gateway, cannot help you much in debugging.
I guess to find out how to calibrate through Z2M one would need to sniff the zigbee packets while setting the limits via TUYA APP.

EddieDSuza, 2022-02-25:
To assist - I have done a [video](https://youtu.be/AHvd7-Q_dCk) on this. May help you'll.

nilvanis, 2022-04-01:
Okay, I managed to find out how to set the limits.
You must add: `tz.tuya_data_point_test ` to: `toZigbee` section under `model: 'TS0601_cover` in
`zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/tuya.js` file to be able to use this feature.
Reload Z2M and publish in MQTT:
`zigbee2mqtt/{device_friendly_name}/set/tuya_data_point_test` with data:
`1,103,1` - to set UP limit
`1,105,1` - to set up DOWN limit

homeseer666, 2022-04-01:
Great work. would be nice to submit a pull request so it gets added to the standard release. cheers,

Gimlir, 2022-04-01:
> Okay, I managed to find out how to set the limits. You must add: `tz.tuya_data_point_test ` to: `toZigbee` section under `model: 'TS0601_cover` in [...]

Thanks a lot! Could you also sniff setting of middle position and reversing of motor direction via app please? Second one is even more important as middle position is available via RF remote, but reverse is only in Tuya app according to seller comment on AliExpress where I get motors.

homeseer666, 2022-04-01:
Updated tuya.js and just tried this with the ZM25TQ 
doesn't seem to work;  the payload is alphanumerical right?
does this look right?

**zigbee2mqtt/{device_friendly_name}/set**
with payload: **{ "options":{ "tuya_data_point_test": "1,103,1"} }**

and must we go through the full reset + some specific sequence (like below) or can the up/down limits be re-set at any moment?

[image]

Gimlir, 2022-04-01:
UPD: Just seen video posted above by @EddieDSuza, seems seller know less about his product and he provided wrong info regarding possibility to reverse motor via RF remote as Eddie just demonstrated hot to do it. Thank you Eddie!

homeseer666, 2022-04-01:
> Thanks a lot! Could you also sniff setting of middle position and reversing of motor direction via app please? Second one is even more important as middle position is available via RF remote, but [...]

Reverse is already available 

https://www.zigbee2mqtt.io/devices/TS0601_cover.html

zigbee2mqtt/friendlyname/set
Payload : { "options":{ "reverse_direction": true} }

EddieDSuza, 2022-04-01:
Your welcome. Cheers

nilvanis, 2022-04-02:
> Great work. would be nice to submit a pull request so it gets added to the standard release. cheers,

I'll try to understand how to add the functionality to z2m and prepare a pull request.

> Thanks a lot! Could you also sniff setting of middle position and reversing of motor direction via app please? Second one is even more important as middle position is available via RF remote, but [...]

Middle is DP 104 :)
See here:
https://github.com/Koenkk/zigbee2mqtt/discussions/12027

> Updated tuya.js and just tried this with the ZM25TQ doesn't seem to work; the payload is alphanumerical right? does this look right?
>
> **zigbee2mqtt/{device_friendly_name}/set** with payload: **{ [...]

My bad!
The data type in MQTT message should be a name rather than an ID :)
Therefore:
`bool,103,1` to set up an upper limit
`bool,105,1` to set up the down limit
`bool,104,1` to set up the middle limit

I saw, that resetting the limits are going thru one zigbee command message, but you can try this:
`bool,103,0` to remove up an upper limit
`bool,105,0` to remove up the down limit
`bool,104,0` to remove up the middle limit

homeseer666, 2022-04-02:
This payload doesn't seem to work and nothing is returned in the console/log:

{ "options":{ "tuya_data_point_test": "bool,103,1"} }

This one 

{ "options":{ "tuya_data_point_test": bool,103,1} }

Returns:

Zigbee2MQTT:error 2022-04-02 12:33:46: Invalid message 'null', skipping...

homeseer666, 2022-04-02:
just in case, here's what my tuya.js looks like:

model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Yushun', model: 'YS-MT750'},
            {vendor: 'Zemismart', model: 'ZM79E-DT'},
            {vendor: 'Binthen', model: 'BCM100D'},
            {vendor: 'Binthen', model: 'CV01A'},
            {vendor: 'Zemismart', model: 'M515EGB'},
            {vendor: 'TuYa', model: 'M515EGZT'},
            {vendor: 'TuYa', model: 'DT82LEMA-1.2N'},
            {vendor: 'Moes', model: 'AM43-0.45/40-ES-EB'},
            {vendor: 'Larkkey', model: 'ZSTY-SM-1SRZG-EU'},
            {vendor: 'Zemismart', model: 'ZM25TQ', description: 'Tubular motor'},
            {vendor: 'Zemismart', model: 'AM43', description: 'Roller blind motor'},
            {vendor: 'Zemismart', model: 'M2805EGBZTN', description: 'Tubular motor'},
            {vendor: 'Zemismart', model: 'BCM500DS-TYZ', description: 'Curtain motor'},
            {vendor: 'A-OK', model: 'AM25', description: 'Tubular motor'},
        ],
        fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options,**tz.tuya_data_point_test**],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')
                .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0)
                    .withValueMax(255)
                    .withDescription('Motor speed'))],
    },

nilvanis, 2022-04-02:
> This payload doesn't seem to work and nothing is returned in the console/log:
> 
> { "options":{ "tuya_data_point_test": "bool,103,1"} }
> 
> This one
> 
> { "options":{ "tuya_data_point_test": bool,103,1} }
> 
> Returns:
> 
> Zigbee2MQTT:error 2022-04-02 12:33:46: Invalid message 'null', skipping...

Just to be sure we are on the same page:
Considering your motor is named "Cover 1" in Z2M and your base_topic is configred as default zigbee2mqtt, to set an upper limit you must publish to:
`zigbee2mqtt/Cover 1/set/tuya_data_point_test` the following message:
`bool,103,1`
nothing less, nothing more. Just the above 10 characters.

If you do that and the payload reaches z2m, you should get a debug log in Z2M saying something like "Received MQQT message xxxxx" where you'll see the topic under which you just posted a message.

Are you doing like the above?

homeseer666, 2022-04-02:
Thanks. was using a different topic. Not home now but will test tomorrow and report back.

Gimlir, 2022-04-04:
Sorry for kind of off-topic but still this is for testing of proposed solution first of all, so let me ask: running Z2M Edge as container installed by HA supervisor add-on manager. Any ideas how to edit code in node_modules/zigbee-herdsman-converters/devices/tuya.js inside container? Tried to find source of container in HA home directory in order to edit and rebuild image but failed to find.

Gimlir, 2022-04-04:
Finally managed to add this 'tuya_data_point_test' using external_converters config parameter of Dockered Z2M.

add name of custom converter file to your `configuration.yaml`
`external_converters:
  - tuya_custom.js
`
then create file with this name in same folder as `configuration.yaml` of Z2M, for Z2M installed as addon in HA by default it will be `homeassistant/zigbee2mqtt` if `homeassistant/` is your home config HA dir.

`tuya_custom.js` content 
cloned from default TS0601 _TZE200_fzo2pocs definition with 2 changes only

- added tz.tuya_data_point_test capability as per @nilvanis 
- removed motor_speed capability as not working for this motor model

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const libColor = require('zigbee-herdsman-converters/lib/color');
const utils = require('zigbee-herdsman-converters/lib/utils');

module.exports = [
    {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
        ],
        fingerprint: [
            // Tubular motors:
            {modelID: 'TS0601', manufacturerName: '_TZE200_fzo2pocs'},
        ],
        model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'CUSTOM Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZM25TQ', description: 'Tubular motor'},
        ],
        fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options,tz.tuya_data_point_test],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
        ],
    },

];
`

Gimlir, 2022-04-04:
Tested on two ZM25TQ, seems working fine, I can confirm it as working method allow to replace RF remote finally with these last remaining config actions

For example for up limit, same applies for down and middle limits setting as well:
- If limit is set via RF before, you should first remove existing limit by publishing `bool,103,0` 
- motor feedback for command is one short up\down motion same like at pairing
- middle limit can be set after up AND down only

@Koenkk looks like this topic is ready for implementation, could you please have a look

Koenkk, 2022-04-04:
Could you make a pr by clicking the pencil icon here? https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js

homeseer666, 2022-04-14:
> toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options,**tz.tuya_data_point_test**],

Was missing a space before " tz.tuya_data_point_test"

toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options, **tz.tuya_data_point_test**],

works now, thank you!

Gimlir, 2022-04-15:
> Could you make a pr by clicking the pencil icon here? https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js

Done,  created PR "ZM25TQ add data_point_test and remove speed ctrl" #4127

Gimlir, 2022-04-18:
Hi @nilvanis as per @Koenkk suggestion there  https://github.com/Koenkk/zigbee-herdsman-converters/pull/4127#issuecomment-1100158485 could you please try to change speed for this motor via native Tuya? Is it working at all?
In PR i've submitted based on your findings on limits I also added removal of speed control for Zemismart ZM25TQ as seems it is not supported - but could you please double check?

nilvanis, 2022-05-01:
> Hi @nilvanis as per @Koenkk suggestion there [Koenkk/zigbee-herdsman-converters#4127 (comment)](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4127#issuecomment-1100158485) could you please try to change speed for this motor via native Tuya? Is it working at all?

I don't see any option in tuya app to change the motor speed for my device. The only options available are:
1. Set a limit: up/down/middle
2. Motor direction: forward/backward
3. Motor mode: inching/linkage (I have no clue what this is about)

>  In PR i've submitted based on your findings on limits I also added removal of speed control for Zemismart ZM25TQ as seems it is not supported - but could you please double check?

First of all, adding tuya DP test is a workaround for now and I understand this should not be implemented as a final solution.
And yes, it looks like the motor speed setting is not available for this device.

What now must be done is to understand what data is being sent back when setting a limit and process it if possible.
Also, I'd like to understand how to code the limit set functionality (and reading the current status of the limit: set/not set). @Koenkk could you point me to some example I can base on?

Foroxon, 2022-05-06:
Looks like this issue was fixed with this PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4127
But how can we set up/down/middle limit with zigbe2mqtt?
Can we add some additional documentation to https://www.zigbee2mqtt.io/devices/ZM25TQ.html ?
Thanks!

Gimlir, 2022-05-11:
> Looks like this issue was fixed with this PR: [Koenkk/zigbee-herdsman-converters#4127](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4127) But how can we set up/down/middle limit with zigbe2mqtt? Can we add some additional documentation to https://www.zigbee2mqtt.io/devices/ZM25TQ.html ? Thanks!

Until we will have limit setting as native functionality use this way of setting limits via DP test (and yes, this is still workaround but allow to set it without RF remote anyway): https://github.com/Koenkk/zigbee2mqtt/issues/11164#issuecomment-1086176970 and https://github.com/Koenkk/zigbee2mqtt/issues/11164#issuecomment-1086580037

closed 2022-06-18 by github-actions[bot]

Albi83, 2023-02-09:
Will the same procedure work for the M515EGZT?

Log:
Received MQTT message on 'zigbee2mqtt/0xb4e3f9fffef4fbff/set/tuya_data_point_test' with data 'bool,105,0'
Publishing 'set' 'tuya_data_point_test' to '0xb4e3f9fffef4fbff'

mihail4anov, 2023-07-10:
**New**

const exposes = require('zigbee-herdsman-converters/lib/exposes');
**const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};**
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const libColor = require('zigbee-herdsman-converters/lib/color');
const utils = require('zigbee-herdsman-converters/lib/utils');

module.exports = [
    {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
        ],
        fingerprint: [
            // Tubular motors:
            {modelID: 'TS0601', manufacturerName: '_TZE200_fzo2pocs'},
        ],
        model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'CUSTOM Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZM25TQ', description: 'Tubular motor'},
        ],
        **fromZigbee: [fz.legacy.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.legacy.tuya_cover_control, tz.legacy.tuya_cover_options, tz.legacy.tuya_data_point_test],**
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
        ],
    },

];

homeseer666, 2023-07-10:
> **New**
>
> const exposes = require('zigbee-herdsman-converters/lib/exposes'); **const legacy = require('zigbee-herdsman-converters/lib/legacy'); const fz = [...]

Can you provide further details on what these changes do? Thx

mihail4anov, 2023-07-10:
> 
> Can you provide further details on what these changes do? Thx

The same that the previous coverter did, but now the startup error is gone

polsup2, 2023-07-15:
@mihail4anov , please give me advice. How to change provided custom converter code to override incorrect position behavior? Now I have position issue. Sending position >=95 causes the blind to close. Only sending topic "open" will actually open the blind. I would like to override this behavior in custom converter. Is it possible?

Roller blind: Zemismart ZM25TQ
Motor position limits: setup completed using remote. Working correctly using remote and "open" "close" topics. 
Z2m and zigbee stick: latest versions

homeseer666, 2023-07-15:
I have the ZM25TQ, everything works with the stock installtion except limits.

to set the limits You must modify the file
zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/tuya.js 
(“toZigbee” section under model “TS0601_cover”)

_toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options, tz.tuya_data_point_test],_

 (The space before " tz.tuya_data_point_test" is important)

Reload Z2M and publish in MQTT: zigbee2mqtt/{device_friendly_name}/set/tuya_data_point_test with data: 

bool,103,1 to set up an upper limit
bool,105,1 to set up the down limit
bool,104,1 to set up the middle limit
bool,103,0 to remove up an upper limit
bool,105,0 to remove up the down limit
bool,104,0 to remove up the middle limit

Considering your motor is named "Cover 1" in Z2M and your base_topic is configured as default zigbee2mqtt, to set an upper limit you must publish to:

zigbee2mqtt/Cover 1/set/tuya_data_point_test the following message:

bool,103,1

No hyphens needed

ndbroadbent, 2023-08-28:
These MQTT commands are working great, thanks! If anyone else uses Home Assistant, I wrote a script with fields to make this easier:

```
alias: Set Zemismart ZM25 Blind Motor Position
fields:
  entity_id:
    required: true
    description: Entity ID of your cover entity
    example: cover.blinds
    selector:
      entity: null
  limit:
    required: true
    description: "Type of limit: upper, lower, middle"
    default: upper
    selector:
      select:
        options:
          - upper
          - middle
          - lower
  action:
    required: true
    description: Set or unset limit
    default: set
    selector:
      select:
        options:
          - set
          - unset
mode: single
icon: mdi:blinds
sequence:
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: >-
        zigbee2mqtt/{{ states[entity_id].attributes.friendly_name
        }}/set/tuya_data_point_test
      payload_template: >-
        bool,{{"103" if limit == "upper" else ("105" if limit == "lower" else
        "104")}},{{"1" if action == "set" else "0"}}     
```

You can open up the Developer Tools and use the form fields to set/unset the upper/middle/lower limits for a given entity ID.

[image]

Now I'm very curious to know if there are other undocumented commands I can set. I would like to be able to set the speed to make them a bit faster if possible.

iamkaransharma, 2023-10-04:
Unfortunately, the `tuya_data_point_test` payloads in the thread don't seem to work for the ZM25RZ01 model. I added an external converter and got Z2M to not throw an error when publishing the payload, but the device doesn't adhere to the limits.

Evkool, 2024-01-28:
Does anyone know? What command should I use to start adding an rf remote control?

Albi83, 2024-11-07:
> These MQTT commands are working great, thanks! If anyone else uses Home Assistant, I wrote a script with fields to make this easier:
> 
> ```
> alias: Update Zemismart ZM25 Blind Motor Limit
> fields:
>   entity_id:
>     required: true
>     description: Entity ID of your cover entity
>     example: cover.blinds
>     selector:
>       entity:
>         filter:
>           domain: cover      
>   limit:
>     required: true
>     description: "Type of limit: upper, lower, middle"
>     default: upper
>     selector:
>       select:
>         options:
>           - upper
>           - middle
>           - lower
>   action:
>     required: true
>     description: Set or unset limit
>     default: set
>     selector:
>       select:
>         options:
>           - set
>           - unset
> mode: single
> icon: mdi:blinds
> sequence:
>   - service: mqtt.publish
>     data:
>       qos: 0
>       retain: false
>       topic: >-
>         zigbee2mqtt/{{ states[entity_id].attributes.friendly_name
>         }}/set/tuya_data_point_test
>       payload_template: >-
>         bool,{{"103" if limit == "upper" else ("105" if limit == "lower" else
>         "104")}},{{"1" if action == "set" else "0"}}
> ```
> 
> ```

It's the only script I have, but after saving it and restarting, I get an error saying the script can't be run:

The script "fields" (script.fields) is not active because the configuration has errors.

Error:extra keys not allowed @ data['action']. Got {'required': True, 'description': 'Set or unset limit', 'default': 'set', 'selector': {'select': {'options': ['set', 'unset']}}} extra keys not allowed @ data['entity_id']. Got {'required': True, 'description': 'Entity ID of your cover entity', 'example': 'cover.blinds', 'selector': {'entity': {'filter': {'domain': 'cover'}}}} extra keys not allowed @ data['limit']. Got {'required': True, 'description': 'Type of limit: upper, lower, middle', 'default': 'upper', 'selector': {'select': {'options': ['upper', 'middle', 'lower']}}} required key not provided @ data['sequence']. Got None.

To fix this error, edit the script to correct it, then save and reload the script configuration.
[image]

Montreal666, 2024-11-07:
Are you just trying to run the script? if so:

" open up the Developer Tools and use the form fields to set/unset the upper/middle/lower limits for a given entity ID."

Albi83, 2024-11-07:
Unfortunately, I can't even add the script, so there's no question of starting/running it.

Montreal666, 2024-11-09:
Paste your code

Montreal666, 2024-11-13:
Pleas format your code (check options above message - three dots)

Albi83, 2024-11-13:
> Please format your code (check options above message - three dots)

Formatted in previous post

Albi83, 2024-11-15:
I managed to add and run the script, but unfortunately Z2M returns this error:

z2m: No converter available for 'tuya_data_point_test' ("bool,103,0")

dantirdea, 2025-02-08:
In case anyone still having issues setting the limit without remote control, this worked for me straight away:

1) FULLY CLOSE BLINDS ( SET UPPER LIMIT )
2) Publish the following message from MQTT

zigbee2mqtt/<entity_id>/set/tuya_data_point_test
bool,103,1

3) FULLY OPEN BLINDS ( SET LOWER LMIT )
4) Publish the following message from MQTT

zigbee2mqtt/<entity_id>/set/tuya_data_point_test
bool,105,1

5) OPEN BLINDS HALF WAY ( SET MIDDLE POSITION )
6) Publish the following message from MQTT

zigbee2mqtt/<entity_id>/set/tuya_data_point_test
bool,104,1

Hope it helps!