Koenkk/zigbee2mqtt #11169: Orvibo RL804QZB state is reset on "get"

brunoleon, 2022-02-01:
### What happened?

I installed and paired an Orvibo RL804QZB to my network.
Pairing works flawlessly.
Device is present with its 3 endpoints (switches).

Failure scenario: 
- turn on `Endpoint: l1`  => it works but...
-  after a few seconds (30 like) the endpoint state is reset to off, even though the hardware itself is not actually toggled to off (no click is heard from the relay)

### What did you expect to happen?

Enpoint should stay on if set as such in Zigbee2mqtt

### How to reproduce it (minimal and precise)

Debugging showed that the toggling to "OFF" from z2m points of view happens when the device re-announces itself.
Tried both close and fare from coordinator => same behavior.

Tried direct state read:
`mosquitto_pub -t 'zigbee2mqtt/0x00124b00251f4790/get' -m '{"state_l1":""}'`

This does toggle the state from ON to OFF form z2m points of view, but not actually on the device.

TLDR: device state should not change (it is supposed to be a read) when calling `get' -m '{"state_l1":""}'``

### Zigbee2MQTT version

1.22.2-dev

### Adapter firmware version

20201026

### Adapter

Slaesh CC2632RB

### Debug log

zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-01 18:22:45: Received MQTT message on 'zigbee2mqtt/0x00124b00251f4790/l1/set' with data '{"state_l1":"ON"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-01 18:22:45: Publishing 'set' 'state' to '0x00124b00251f4790'
zigbee2mqtt      | 2022-02-01T17:22:45.652Z zigbee-herdsman:controller:endpoint Command 0x00124b00251f4790/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt      | 2022-02-01T17:22:45.652Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b00251f4790:49697/1 (0,0,1)
zigbee2mqtt      | 2022-02-01T17:22:45.652Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":49697,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":112,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,178,1]}}
zigbee2mqtt      | 2022-02-01T17:22:45.652Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,33,194,1,1,6,0,112,0,30,3,1,178,1,18]
zigbee2mqtt      | 2022-02-01T17:22:45.659Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt      | 2022-02-01T17:22:45.659Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt      | 2022-02-01T17:22:45.659Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt      | 2022-02-01T17:22:45.659Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt      | 2022-02-01T17:22:45.659Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | 2022-02-01T17:22:45.663Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,112,182]
zigbee2mqtt      | 2022-02-01T17:22:45.663Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,112,182]
zigbee2mqtt      | 2022-02-01T17:22:45.663Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,112] - 182
zigbee2mqtt      | 2022-02-01T17:22:45.663Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":112}
zigbee2mqtt      | 2022-02-01T17:22:45.663Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | 2022-02-01T17:22:45.688Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,6,0,33,194,1,1,0,120,0,217,246,4,0,0,5,24,178,11,1,0,33,194,29,49]
zigbee2mqtt      | 2022-02-01T17:22:45.688Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,6,0,33,194,1,1,0,120,0,217,246,4,0,0,5,24,178,11,1,0,33,194,29,49]
zigbee2mqtt      | 2022-02-01T17:22:45.688Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,33,194,1,1,0,120,0,217,246,4,0,0,5,24,178,11,1,0,33,194,29] - 49
zigbee2mqtt      | 2022-02-01T17:22:45.688Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":49697,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":120,"securityuse":0,"timestamp":325337,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,178,11,1,0]}}
zigbee2mqtt      | 2022-02-01T17:22:45.688Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":178,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":1,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":49697,"endpoint":1,"linkquality":120,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
zigbee2mqtt      | 2022-02-01T17:22:45.688Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | Zigbee2MQTT:info  2022-02-01 18:22:45: MQTT publish: topic 'zigbee2mqtt/0x00124b00251f4790', payload '{"linkquality":120,"state_l1":"ON","state_l2":"OFF","state_l3":"OFF"}'
zigbee2mqtt      | 2022-02-01T17:22:45.696Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,33,194,1,10,0,117,0,184,248,4,0,0,7,24,0,10,0,0,16,1,33,194,29,251]
zigbee2mqtt      | 2022-02-01T17:22:45.696Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,33,194,1,10,0,117,0,184,248,4,0,0,7,24,0,10,0,0,16,1,33,194,29,251]
zigbee2mqtt      | 2022-02-01T17:22:45.696Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,33,194,1,10,0,117,0,184,248,4,0,0,7,24,0,10,0,0,16,1,33,194,29] - 251
zigbee2mqtt      | 2022-02-01T17:22:45.696Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":49697,"srcendpoint":1,"dstendpoint":10,"wasbroadcast":0,"linkquality":117,"securityuse":0,"timestamp":325816,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,0,10,0,0,16,1]}}
zigbee2mqtt      | 2022-02-01T17:22:45.696Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":0,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":49697,"endpoint":1,"linkquality":117,"groupID":0,"wasBroadcast":false,"destinationEndpoint":10}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-01 18:22:45: Received Zigbee message from '0x00124b00251f4790', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt      | 2022-02-01T17:22:45.697Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... 13 lines cut ...]
zigbee2mqtt      | 2022-02-01T17:23:21.186Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b00251f4790:49697/1 (0,0,1)
zigbee2mqtt      | 2022-02-01T17:23:21.186Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":49697,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":114,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,1
80,0,0,0]}}
zigbee2mqtt      | 2022-02-01T17:23:21.186Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,33,194,1,1,6,0,114,0,30,5,16,180,0,0,0,2]
zigbee2mqtt      | 2022-02-01T17:23:21.193Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt      | 2022-02-01T17:23:21.193Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt      | 2022-02-01T17:23:21.193Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt      | 2022-02-01T17:23:21.193Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt      | 2022-02-01T17:23:21.193Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | 2022-02-01T17:23:21.198Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,114,180]
[... cut ...] 1}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-01 18:23:21: Received Zigbee message from '0x00124b00251f4790', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt      | 2022-02-01T17:23:21.224Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | Zigbee2MQTT:info  2022-02-01 18:23:21: MQTT publish: topic 'zigbee2mqtt/0x00124b00251f4790', payload '{"linkquality":117,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'

closed 2022-03-13 by github-actions[bot]

bed31, 2022-09-18:
Hi,
I have exactly the same issu with 2 Orvibo RL804QZB.
@brunoleon Do you have solve he problem ?

Eric

brunoleon, 2022-09-18:
Hello,

I discovered that only l1 and l2 have the problem.
l3 is behaving properly so I use it as I only need one for now.

Bruno