Koenkk/zigbee2mqtt #11171: Unable to add scene to group - No network route (205)

shrx, 2022-02-01:
### What happened?

Following these instructions to set up color temperature scenes for the Ikea Tradfri bulb, I performed these steps:
- created a group named "tradfri_scenes" with ID 65289
- added my bulb to the group
- tried to add a scene to the group by running `mqttcli pub -t zigbee2mqtt/tradfri_scenes/set -m '{"scene_add": {"ID": 0, "color_temp": 370}}'`
- The scene is not added and I get this in the log:
```
Zigbee2MQTT:error 2022-02-01 19:30:16: Publish 'set' 'scene_add' to 'tradfri_scenes' failed: 'Error: Command 65289 genScenes.remove({"groupid":65289,"sceneid":0}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:info  2022-02-01 19:30:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'scene_add' to 'tradfri_scenes' failed: 'Error: Command 65289 genScenes.remove({\"groupid\":65289,\"sceneid\":0}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"tradfri_scenes"},"type":"zigbee_publish_error"}'
```

### What did you expect to happen?

I expect the scene to be added to the group

### Zigbee2MQTT version

1.22.2-dev commit: 59ef38f0

### Adapter firmware version

20180507

### Adapter

Sonoff CC2531

shrx, 2022-02-01:
I tried the procedure descibed here https://github.com/Koenkk/zigbee2mqtt/issues/4135#issuecomment-682092658 but I get the same error 205 even with the empty group withour adding a bulb to it.

shrx, 2022-02-01:
Same issue also after updating coordinator firmware to 20211115.

Littlesloths, 2022-07-04:
the "no network route" error is still active.... I have three Ikea tradfri remotes E1810. They worked before in that regard. Now I changed to ESP32 Zigbee Bridge Pro and 2/3 are not working.
There are discussions on
https://github.com/Koenkk/zigbee2mqtt/issues/7716
others think it is related to the IKEA firmware on the remote control.
Any new ideas?

kapabel, 2022-07-20:
I have the same problem. Also using the CC2531 and having troubles with IKEA products (blinds E1757, remote E1766 and signal repeater E1746)

Littlesloths, 2022-07-23:
did you update the individual FW versions? what versions do you use? Finally, I lost interest in IKEA remotes and use other remotes (not that stylish though)

kapabel, 2022-07-23:
I ended up formatting my whole setup and starting clean from scratch with an installation via docker. Problem solved and no need to upgrade the firmware of the dongle.

shrx, 2023-03-02:
Is it possible to reopen the issue? It's still present with zigbee2mqtt 1.30.1.