Koenkk/zigbee2mqtt #11172: TPZRCO2HT-Z3 not reporting battery

4nd3r, 2022-02-01:
### What happened?

I have three TPZRCO2HT-Z3 devices (firmware build date: 20190501).

According to [documentation](https://www.zigbee2mqtt.io/devices/TPZRCO2HT-Z3.html) this device should expose battery, but value for all three devices is `N/A`.

State has following:
```
{
    "co2": 668,
    "humidity": 24.5,
    "linkquality": 171,
    "temperature": 23.52
}
```

I also wrote to Titan Products to clarify if battery value is actually sent by devices, but maybe someone here has some knowledge about this device?

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

zzh!

4nd3r, 2022-02-08:
Response from Titan Products:
> Battery voltage can be obtained in the following way:
> First Bind the Power Configuration Cluster (0x0001) in End Point 1 2. Then read the Battery Voltage attribute (0x0020) in the Power configuration cluster (0x0001) of End Point 1. The ZED/end device will then return the battery level as a means of a voltage and not as a percentage.

mog422, 2022-02-08:
After I submitted this PR, I have modified the device to use DC power instead of battery, so I have no way to check it.

mog422, 2022-02-08:
I tried some more.

Try it as a branch of this PR.
```zigbee2mqtt/bridge/request/device/configure``` may be required if the device is already paired.
Koenkk/zigbee-herdsman-converters#3845

```{"battery_low":false,"co2":745,"humidity":45.02,"linkquality":112,"temperature":26.06,"voltage":3600}```
In my environment, it seems to report expected value.

4nd3r, 2022-02-08:
With https://github.com/Koenkk/zigbee-herdsman-converters/pull/3845 I can see voltage of all 3 devices. Thank you, @mog422.