Koenkk/zigbee2mqtt #11173: After updating CC2652R_coordinator_20211217 board not available in ioBroker

hytrek, 2022-02-01:
### What happened?

Hi there
i just updated the board to the latest Firmware using the CC2652R_coordinator_20211217.hex 
While installing I was prompted to also updated the Firmware of the XDS110 from xxx13 to 15. I accepted.
Board was not found by ioBroker so I checked the address and  ls -l /dev/serial/by-id and found it was changed from 13-15

Back in ioBroker I changed it accordingly. Board now was accessable but all connected devices were not available. (visible in the Adapter but not connected)

Can you help please?

### Zigbee2MQTT version

1.6.6

### Adapter firmware version

20211217

### Adapter

CC26X2R1

Koenkk, 2022-02-01:
This is not the iobroker issue tracker, please create it there.

closed 2022-02-01 by Koenkk

hytrek, 2022-02-01:
Sorry... where to find the right place?

Koenkk, 2022-02-02:
https://github.com/ioBroker/ioBroker.zigbee/issues