Koenkk/zigbee2mqtt #11177: Develco Energy Monitor cannot set pulse configuration

danmed, 2022-02-01:
### What happened?

I was messing with the slider for pulse_configuration and (as expected) it wildy altered my readings.. so i wanted to set it back 0.. but it seems i cannot set it any lower than 50.

 Publish 'set' 'pulse_configuration' to 'Frient Energy Monitor' failed: 'Error: Write 0x0015bc001b022574/2 seMetering({"develcoPulseConfiguration":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'

### What did you expect to happen?

I expected to be able to set the slider back to 0 or NULL

### How to reproduce it (minimal and precise)

move the slider and then try to set it back to 0

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

 20210120

### Adapter

Tubeszb ethernet coordinator

### Debug log

[frient.txt](https://github.com/Koenkk/zigbee2mqtt/files/7982284/frient.txt)

GSzabados, 2022-05-28:
The default pulse configuration is 1000. The value represents how many pulses mean 1kWh of energy.

0 is not really an option, but I guess you figured this out already.

danmed, 2022-05-28:
> The default pulse configuration is 1000. The value represents how many pulses mean 1kWh of energy.
> 
> 0 is not really an option, but I guess you figured this out already.

Yeah.. i figured it out.. When the device is first paired, it is set to 0 in the GUI which i *think* means it's jsut at default (1000).. setting it at 2000 works fine.