Koenkk/zigbee2mqtt #11178: Can't set minimum reporting setting on Nous E6

stomko11, 2022-02-01:
### What happened?

I am unable to set minimum reporting interval. Device is quite chatty, so I would like to set minimal reporting interval to 600 seconds instead of default 60.

When I select endpoint 1 and "ManuSpecificTuya" cluster, it does not offer any attributes. If I change interval (60 to 600) I get following errors:

[image]

```
Error Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0xa4c138541c4d569d/1 manuSpecificTuya([{"attribute":"","minimumReportInterval":"600","maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Don't know value type for 'undefined')'
Debug Error: ConfigureReporting 0xa4c138541c4d569d/1 manuSpecificTuya([{"attribute":"","minimumReportInterval":"600","maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Don't know value type for 'undefined') at Object.IsDataTypeAnalogOrDiscrete (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:30:15) at /app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:354:38 at Array.find (<anonymous>) at Function.conditionsValid (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:338:58) at ZclFrame.writePayloadGlobal (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:113:35) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:76:18) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:579:28) at /app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:642:39 at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:641:39)
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"ConfigureReporting 0xa4c138541c4d569d/1 manuSpecificTuya([{\"attribute\":\"\",\"minimumReportInterval\":\"600\",\"maximumReportInterval\":3600,\"reportableChange\":0}], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Don't know value type for 'undefined')","status":"error","transaction":"rzlzl-1"}'
```

### What did you expect to happen?

I expect the value to be changed. I did wake up device prior to setting it.

### How to reproduce it (minimal and precise)

Set mininal reporting interval to 600, wake up device, hit Apply

### Zigbee2MQTT version

1.22.2-dev commit: 2721cc1

### Adapter firmware version

0x26700700

### Adapter

 ConBee2/RaspBee2

### Debug log

```
Error Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0xa4c138541c4d569d/1 manuSpecificTuya([{"attribute":"","minimumReportInterval":"600","maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Don't know value type for 'undefined')'
Debug Error: ConfigureReporting 0xa4c138541c4d569d/1 manuSpecificTuya([{"attribute":"","minimumReportInterval":"600","maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Don't know value type for 'undefined') at Object.IsDataTypeAnalogOrDiscrete (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:30:15) at /app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:354:38 at Array.find (<anonymous>) at Function.conditionsValid (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:338:58) at ZclFrame.writePayloadGlobal (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:113:35) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:76:18) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:579:28) at /app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:642:39 at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:641:39)
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"ConfigureReporting 0xa4c138541c4d569d/1 manuSpecificTuya([{\"attribute\":\"\",\"minimumReportInterval\":\"600\",\"maximumReportInterval\":3600,\"reportableChange\":0}], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Don't know value type for 'undefined')","status":"error","transaction":"rzlzl-1"}'
```

stomko11, 2022-03-04:
Anyone able to help on this?

closed 2022-04-11 by github-actions[bot]

samskiter, 2023-10-24:
I'm also having this issue for another device - the Eurotronic Comet TRV (which is has an identical Zigbee interface to the Spirit model)

It seems that I am unable to adjust or delete reporting that is configured for a 'custom' attribute that isn't previously known to zigbee2mqtt.

I understand not being able to set/update a reporting but preventing removing a reporting is a pain.

This issue is repeatable in the zigbee2mqtt frontend *and* directly via the configure_reporting topic

midlan, 2024-02-25:
@stomko11 I have same issue. In my case the apply button is just disabled, so I can't do anything. Did you find any workaround?