Koenkk/zigbee2mqtt #11179: Tuya TS0121 'wall switch' injecting error(s) in z2m logs regularly

sermayoral, 2022-02-02:
### What happened?

Hi, using latest 1.23.0, I'm getting regular error in the logs:

`Zigbee2MQTT:error 2022-02-02 01:38:19: Failed to configure 'bw_shp13_lavavajillas_enchufe', attempt 3 (Error: ConfigureReporting 0x842e14fffe51c642/1 seMetering([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 39642 - 1 - 16 - 1794 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))`

The switch works fine most of the time. Sometimes I need to click a 2nd time, as if the 1st action didn't register.
Only bound to the coordinator, cluster PowerCfg.

Thanks!!

### What did you expect to happen?

No error in the log

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

zzh!

### Debug log

Zigbee2MQTT:error 2022-02-02 01:38:19: Failed to configure 'bw_shp13_lavavajillas_enchufe', attempt 3 (Error: ConfigureReporting 0x842e14fffe51c642/1 seMetering([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 39642 - 1 - 16 - 1794 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))

Evil95, 2022-02-02:
My BlitzWolf BW-SHP13 works without any issues and reports it's power usage every 15 seconds, which is great. The Vendor of my Plugs are _TZ3000_g5xawfcq.

Koenkk, 2022-02-02:
Do the power measurements work?

sermayoral, 2022-02-02:
> Do the power measurements work?

Hi @Koenkk 

Yes, it does. Everything seems to work right, but everytime I have to restart zigbee2mqtt, the logs throws these errors. I have tried to repair the device several times with no success. 

I have other BW-SHP13 that i bought some months before this one that works perfectly and it doesn't throw any error. database.db is:

`
{"id":2,"type":"Router","ieeeAddr":"0x842e14fffe36663a","nwkAddr":47888,"manufId":4098,"manufName":"_TZ3000_g5xawfcq","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TZ3000_g5xawfcq","stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,2963],"divisor":100,"multiplier":1}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsVoltage":228,"rmsCurrent":0,"activePower":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17ef09","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17ef09","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001e17ef09","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":545190599},"lastSeen":1643817659583,"defaultSendRequestWhen":"immediate"}
`

The vendor of this plug is the same as @Evil95 one. It seems there is something that z2m doesn't interpret right only with the vendor _TZ3000_3ooaz3ng

Koenkk, 2022-02-02:
@sermayoral should be fixed now, just to double check, does it also report an `energy` value?

sermayoral, 2022-02-03:
@Koenkk yes, it does:

`MQTT publish: topic 'zigbee2mqtt/bw_shp13_lavavajillas_enchufe', payload '{"current":0.05,"energy":30.82,"last_seen":"2022-02-03T01:29:40+01:00","linkquality":65,"power":1,"power_outage_memory":null,"state":"ON","voltage":219}'`

Koenkk, 2022-02-03:
good, can you confirm this error has been fixed in the latest-dev? Then  this can be closed. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sermayoral, 2022-02-03:
```
info  2022-02-03 20:53:13: Starting Zigbee2MQTT version 1.23.0-dev (commit #9fb1eb4)
info  2022-02-03 20:53:13: Starting zigbee-herdsman (0.14.11)
info  2022-02-03 20:53:14: zigbee-herdsman started (resumed)
info  2022-02-03 20:53:14: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}'
info  2022-02-03 20:53:15: Configuring 'bw_shp13_lavavajillas_enchufe'
info  2022-02-03 20:53:38: Successfully configured 'bw_shp13_lavavajillas_enchufe'
```

Thank you very much @Koenkk :-)

closed 2022-02-03 by sermayoral