Koenkk/zigbee2mqtt #11180: Aqara vibration sensor: Sensitivity cannot be set

kremik, 2022-02-02:
### What happened?

If you try to set the sensitivity of the sensor (low, medium, high) it's not possible and it reports an error (see debug log below). It is Xiaomi/Aqara DJT11LM Vibration sensor. I don't know if it's a problem of the device or some problem in Z2M. Just to let you know.

### What did you expect to happen?

Set the sensitivity to a higher or lower level and not to get an error.

### How to reproduce it (minimal and precise)

Change the sensitivity e.g. in the Z2M web interface.

### Zigbee2MQTT version

 1.23.0 commit: 6ea61bae

### Adapter firmware version

20210211

### Adapter

CC2652P

### Debug log

Publish 'set' 'sensitivity' to 'aqaravibration1' failed: 'Error: Write 0x00158d0006d33d4a/1 genBasic({"65293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'

Koenkk, 2022-02-02:
Make sure to wakeup the sensor right before sending the command by triggering it or pressing a button on it.

Koenkk, 2022-02-02:
@kremik something that might help is to permit joining, factory reset the device while holding it against the coordinator and set the sensitivity

atv2016, 2022-02-03:
i've noticed the same, it would also be nice if the gui would show that the sensitivity is set (maybe i don't see it because it shows an error) but i can't set it either. Maybe i'll have to pair it first with the aqara app or something.

atv2016, 2022-02-03:
i seem to have set it to high now (stays black) but even then sensitivity doesn't seem to be very high. not sure what consists of a vibration but i have to put in a lot of effort to make it go to true..

closed 2022-03-14 by github-actions[bot]