Koenkk/zigbee2mqtt #11181: Troubleshooting a ZigBee network

swiergot, 2022-02-02:
### What happened?

Hi,

I'm experiencing constant stability issues with my network. It was all fine when I had only one switch and two buttons (at that time the coordinator was CC2531, I also had an IKEA router). Then I started adding more switches and buttons, and it was no longer so great. I thought the coordinator was to blame, so first I replaced it with CC2538, and when that didn't help, I bought SONOFF Zigbee 3.0 USB Dongle Plus, but sadly nothing has improved.

How my network looks like today:

I have a three story house. The coordinator (SONOFF) is at the top floor where I also have one switch. On the middle floor I have the IKEA router and a switch, a Tuya button, a SONOFF button (those two buttons work without issues as far as I can tell, also the switch seems to be always responsive). Problems are observed on the ground floor: three Tuya buttons and three switches. Either button presses do not reach Z2M (no events appear in the log) or the switches do not respond at all or their response is much delayed (happens even when I try to control them from HA). It's all very disappointing and unacceptable. I sincerely hope this is not the way ZigBee normally works.

Now, the question is: how to troubleshoot it? I should also mention that initially I wanted to get rid of the IKEA router but I was unable to pair the devices on the ground floor when the coordinator was at the top floor, so I joined the IKEA router on the middle floor again and then all devices on the ground floor paired without any problem.

Is there any debug logging on Z2M that will be sufficient to find the root cause? Or do I need a sniffer?

Any help will be greatly appreciated.

Below is a detailed list of my devices and a map:

Name | Type | Manufacturer | Model | LQI
-- | -- | -- | -- | --
Światło w sypialni | switch | Lonsonho | QS-Zigbee-S04-2C-LN | 91
Przycisk 1 | button | SONOFF | SNZB-01 | N/A
Światło w salonie | switch | Lonsonho | QS-Zigbee-S04-2C-LN | 87
Światło w wiatrołapie | switch | Lonsonho | QS-Zigbee-S04-2C-LN | 91
Światło w kuchni | switch | SONOFF | ZBMINI | 87
Pilot w salonie | button | TuYa | TS0043 | 87
Pilot w kuchni | button | TuYa | TS0043 | 91
Pilot w wiatrołapie | button | TuYa | TS0043 | 87
Router | router | IKEA | E1746 | 91
Czujnik Aqara 1 | sensor | Xiaomi | WSDCGQ11LM | 40
Pilot | button | TuYa | TS0043 | 87
Światło na tarasie górnym | switch | SONOFF | ZBMINI | 3
Okno dachowe | sensor | Xiaomi | MCCGQ11LM | 116

[image]

### What did you expect to happen?

A stable, always responsible network.

### How to reproduce it (minimal and precise)

Happens all the time on my network.

### Zigbee2MQTT version

multiple, regular updates as a HA addon

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

KrzysztofHajdamowicz, 2022-02-14:
Try to build network from scratch- start with coordinator, then pair devices sorting by their proximity to coordinator. In my opinion, main concern in your case is bridging network from building one level to another level. Best idea is to put a router on each staircase (or even half-level if staircase turn by 180 degree).

KrzysztofHajdamowicz, 2022-02-14:
From my experience IKEA routers are OK, you can buy CC2562P from Grzegorz Sterniczuk and flash it via USB with Koenkk's router firmware. They have great antennas and good radios. I use them to bridge two parts of a building where I have 5 walls in a 5m distance. Other than that there are 60 more devices on that stick with coordinator firmware.

swiergot, 2022-02-14:
After writing my previous post I grabbed the IKEA router I had on the 1st floor and took it downstairs to plug into a socket roughly in the middle of the ground floor (and close to the staircase). So far it's been working flawlessly.

To be honest I didn't even consider that it could be due to weak links. It didn't look that bad on the map to me, or maybe I just lack experience with ZigBee networks. I suspected buggy routers. I'm so glad I was wrong.

Thank you once again for help!

KrzysztofHajdamowicz, 2022-02-14:
I'm glad it helped. It's good for Zigbee network to have many possible routes to transmit data around, so think about expanding your network around that tough areas. I can guarantee you that Sonoff dongle with 3+ dBi antenna and router firmware will outperform IKEA router with a night-and-day difference :-)

swiergot, 2022-02-14:
Sure, as I wrote before Sonoff is my third coordinator and I wish I had some better coordinator hardware from the start, it would have saved me a lot of trouble.

IKEA routers are just easy do deploy and nicely compact due to integrated power supply.

closed 2022-02-21 by swiergot