Koenkk/zigbee2mqtt #11184: Container restarting with version 1.23

ptrdvds, 2022-02-02:
### What happened?

Image pull : image: koenkk/zigbee2mqtt:latest
After docker-compose up -d the container keeps restarting (due to restart: unless-stopped)
Rollback to 1.22.2 works without any problems.

### What did you expect to happen?

I expected the container to be up without errors and restarts

### How to reproduce it (minimal and precise)

Pull image 1.23 or latest and start the container.

### Zigbee2MQTT version

1.23

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Debug log

zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/bridge/groups' with data '[{"friendly_name":"default_bind_group","
id":901,"members":[],"scenes":[]}]'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/bridge/extensions' with data '[]'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"6ea61ba","coordinator":{"meta
":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"chann
el":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.23.0"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/sen05.outdoor_room/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/sen06.bedroom/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/plu01.living_room.oslo/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/plu02.hall.wall_light/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/but01' with data '{"action":null,"battery":60,"linkquality":null,"
update":{"state":"idle"},"update_available":false}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/but02.living_room.oslo' with data '{"action":null,"battery":87,"li
nkquality":null,"update":{"state":"idle"},"update_available":false}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/led01' with data '{"brightness":100,"color":{"h":199,"hue":199,"s"
:5,"saturation":5,"x":0.3176,"y":0.3276},"color_mode":"color_temp","color_temp":160,"color_temp_startup":160,"linkquality":null,"state":"OFF"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/sen07.laundry_room/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/sen08.bathroom_south/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/sen09.bathroom_north/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/sen10.tv_room/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/sen11.living_room/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/lam01.tv_room/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/lig01.tv_room/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/lig02.living_room.oslo/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/plu03.living_room.flos/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/but04/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/plu01.living_room.paraplu/availability' with data 'online'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/but03' with data '{"action":null,"battery":11,"linkquality":null,"
update":{"state":"idle"},"update_available":false}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Received MQTT message on 'zigbee2mqtt/but03/availability' with data 'offline'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-02-02 10:51:23: Active device 'plu01.living_room.paraplu' was last seen '3.49' minutes ago.
zigbee2mqtt      | Zigbee2MQTT:error 2022-02-02 10:51:24: Failed to call 'Availability' 'start' (TypeError: Cannot read property 'active' of undefined
zigbee2mqtt      |     at Availability.getTimeout (/app/lib/extension/availability.ts:21:50)
zigbee2mqtt      |     at Availability.isAvailable (/app/lib/extension/availability.ts:32:27)
zigbee2mqtt      |     at Availability.publishAvailability (/app/lib/extension/availability.ts:133:32)
zigbee2mqtt      |     at Availability.start (/app/lib/extension/availability.ts:109:22)
zigbee2mqtt      |     at Controller.callExtensions (/app/lib/controller.ts:315:40)
zigbee2mqtt      |     at processTicksAndRejections (internal/process/task_queues.js:95:5)
zigbee2mqtt      |     at Controller.start (/app/lib/controller.ts:152:9)
zigbee2mqtt      |     at start (/app/index.js:101:5))
zigbee2mqtt      | Zigbee2MQTT:info  2022-02-02 10:51:24: Started frontend on port 0.0.0.0:8080
zigbee2mqtt      | Zigbee2MQTT:info  2022-02-02 10:51:24: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"6ea61ba","coordinator":{"meta":{"ma
intrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"channel":11:

ptrdvds, 2022-02-02:
homeassistant: true
permit_join: false
frontend:
  port: 8080
  host: 0.0.0.0
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto:1883
keepalive: 60
serial:
  # port: /dev/ttyUSB0
  port: /dev/ttyACM0
  adapter: deconz
advanced:
  ikea_ota_use_test_url: true
  log_level: debug
devices:
  '0x0017880109d31152':
    friendly_name: plu01.living_room.paraplu
    availability: true
  '0x0017880108fb4f5c':
    friendly_name: plu02.hall.wall_light
    availability: true
  '0x00158d00079ae081':
    friendly_name: sen01.office
  '0xb4e3f9fffec8efa2':
    friendly_name: but01
    retain: true
  '0x2c1165fffe02833a':
    friendly_name: but02.living_room.oslo
    retain: true
    legacy: false
  '0x0c4314fffe49a529':
    friendly_name: but03
    retain: true
  '0xb4e3f9fffeb294f2':
    friendly_name: led01
    retain: true
  '0x001788010b0942d9':
    friendly_name: sen02.hall
  '0x00158d0006f23200':
    friendly_name: sen03.kitchen
  '0x00158d0006f84066':
    friendly_name: sen04.guest_room
  '0x00158d0006f22e6b':
    friendly_name: sen05.outdoor_room
  '0x00158d0006fbe88d':
    friendly_name: sen06.bedroom
  '0x00158d0006f85b8f':
    friendly_name: sen07.laundry_room
  '0x00158d0006f45ae6':
    friendly_name: sen08.bathroom_south
  '0x00158d0006f8420c':
    friendly_name: sen09.bathroom_north
  '0x00158d0006f231f9':
    friendly_name: sen10.tv_room
  '0x00158d0006f45ca0':
    friendly_name: sen11.living_room
  '0x001788010b0915d9':
    friendly_name: lig01.tv_room
  '0x001788010b2279dc':
    friendly_name: lig02.living_room.oslo
  '0x0017880109d66abf':
    friendly_name: plu03.living_room.flos
  '0x00158d000753dd91':
    friendly_name: but04

Koenkk, 2022-02-02:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ptrdvds, 2022-02-02:
Hi Koen,
That did the trick, the container stayed up.
Thank you very much.
Regards Peter

closed 2022-02-03 by Koenkk