Koenkk/zigbee2mqtt #11186: [New device support]: Philips Hue Gradient Signe floor lamp (Black)

Klangen82, 2022-02-02:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-gradient-signe-floor-lamp/8718696176283

### Database entry

 Device '0x001788010b9153db' joined Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010b9153db","ieee_address":"0x001788010b9153db"},"type":"device_joined"}' Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788010b9153db"},"type":"device_connected"}' Info Starting interview of '0x001788010b9153db' Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010b9153db","ieee_address":"0x001788010b9153db","status":"started"},"type":"device_interview"}' Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788010b9153db"},"type":"pairing"}' Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010b9153db","ieee_address":"0x001788010b9153db"},"type":"device_announce"}' Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x001788010b9153db"},"type":"device_announced"}' Info Successfully interviewed '0x001788010b9153db', device has successfully been paired Warning Device '0x001788010b9153db' with Zigbee model '915005987201' and manufacturer name 'Philips' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x001788010b9153db","ieee_address":"0x001788010b9153db","status":"successful","supported":false},"type":"device_interview"}' Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x001788010b9153db","supported":false},"type":"pairing"}'

### Comments

Believe that Philips started to come out with some gen 2 of the same devices as my Philips Hue Gradient Signe floor lamp (Black) reports model number 915005987201 and not 4080248U9

https://www.zigbee2mqtt.io/devices/4080248U9.html

Please let me know if more info is needed!

### Supported color modes

light (state, brightness, color_temp, color_temp_startup, color_xy, color_hs), effect, linkquality

### Color temperature range

 [hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]})]

Koenkk, 2022-02-02:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-02 by Koenkk

Klangen82, 2022-02-02:
Thanks @Koenkk fast you are = I like :) Keep up the great work!