Koenkk/zigbee2mqtt #11190: TS0601_TZE200_iossyxra lost cover entity in 1.23.0

VladNord, 2022-02-02:
### What happened?

After updating to latest z2m 1.23.0 TS0601_TZE200_iossyxra lost cover entity in latest update.
just select entities.
[image]

in previous version everything is ok

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26720700

### Adapter

conbee2

Koenkk, 2022-02-02:
This was done in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3677 by @Burki24. @Burki24 could make a pr to correct this?

Burki24, 2022-02-02:
> This was done in [Koenkk/zigbee-herdsman-converters#3677](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3677) by @Burki24. @Burki24 could make a pr to correct this?

Yes, I'm working on it ASAP.

Burki24, 2022-02-08:
@VladNord 

Work is done: 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3836#issuecomment-1031644151
[image]

VladNord, 2022-02-11:
> @VladNord
> 
> Work is done, please verify that local, so we can close this issue, thanks. [Koenkk/zigbee-herdsman-converters#3836 (comment)](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3836#issuecomment-1031644151) [image] [image]

Thanks!
Can you send manual how I can test this improvement before release, please?

Burki24, 2022-02-18:
@VladNord This update was merged 10 days ago. So it's allready included in mster-branch.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3836#issuecomment-1031644151

closed 2022-03-29 by github-actions[bot]