Koenkk/zigbee2mqtt #11191: Availability feature for only certain devices: HA entity not available

kauthmbt, 2022-02-02:
### What happened?

After updating to latest version the selected lights marked with `availabilty: true` became permanently unavailable in HA.
In the Zigbee2MQTT frontend itself the lights can still be turned on/off. But even if I turn it on via the Z2M GUI it still doesn't reflect the correct HA entity state.

A fallback to the previous version solves the problem again.

Is there a relation to #10437 ?

### What did you expect to happen?

A way to just mark individual devices `availabilty: true` and to reflect the current state of the device in HA 

### How to reproduce it (minimal and precise)

Adding `availability: true` in `devices.yaml` for just a specific device like described in [Availability advanced configuration](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)

As soon as the `availability: true` option is removed for the specific device in `devices.yaml` the HA entity becomes usable again.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Veldkornet, 2022-02-04:
For what it's worth, I also use availability with the same coordinator and I don't have this problem.

kauthmbt, 2022-02-04:
@Veldkornet - also just only for specific devices?
Or the general availability function which tracks availability of **all** devices? Because the later one works also for me but not the option of using it for just specific devices.

Veldkornet, 2022-02-04:
> @Veldkornet - also just only for specific devices?
> 
> Or the general availability function which tracks availability of **all** devices? Because the later one works also for me but not the option of using it for just specific devices.

Ah I see. I have it enabled "globally". I haven't set anything specific for each device.

closed 2022-03-14 by github-actions[bot]