Koenkk/zigbee2mqtt #11197: [New device support]: Aubess Tuya _TZE200_locansqn temperature and humidity sensor

Math1010, 2022-02-02:
### Link

https://fr.aliexpress.com/item/1005003652886345.html?gatewayAdapt=glo2fra&spm=a2g0o.9042311.0.0.4aa06c37a1GwjK

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0xa4c1380e3b04539a","nwkAddr":58784,"manufId":4417,"manufName":"_TZE200_locansqn","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":49,"65508":0,"appVersion":67,"modelId":"TS0601","manufacturerName":"_TZE200_locansqn","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0022811bd2","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1643827716736,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new device page but values displayed are N/A.
Here is the tuya dump result :
https://pastebin.com/EVdFe3Sk
And here the result of the script  read_tuya_dump.py : 
https://pastebin.com/e9BzTk4M

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_locansqn'
        },
    ],
    model: 'TS0601_temp_humidity_sensor',
    vendor: 'TuYa',
    description: 'Aubess Temperature & humidity sensor',
    fromZigbee: [
		fz.tuya_temperature_humidity_sensor,
    ],
    toZigbee: [
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
   
    exposes: [e.temperature(), e.humidity(), e.battery()],
};

module.exports = definition;
```

coph, 2022-03-08:
Hi,

It working but I'm getting some message:

Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #9 with data {"dp":9,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #10 with data {"dp":10,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,134]}}
Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #11 with data {"dp":11,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #12 with data {"dp":12,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,60]}}
Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #13 with data {"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,20]}}
Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #17 with data {"dp":17,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,120]}}
Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #18 with data {"dp":18,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,120]}}
Zigbee2MQTT:warn  2022-03-08 17:53:13: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #19 with data {"dp":19,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,6]}}
Zigbee2MQTT:warn  2022-03-08 17:53:14: zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,6]}}

coph, 2022-03-18:
Are you guys facing the issue and warning messages ?

MarkKoshelev, 2022-03-27:
The same
zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #18 with data {"dp":18,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,120]}}
Warning zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #19 with data {"dp":19,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,6]}}
Warning zigbee-herdsman-converters:maa_tuya_temp_sensor: NOT RECOGNIZED DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,6]}}

Koenkk, 2022-03-27:
To figure out what these data points mean, you can use https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

coph, 2022-03-27:
I've tried, get some error message documented by other users "Login error (2406): skill id invalid" trying to log to my Tuya project. Have followed instruction documented by other users... I still not working. Maybe Mark Koshelev will have more success than me ?

Carontes, 2022-04-01:
I've bought the same sensor but it doesnìt work, I see a lot of error messages, the only value Zigbee2mqtt shows is the linkquality. Is it possible to modify a working external converter file of another _TZE200_ model that we know it works in Zigbee2mqtt, or is it impossible? It's a shame to have it and not being able to use it, it looks a valid sensor

coph, 2022-04-01:
Hi Carontes, It's working if you build an external converter using the code above provided by Mark. I can get Temperature / Humidity / Battery level and Link Quality. My device is showing the Time - and it's synchronize properly with local time from my server (HA). I do have some error messages but it's working. Attached is a picture of my device.

[image]

coph, 2022-04-01:
Have you re-start the core after the changes made to the configuration ? If not sure, re-start them all.  Yes I see those devices in zigbee2mqtt device list. Sorry device name in French:

[image]
and:

[image]

Carontes, 2022-04-01:
Thanks for your reply. I meant that I can't add the sensor in the Overview section in HA, I don't know if you understand what I mean. I'm not an expert of HA, this is my first time using it since I wanted to create automations between TRVs, window sensors and temperature sensors

coph, 2022-04-01:
Carontes, It's not the tool to ask for help. I do not see option to send direct message on Github. I might help but would need an email address. Otherwise you can have a look on youtube and /or HA Documentation.

Carontes, 2022-04-03:
Sorry, you're right. This is my email arturogellini[at]gmail.com. Thank you for your help

IIPoliII, 2022-05-05:
For me it's not getting the battery level for some reasons, It's the first time I use Zigbee2MQTT so quite unsure what to do about it

closed 2022-05-13 by github-actions[bot]