Koenkk/zigbee2mqtt #11199: [New device support]: Please add support for Yale YMF40A RL

juslex, 2022-02-02:
### Link

https://www.yalehome.com.br/pt-BR/produtos1/products/fechaduras-digitais-biometricas/fechadura-digital-ymf40arl/

### Database entry

{"id":24,"type":"EndDevice","ieeeAddr":"0x000d6f00168a4334","nwkAddr":58632,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"06ffff2027","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"06ffff2027","manufacturerName":"ASSA ABLOY iRevo","powerSource":3,"zclVersion":1,"appVersion":11,"hwVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":360}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}},"closuresDoorLock":{"attributes":{"lockState":1}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0f070","endpointID":1},{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0f070","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0f070","endpointID":1}],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":11,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":923668394},"lastSeen":1643830970574,"defaultSendRequestWhen":"fastpoll"}

### Comments

I added the following entry to yale.js using the same code as YMF40/YDM4109+ but I don't have correct information about the lock battery. apart from this error, the lock seems to work correctly

### External converter

```shell
{
        zigbeeModel: ['06ffff2027'],
        model: 'YMF40A',
        vendor: 'Yale',
        description: 'Real living lock / Intelligent biometric digital lock',
        extend: lockExtend(),
    },
```

juslex, 2022-02-02:
[image]

juslex, 2022-02-02:
[image]

Koenkk, 2022-02-03:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3802

closed 2022-02-03 by Koenkk

cmagno369, 2024-04-08:
@Koenkk 
Can you please add for YMF40A RL:

 {
        zigbeeModel: ['06e01d220c'],
        model: 'YMF40A RL',
        vendor: 'ASSA ABLOY iRevo',
        description: 'Real living lock / Intelligent biometric digital lock',
        extend: lockExtend(),
    }

Koenkk, 2024-04-09:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.