Koenkk/zigbee2mqtt #11200: No converter available for 'transition' (10) from a TuYa TS0601_dimmer

cholawo, 2022-02-02:
### What happened?

I sent `{"brightness": 75, "transition":10 }`.
The transition time is ignored and the brightness changes immediately.

### What did you expect to happen?

Should transition

### How to reproduce it (minimal and precise)

Send `{"brightness": 75, "transition":10 }`.

In case this is relevant, the actual device I bought is a [Moes Go Rotary Dimmer](https://www.amazon.co.uk/gp/product/B091GG6Q2T/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1) but it was detected as the TuYa TS0601_dimmer.

### Zigbee2MQTT version

 1.23.0-dev commit: ddcb8e12

### Adapter firmware version

20211217

### Adapter

Sonoff zigbee 3.0 usb

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7990069/log.txt)

Koenkk, 2022-02-03:
This device uses non-standard converters for controlling the brightness. At the moment they do not support transition. I'm not sure if these device even do support it, someone has to check whether this is possible when using the device with the original gateway?

cholawo, 2022-02-03:
Thanks for responding so quickly!  I can see no mention of transition in [this review](https://www.ixbt.com/live/chome/zigbee-dimmer-moes-delaem-osveschenie-umnym-tuya-smart-integraciya-v-home-assistant.html) so I it probably isn't supported by the device.   I also note that brightness_move and brightness_step don't work.   Do you think these commands _do_ work with _some_ of the TS0601_dimmer models and I was just unlucky with the one I bought?

Koenkk, 2022-02-04:
I expect it works with none, since it doesn't use standard zigbee commands

closed 2022-02-07 by cholawo