Koenkk/zigbee2mqtt #11201: value_template removed for light in HA 2022.2

MiguelAngelLV, 2022-02-02:
### What happened?

My livolo switches configured as light fail in Home Assistant 2022.2:

```
The 'value_template' option has been removed, please remove it from your configuration
```

### What did you expect to happen?

Light add to HA

### How to reproduce it (minimal and precise)

Configure a switch as light and use HA 2022.2

### Zigbee2MQTT version

1.23

### Adapter firmware version

20211210

### Adapter

cc2351

The-Deep-Sea, 2022-02-02:
See #8033.

MiguelAngelLV, 2022-02-02:
@The-Deep-Sea  thanks you. it works after edit the device config.

closed 2022-02-03 by Koenkk

Diddlik, 2022-02-05:
I have this problem too, but can solve it.
I have this in z2m config
```
'0x842e14fffe0febde':
  friendly_name: Light.Abstellraum
  homeassistant:
    switch:
      type: light
      object_id: light
    light:
      name: Light.Abstellraum
      state_value_template: '{{ value_json.state }}'
```
in HA MQTT Topic it is discovered like:

```
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state"
    }
  ],
  "command_topic": "zigbee2mqtt/Light.Abstellraum/set",
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x842e14fffe0febde"
    ],
    "manufacturer": "TuYa",
    "model": "Wall switch module (WHD02)",
    "name": "Light.Abstellraum"
  },
  "name": "Light.Abstellraum",
  "payload_off": "OFF",
  "payload_on": "ON",
  "state_topic": "zigbee2mqtt/Light.Abstellraum",
  "state_value_template": "{{ value_json.state }}",
  "unique_id": "0x842e14fffe0febde_light_zigbee2mqtt",
  "value_template": "{{ value_json.state }}"
}
```
and in HA logs I get following error message:
```
 The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred:
```

MiguelAngelLV, 2022-02-06:
@Diddlik check the updated guide:

https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#exposing-switch-as-a-light

You need add `value_template: null`

```
'0x842e14fffe0febde':
  friendly_name: Light.Abstellraum
  homeassistant:
    switch:
      type: light
      object_id: light
    light:
      name: Light.Abstellraum
      value_template: null
      state_value_template: '{{ value_json.state }}'
      
 ```

Diddlik, 2022-02-06:
I stopped ha and z2m. Deleted discovered topic from z2m. everything from this switch. 
Updated z2m device.yaml to:
```
'0x842e14fffe0febde':
  friendly_name: Light.Abstellraum
  homeassistant:
    switch:
      type: light
      object_id: light
    light:
      name: Light.Abstellraum
      value_template: null
      state_value_template: '{{ value_json.state }}'
```
than started HA and z2m....the same problem.

Diddlik, 2022-02-06:
ok, it is not completly broken. Rhis solution works for single switch. For double switch it does not work. In the discovery topic I have old values with value_template:
[image]

@Koenkk can you confirm, that it works for double switch? 

```

'0x086bd7fffe5d8eaf':
  friendly_name: Switch.Eingang
  homeassistant:
    switch_left:
      type: light
      object_id: light_eingang
    light_left:
      name: Switch.Eingang_left
      value_template: null
      state_value_template: '{{ value_json.state_left }}'
    switch_right:
      type: light
      object_id: light_hausnummer
    light_right:
      name: Switch.Eingang_right
      value_template: null
      state_value_template: '{{ value_json.state_right }}'
```

MiguelAngelLV, 2022-02-07:
@Diddlik I used double switch (livolo 2 gang) and it works for me.

Diddlik, 2022-02-07:
Hmm, ok, I try to delete and reconfig it again...