Koenkk/zigbee2mqtt #11202: Gledopto GL-C-002P (Mini 5in1) unpairable

MartB, 2022-02-02:
### What happened?

Trying to pair the GL-C-002P aka Mini version of the 5in1 controller (GL-C-001P).
It initially pairs once but never works after a powercycle, when trying to re-pair the device it never gets paired correctly.
It constantly leaves the network and re-joins as long as the joining is enabled from within z2m
The device gets detected as GL-C-008P correctly but it doesnt "stick" to the network.

### What did you expect to happen?

Expected a normal pairing proccess to happen just like with the older controllers. 

### How to reproduce it (minimal and precise)

I guess you just have to buy the mini 5in1 pair it once, power cycle or reset it and it never joins your network again.
Until anyone confirms this device works i assume this is the case for all of these new controllers.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20220103

### Adapter

SLAESH CC2652RB 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7990528/log.txt)

Koenkk, 2022-02-03:
Could you provide the herdsman debug logging when pairing this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

MartB, 2022-02-03:
[debug_glc002p.log](https://github.com/Koenkk/zigbee2mqtt/files/7996217/debug_glc002p.log)
@Koenkk Thanks for your help, i appreciate it. 

I kept everything in there as im going to change secrets anyway, this is the full docker log from the addon with like 3-4 pairing attempts. Started from a completely clean state besides the coordinator firmware flash.

Koenkk, 2022-02-04:
@MartB seems the device just leaves the network, since gledopto advertises that their devices work with the hue bridge, do you have one? If yes, please check if it working by pairing it to the hue bridge
.

MartB, 2022-02-04:
@Koenkk I tried with a borrowed echo show and it seems to work there, did not work on the first pairing attempt but it got picked up. Z2M doesnt work tho, sad so i guess theres nothing you can do?

Gledopto does not seem to be super keen on trying to fix it as well.

Koenkk, 2022-02-04:
@MartB do you maybe have a CC2531 stick? If yes, try sniffing the traffic when pairing the device with the echo. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

MartB, 2022-02-04:
@Koenkk  Sadly i dont, can you recommend an amazon listing / source to obtain this from within europe?
I guess i need the debugger + stick if i want to try?

Koenkk, 2022-02-05:
@MartB I usually buy from aliexpress, you indeed need the debugger + stick. But another sonoff dongle with the sniffer firmware should also do it

MartB, 2022-02-05:
> @MartB I usually buy from aliexpress, you indeed need the debugger + stick. But another sonoff dongle with the sniffer firmware should also do it

Aliexpress would have been viable, but the holidays are delaying things.
I got one from amazon and im gonna see if it works haha, should be there next week and then im gonna capture both pairing processes.

Thanks for your help so far, maybe we can figure sth out, now that a second user reported the same im sure its not a "me" problem.

basmeyer, 2022-02-05:
Would the Sonoff ZBDongle Plus (CC2652) work to flash with a sniffer firmware? @Koenkk 
Found this https://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/3.20.00.68/exports/docs/zigbee/html/zigbee/packet_sniffer.html at Texas Instruments, but PC... the thing is, all Mac here. But plenty of Pi's. :)
Any links?

basmeyer, 2022-02-05:
As a summary to my experience with these fine Zigbee low voltage DC LED controllers, of which we probably hear more soon because the features they posses together with the purchase price is almost mind blowing, I currently have 5, of which I was able to pair 2 completely successfully with a tiny hickup. But 3 of them are "expelled" from proper pairing. They are recognised but depending of the integration not working or kicked out of the network almost straight away.

ZHA perfectly works with 2 out of 5 of the properly paired devices.
Zigbee2MQTT currently works with neither of them, not even the 2 same mentioned above. After pairing they last 10-15 seconds and leave the network.

I hope I can assist with the Zigbee sniffing.

Bysmyyr, 2022-02-06:
Same happens to me, connect and then disconnect in about a minute. I have Tubes cc2652 lan connected adapter.

Koenkk, 2022-02-06:
> Would the Sonoff ZBDongle Plus (CC2652) work to flash with a sniffer firmware? @Koenkk Found this [...]

yes that should do it

wtip, 2022-02-19:
@MartB any luck with the sniffing?

basmeyer, 2022-02-19:
Has anyone contacter the manifacturer to ask what they have changed compared to the 001P?

Bas

wtip, 2022-02-20:
@basmeyer 
I just sent an email to `service@gledopto.com` but I'm doubtful that I'll get a useful response.

basmeyer, 2022-02-20:
Thanks.
I hope they realize that possible use of this product in open source systems will always contribute in a sales boost.

kirovilya, 2022-02-20:
I will receive such a device today. Successfully paired in z2m on stick 2531.
When the operating mode is changed, the device leaves the network and pairs again with a new behavior model.
RGBCCT - GL-C-008P
RGBW - GL-C-007P
RGB - GL-C-003P
CCT - GL-C-006P
Dimmer - GL-C-009P

kirovilya, 2022-02-20:
As for pairing problems, you may need to pay attention to RF noise, since this device can work with RF remotes, maybe this somehow interferes with the process and throws it out of the network.

I did not find such problems with my device - everything works fine

wtip, 2022-02-20:
@kirovilya Can you share your Gledopto device firmware version?

Mine is:
```
firmware build date:20211223	
firmware version: 10251203
```
and if it matters I'm also using a SLAESH CC2652RB usb adapter as the coordinator.

kirovilya, 2022-02-20:
the same
[image]

MartB, 2022-02-20:
> @MartB any luck with the sniffing?

Yes, im going to sniff the amazon echo device tomorrow but i sniffed the pairing process with my network and the device leaves after sending some APS: Command thats encrypted, i can provide the decrypted dump to @Koenkk if he wants to take a look at it.

> Has anyone contacted the manifacturer to ask what they have changed compared to the 001P? 

I did and they do not offer support past the listed available hubs, im afraid this one is gonna be on us.

> As for pairing problems, you may need to pay attention to RF noise, since this device can work with RF remotes, maybe this somehow interferes with the process and throws it out of the network.
> 
> I did not find such problems with my device - everything works fine

I doubt RF noise is related, i do not own any of this Mi-Boxer RF protocol 2.4 ghz remotes and if you check the video i made it doesnt get any better than this RF wise. There is some software incompatibility between this telink stuff, maybe its related to the ZCL Green Power support this SOC has. 

If you want to risk "soft-bricking" the device, kick it from your network and reset it and then try to re-pair it again.
And if you have a working sniffing setup please check if it also sends the Update Device, Device Left packet straigth after sending the same APS: Command.
[image]

kirovilya, 2022-02-20:
I confirm! the problem is observed on the stick cc2652 (zstack 3) and is absent on cc2531 (zstack 1.2)
[gl-c-002p.zip](https://github.com/Koenkk/zigbee2mqtt/files/8104754/gl-c-002p.zip)

kirovilya, 2022-02-20:
one more test - it works with EFR32 EZSP stick
[gl-c-002p_ezsp.zip](https://github.com/Koenkk/zigbee2mqtt/files/8104770/gl-c-002p_ezsp.zip)

MartB, 2022-02-20:
> I confirm! the problem is observed on the stick cc2652 (zstack 3) and is absent on cc2531 (zstack 1.2) [gl-c-002p.zip](https://github.com/Koenkk/zigbee2mqtt/files/8104754/gl-c-002p.zip)

Thank you for giving this a shot!
I think this is gonna be the "missing link" because you have the keys to decrypt the actual traffic for 1) failure and 2) success here.
Let's hope you or someone else figure this out. If you are willing to share your network key in private with koenkk or anyone else trusted and capable enough this could shine some more light on the issue.

lorenz, 2022-02-20:
@MartB The key is in the capture, just decrypt with the default link key and then grab it from the transport key message.

MartB, 2022-02-20:
> @MartB The key is in the capture, just decrypt with the default link key and then grab it from the transport key message.

Thanks for that much appreciated!

lorenz, 2022-02-20:
I've started digging around the capture and I think packet 122 in the 2652 capture is interesting. It's extremely short, not decryptable and doesn't occur at all on the 2531 capture.  Also the captures start to heavily diverge afterwards. Directly preceding it is the node descriptor info from the coordinator where the only significant difference is the stack compliance revision increasing from 0 to 22. The device then keeps issuing node descriptor requests and these weird undecryptable command frames every 5 seconds for 4 cycles before leaving exactly 20s after it has issued the first node descriptor request.

MartB, 2022-02-21:
@lorenz This might explain it:
[image]

I guess we are seeing the attempted (or working) key update, as the stack compat level is high enough?

lorenz, 2022-02-22:
But TI wrote that article so I'd assume that their Z-Stack implementation does not have a broken key update function. Maybe this is something Zigbee2MQTT needs to handle? I'm not very familiar with the interface between Zigbee2MQTT and Z-Stack.

lorenz, 2022-02-26:
I got it to work by hacking the coordinator firmware (just zeroed out the Zigbee stack version for GLEDOPTO OIDs), but that's not really a good solution. If you want to try, you can grab the hacked firmware at https://blob.dolansoft.org/public/cc13_2652p-coordinator-with-manager.hex.

The issue is really weird as for the EZSP trace posted above the device behaves normally, but for the TI trace it uses an unknown link key to encrypt the key request. For the Echo it does the same but the Echo is somehow able and willing to decrypt the response with the weird link key. Since pairing worked for the first time my best guess is that the stack does not delete the link key upon leaving the network and reuses that link key when requesting a new one. But Zigbee2MQTT has already deleted that link key and thus Z-Stack can't decrypt the request. Maybe the Echo keeps the link key around indefinitely and is thus able to decrypt and just ignores the protocol violation?

lorenz, 2022-02-26:
@MartB Yeah, I'm pretty sure that that whatever ZigBee Stack the device is using is not clearing the link key when it's leaving. But without a dump from a fresh pairing process that's hard to confirm and I currently don't have two USB sticks so I can't sniff. I've been looking at what the Echo does and it looks like it just answers with the default link key (which we could patch into the firmware), but the device does not want to verify that key so this key exchange doesn't seem to actually work. The one that's working is using unknown link keys. So this probably needs further sniffing with a coordinator that was not previously paired (or in case that key is not per-coordinator even a new device).

MartB, 2022-02-27:
> @MartB Yeah, I'm pretty sure that that whatever ZigBee Stack the device is using is not clearing the link key when it's leaving. But without a dump from a fresh pairing process that's hard to [...]

Trying once more to get gledopto involved, im pretty sure i can "factory new" pair one more of these until i run out of attempts.
So the following is on my todo list now:

## General
- [ ] Figure out if this is some sort of special bdb device comissioning (unlikely)
- [ ] Check for possible explanations which keys it might generate and use

## Sniffing to do list
- [ ] existing device -> reset -> pair with echo and sniff from start to finish
- [ ] factory new device FN -> pair with z2m (believed to be working)
- [ ] reset FN device -> re-pair with z2m
- [ ] FN never paired with echo -> reset FN -> pair FN with echo
- [ ] remove FN from echo -> reset FN -> pair FN with echo

patrikulus, 2022-03-25:
> Minor victory for people fond of hardware tinkering that want to learn some new stuff. **tldr: The device does not fully reset its persistent zigbee network data on factory reset, somehow breaking the key exchange in the process, manually full resetting the flash area solves this.**
> 
> ## Preface
> I could not keep my hands off this device, i dumped the flash of the device and discovered a region that is saving information about the zigbee networks it was connected to.
> 
> ## Flash analysis
> [image] (Top contains my zeroed memory, bottom is a leftover taken straigth after a firmware reset)
> 
> It seems like the deletion on factory reset is off by one, as it never gets fully reset. I cleared 24 KiB of flash area at offset `0x7A000` and the device instantly paired with z2m again and stays paired.
> 
> I later found a manual that mentions this too (see page 23): http://wiki.telink-semi.cn/tools_and_sdk/Demo/B91_Zigbee/Telink_Zigbee_Overview_CN.pdf
> 
> So i can at least revive / full reset the module at will by using https://github.com/pvvx/TlsrComSwireWriter
> 
> 1. Building a small board and grabbing a UART serial adapter
> 2. Opening the case (a pain)
> 3. Soldering 3 wires to the exposed GND, SWS and VCC pads.
> 4. Hooking it up to the pc and running
>    `python ./TLSR825xComFlasher.py -p /dev/ttyUSB1 --tact 70 es 0x7A000 0x6000`
> 
> ## How is this relevant?
> Well @lorenz and me were right and the device indeed remembers parameters it should not, and i would not be surprised if these remembered bytes are the missing keys (not sure if in the clear) or other parameters that the device uses to encrypt the AP message for request key with.
> 
> I hope Gledopto will take a look at this and at least fix this bug. Now we only need to figure out how the echo device "overrides" this faulty behavior in a way that it still pairs, but that does explain why the pairing also fails there a few times.

I'm confused about the schema, should I connect only TX to SWS or also RX? Sorry for dump question, but I'm new into flashing stuff, so far I have only "tasmotized" few ESP devices.

MartB, 2022-03-26:
@patrikulus You should have success when connecting it like this: https://github.com/pvvx/TlsrComSwireWriter/raw/master/schematicc.gif

Basically, the SWS pad on the controller" the controller needs to be run to RX of the UART controller and to the TX part, with a 1 - 1.8k resistor in between.

Koenkk, 2022-03-26:
@lorenz could you try hacking the manufacturer code in the coordinator firmware? The echo sends [0x1217](https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1046387855), maybe the default `0x0000` manufacturer code triggers this bug on the gledopto device.

MartB, 2022-03-27:
@Koenkk i did and it did not work, same behavior.

[znp_LP_CC2652RB_tirtos_ccs.zip](https://github.com/Koenkk/zigbee2mqtt/files/8356639/znp_LP_CC2652RB_tirtos_ccs.zip)
k

(Compiled with compiler speed optimizations enabled, thats why its bigger)

Koenkk, 2022-03-27:
@MartB and what about the `Network Manager: True`? In [your post](https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1046387855) it is true for Echo and false for z2m. In the [EZSP sniff](https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1046303717) from @kirovilya it is also true:

[image]

lorenz, 2022-03-27:
I hacked in network manager support already, doesn't do anything

Koenkk, 2022-03-28:
If I understand correctly, the GL-C-002P doesn't show this behaviour with EZSP right? (sniff: https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1046303717)

lorenz, 2022-03-28:
Me and @MartB think since it's related to the device retaining link keys even after disassociation it would happen to the EZSP too if reconnected. It also happens with the Echo, but it responds differently and after a few tries it recovers.

lorenz, 2022-03-29:
@czdaniel If you have a 3.3V serial adapter you can try the physical reset procedure which resets the flash of the unit. After that switch the unit to the right mode and then pair it. It will work the first time.

basmeyer, 2022-03-29:
Will this also work with like 5 of them? After the first succesful pairing I had the feeling it did not want to pair with more devices. But I could be wrong as things got messy when I also tried other modes at the same time. In he end none of them was working any more with neither z2m nor ZHA.
Yeah, perhaps a bit brave to order 5 of these already. ;)

Bas

czdaniel, 2022-03-29:
I got a hue bridge just to test if it works with that and it does. My amazon echo might have other problems.
My power supply cables are too beefy for these "easy to use" push in connectors, which do not inspire confidence and I accidentally fried one already. I was under the impression that these could seperately dim 5 leds strips by the barebones manual online, but this is not the case either. I have a good 140W meanwell power supply but when I dim the light the gledopto emits a high pitched whine, possibly due to that bad connection on the 12v input. I've ordered a few Miboxer FUT036Zs and we'll see if they perform better.

Koenkk, 2022-03-29:
@ptvoinfo any chance your firmware could be flashed on these devices?

ptvoinfo, 2022-03-30:
@Koenkk Sorry, but the firmware does not support TELINK chips.

Gudde660, 2022-03-31:
I'm also experiencing this issue with the gl-s-004p from Gledopto. They paired succesful the first time and after a factory reset they pair join the network and disconnect after a few seconds.

Is there something i can do to help?

MartB, 2022-03-31:
Currently trying to flash the TL SDK RGB light sample to the device, I extracted the original firmware, and it's working, okay so I cant really brick it I hope haha. Maybe this shines more light on what is going on.

@Gudde660 You can try to find out if this device also uses a telink SOC

lorenz, 2022-03-31:
@MartB I would like to tinker with the code as well, can you tell me how you got the SDK reference code running?

MartB, 2022-04-01:
> @MartB I would like to tinker with the code as well, can you tell me how you got the SDK reference code running?

Sure! Feel free to contact me on the z2m discord (`@MartB - Martin#2975`), its a little bit too off-topic for this issue. It wasnt particulary hard but the sample lacks a lot of functionality, i figured out the right PWM and button mapping and the device performs just like i expect it to in color temp mode (without transitions). Reconnecting fine every time, no matter what how i reset/remove it.

Im going to analyze the flash of this firmware in a few days and compare it to what gledopto did on the factory resets.

Koterak, 2022-04-16:
Have same issue with GL-B-007P 6W bulb. I opened one bulb and found that it's using TLSR8258 chip. Is there anything I can do to get it working?

lorenz, 2022-04-16:
@Koterak You can try my modified firmware (see https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1052137967) but be aware it might break any already-paired GLEDOPTO devices. It rewrites node information so that the GLEDOPTO device doesn't attempt a Zigbee 3.0 key exchange. Other options are resetting the flash manually (see https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1053144581) or (not fully tested) temporarily joining the bulb to a different coordinator).  Long-term this needs a firmware patch either from GLEDOPTO or from the community to fix their broken state management.

xking3, 2022-05-06:
I have also the same issue and in total 120 Spott whit this problem. How can i build that cable and stream line the prozess. Could anyone please help me with a description of the prozess which makes it possible to follow it for somebody with less HW/SW skills ?
Thanks

xking3, 2022-05-07:
Meanwhile  have talked to Gledopto. They told me that this is not the case with a HUE Hub.  I have one and will try if I can connect with it, than give it free again and try to reconnect with zigbee2mqtt.
If someone has any idea if this could work I would be glad to know. I will be able to perform a test thisMonday.
@MartB @Koenkk

MartB, 2022-05-08:
> I have also the same issue and in total 120 Spott whit this problem. How can i build that cable and stream line the prozess. Could anyone please help me with a description of the prozess which makes it possible to follow it for somebody with less HW/SW skills ? Thanks

I would advise against trying this if your primary goal is to get a working device. You can flash the Firmware provided by @lorenz further up in this thread or use an older coordinator chipset e.g cc2531.

If you want to play around with it, let me know and I can fill in the underexplained gaps.

> Meanwhile have talked to Gledopto. They told me that this is not the case with a HUE Hub. I have one and will try if I can connect with it, than give it free again and try to reconnect with [...]

Yeah Gledopto is stubborn and insists they did not do anything wrong.
Which is arguably somewhat true, as the device works as advertised for outdated chipsets (CC2531) and devices with countless of interop. workarounds in place (Amazon Alexa, Philips Hue, ...).

You could theoretically try to "overwrite" the first entry of the "past network" data by resetting the device and pairing it to a working device X (to be determined) times. I could try to find out the exact number of resets required, would be a unusable but funny workaround tho haha. A single reset is not gonna be enough, as the list gets appended.

xking3, 2022-05-08:
> If you want to play around with it, let me know and I can fill in the underexplained gaps.

I would very much appreciate,  if you could help. To begin it would be very help full if you could explain more in detail what hardware I need to make such a cable. Do I need some sort of a USB Dongle to start with? And then from there do the soldering as shown on his picture?

>Yeah Gledopto is stubborn and insists they did not do anything wrong.
Which is arguably somewhat true, as the device works as advertised for outdated chipsets (CC2531) and devices with countless of interop. workarounds in place (Amazon Alexa, Philips Hue, ...).

Well I hope to talk to one of the software guys from them tomorrow. I already did send them this thread. I think this gives them a more detailed view of the problem, thanks to your detailed work here. I hope that the help to solve the problem, as it would show the community that they really care about there product and there clients. This would be also a great opportunity for them. I think in the future all the products with the old chipset will be needed to be updated and then their devices would stop running. If this happens nobody would by able to use the LED's form Geldopto anymore and that would be very bad. Because they are in total very good quality. 

> You could theoretically try to "overwrite" the first entry....

I will try. This is the fastest thing I can do :-) So I will let you know how many times I have to reset, as long it is in between 2-10 times. ;-)

Now one other question. Let's say I can connect it to a Hue Hub. Would it be possible to do the firmware update from there?

xking3, 2022-05-10:
I got an updated firmware from gledotpto today to test. I am now organizing a conbee ii stick to update the firmware. I will let you know what the result as soon I have succeeded to flash the firmware.

basmeyer, 2022-05-10:
Did Gledopto give any release notes with this update? Is it beta? And if so, do restrictions to share this firmware apply?

MartB, 2022-05-10:
> I got an updated firmware from gledotpto today to test. I am now organizing a conbee ii stick to update the firmware. I will let you know what the result as soon I have succeeded to flash the firmware.

Are you allowed to provide the firmware?
I can verify "in-flasah" if it changes the clearing behavior.

xking3, 2022-05-10:
The file is for GL-D-004P V105 20220427
I don't know if it is a beta. I can not share it for the moment because I don't know if I have permission. But I will ask them.

xking3, 2022-05-11:
Following news: The GL-D-004P I have is now updated with a 0x1452 version. I don't know if this is the newest image. But i think not. On the LED GL-D-007P I have an image 0x1455 right now. 
I have updated the 004P with a Conbee II stick over deCONZ and it works now with the zigbee2mqtt with a 3.0 Sonoff stick.
I am waiting now for a firmware for the 007P LED. If this works too, I will proceed to ask them if I can release the Firmware for these LED's

But right now i am asking my self if it would not make sense to connect the LED's over Conbee II to Loxone instead with the 3.0 Sonoff. Perhabs somebody can tell me if that would be better... ore in some way easier when i have to program the routines of the light's.

[31 more replies]