Koenkk/zigbee2mqtt #11204: Xiaomi Aqara wireless remote switch H1 (double rocker) (WRS-R02) Missing Hold Actions

waihsing, 2022-02-03:
### What happened?

When adding Aqara wireless remote switch H1, the hold actions are missing.

### What did you expect to happen?

Hold actions such as hold_left, hold_right, hold_both to be available as it was in 1.22.2

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

MQTT Messages

Successfully interviewed '0x54ef4410001afee1', device has successfully been paired
Device '0x54ef4410001afee1' is supported, identified as: Xiaomi Aqara wireless remote switch H1 (double rocker) (WRS-R02)
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara wireless remote switch H1 (double rocker)","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["single_left","single_right","single_both","double_left","double_right","double_both","triple_left","triple_right","triple_both"]},{"access":7,"description":"Click mode, fast: only supports single click which will be send immediately after clicking.multi: supports more events like double and hold","name":"click_mode","property":"click_mode","type":"enum","values":["fast","multi"]},{"access":7,"description":"Operation mode, select \"command\" to enable bindings (wake up the device before changing modes!)","name":"operation_mode","property":"operation_mode","type":"enum","values":["command","event"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WRS-R02","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for illuminance, takes into effect on next report of device.","name":"illuminance_precision","property":"illuminance_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the illuminance value (percentual offset), takes into effect on next report of device.","name":"illuminance_calibration","property":"illuminance_calibration","type":"numeric"}],"supports_ota":false,"vendor":"Xiaomi"},"friendly_name":"0x54ef4410001afee1","ieee_address":"0x54ef4410001afee1","status":"successful","supported":true},"type":"device_interview"}'

z-x-f, 2022-02-03:
Switch 'multi_mode' to enable double-click and hold

z-x-f, 2022-02-03:
I think I get what you mean.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3806

closed 2022-02-04 by Koenkk

svenjochems, 2023-02-08:
Hi

Is it also possible to add the 'hold' and 'release' actions for the WS-EUK04 model?