Koenkk/zigbee2mqtt #11206: [New device support]: Profalux Visio Cover

glamirand, 2022-02-03:
### Link

https://www.profalux.com/produits/volets-roulants/visio/

### Database entry

        "0x20918a0000045a95": {             "date_code": "20151006091b090",             "definition": {                 "description": "Profalux cover motor with position control",                 "exposes": [                     {                         "features": [                             {                                 "access": 3,                                 "name": "state",                                 "property": "state",                                 "type": "enum",                                 "values": [                                     "OPEN",                                     "CLOSE",                                     "STOP"                                 ]                             },                             {                                 "access": 3,                                 "description": "Position of this cover",                                 "name": "position",                                 "property": "position",                                 "type": "numeric",                                 "value_max": 100,                                 "value_min": 0                             }                         ],                         "type": "cover"                     },                     {                         "access": 1,                         "description": "Link quality (signal strength)",                         "name": "linkquality",                         "property": "linkquality",                         "type": "numeric",                         "unit": "lqi",                         "value_max": 255,                         "value_min": 0                     }                 ],                 "model": "Visio",                 "options": [                     {                         "access": 2,                         "description": "Inverts the cover position, false: open=100,close=0, true: open=0,close=100 (default false).",                         "name": "invert_cover",                         "property": "invert_cover",                         "type": "binary",                         "value_off": false,                         "value_on": true                     }                 ],                 "supports_ota": false,                 "vendor": "Profalux"             },             "endpoints": {                 "1": {                     "bindings": [                         {                             "cluster": "genOnOff",                             "target": {                                 "endpoint": 1,                                 "ieee_address": "0x00124b00237ba1a2",                                 "type": "endpoint"                             }                         },                         {                             "cluster": "genLevelCtrl",                             "target": {                                 "endpoint": 1,                                 "ieee_address": "0x00124b00237ba1a2",                                 "type": "endpoint"                             }                         }                     ],                     "clusters": {                         "input": [                             "genBasic",                             "genIdentify",                             "genGroups",                             "genScenes",                             "genOnOff",                             "genLevelCtrl",                             "genTime",                             "genCommissioning",                             "closuresShadeCfg",                             "64544",                             "64545"                         ],                         "output": [                             "genIdentify",                             "64544"                         ]                     },                     "configured_reportings": [                         {                             "attribute": "onOff",                             "cluster": "genOnOff",                             "maximum_report_interval": 3600,                             "minimum_report_interval": 0,                             "reportable_change": 0                         },                         {                             "attribute": "currentLevel",                             "cluster": "genLevelCtrl",                             "maximum_report_interval": 3600,                             "minimum_report_interval": 0,                             "reportable_change": 0                         }                     ],                     "scenes": []                 }             },             "friendly_name": "Bureau/Volet",             "ieee_address": "0x20918a0000045a95",             "interview_completed": true,             "interviewing": false,             "network_address": 63947,             "power_source": "Mains (single phase)",             "supported": true,             "type": "Router"         }

### Comments

I tried (and managed) to integrate my covers using the following converter.

The cover doesn't seems to provide modelId nor ManufId which make things more difficult..

I'm stuck using ieee addresses as fingerprint and I'm looking to use something more generic like manufacturer specific endpoints in order to make this converter work for anyone but I don't manage to find the propper syntax (and I basically don't know where to find any kind of documentation about fingerprints better than exploring other converters ..)

Thanks for your help!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils')
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    profalux_cover_state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            switch (value) {
                case 'OPEN':
                    await entity.command('genOnOff', 'on', {}, utils.getOptions(meta.mapped, entity));
                    break;
                case 'CLOSE':
                    await entity.command('genOnOff', 'off', {}, utils.getOptions(meta.mapped, entity));
                    break;
                case 'STOP':
                    await entity.command('genLevelCtrl', 'stop', {}, utils.getOptions(meta.mapped, entity));
                    break;
                default:
                    throw new Error(`Value '${value}' is not a valid cover position (must be one of 'OPEN' or 'CLOSE' or 'STOP')`);
                };
        },
    },
};

const definition = {
    fingerprint: [
        //{dateCode: '20151006091b090'},
        //{dateCode: '20151006091c090'}
        {ieeeAddr: '0xXXXXXXXXXXXXXXXX'},   //Salon 1
        {ieeeAddr: '0xXXXXXXXXXXXXXXXX'},   //Salon2
        {ieeeAddr: '0xXXXXXXXXXXXXXXXX'},   //Chambre
        {ieeeAddr: '0xXXXXXXXXXXXXXXXX'},   //Bureau
        {ieeeAddr: '0xXXXXXXXXXXXXXXXX'}    //Cuisine      
        //{endpoints: {1: {inputClusters: ["genBasic", "genOnOff", "genLevelCtrl", "64544", "64545"]}}}
    ],

    model: 'Visio',
    vendor: 'Profalux',
    description: 'Profalux cover motor with position control',
    fromZigbee: [fz.identify, fz.cover_state_via_onoff, fz.cover_position_via_brightness],
    toZigbee: [tzLocal.profalux_cover_state, tz.cover_via_brightness],
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET)],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);

        await endpoint.configureReporting('genOnOff', [{
            attribute: 'onOff',
            minimumReportInterval: 0,
            maximumReportInterval: 3600,
            reportableChange: 0,
        }]);

        await endpoint.configureReporting('genLevelCtrl', [{
            attribute: 'currentLevel',
            minimumReportInterval: 0,
            maximumReportInterval: 3600,
            reportableChange: 0,
        }]);
    }
};

module.exports = definition;
```

glamirand, 2022-02-09:
@Koenkk , any suggestions that might help? :)

Koenkk, 2022-02-09:
@glamirand can you provide the data/database.db entry? I see you included something in the OP but this doesn't look like the data/database.db entry (looks from `zigbee2mqtt/bridge/devices`?)  You can open the `data/database.db` file with a text editor.

glamirand, 2022-02-09:
@Koenkk , I just updated the OP with the expected data from database.db

Koenkk, 2022-02-10:
```js
fingerprint: [
    {type: 'Router', endpoints: [
        {ID: 1, profileID: 260, deviceID: 512, inputClusters: [0, 3, 4, 5, 6, 8, 10, 21, 256, 64544, 64545], outputClusters: [3, 64544]},
    ]},
],
``` 

should do it

Koenkk, 2022-02-13:
great, lets continue there.