Koenkk/zigbee2mqtt #11211: Changing type to light with HA 2022.2.0 not working

KennieNL, 2022-02-03:
### What happened?

Hi,

I have some switch devices that I expose to Home Assistant as light using the following in devices.yaml:
`"0xabcdef":
  friendly_name: bliepbloop
  homeassistant:
    switch:
      type: light
      object_id: light`

This has been working fine until updating to Home Assistant 2022.2.0, since then the devices are no longer shown in Home Assistant.

I don't really see anything related in the logs, but I see in the (breaking) changelog of home assistant that the use of value_template has been removed since this version: https://github.com/home-assistant/core/pull/62682

I suspect this is the issue.

### What did you expect to happen?

My switch devices being exposed as light in home assistant

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB

LiquidBlackNL, 2022-02-03:
Same problem, with the same config as mentioned by @KennieNL after upgrade to 2022.2.0.
Using a TuYa TS0121_plug, White-label: BlitzWolf BW-SHP13.

Zigbee2mqtt: 1.23.0-1
Adapter: slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff
Firmware version: 20210120

srozb, 2022-02-03:
Exactly the same issue with the latest stable z2m, ha 2022.02 and sonoff zbmini switch exposed as light class.

The-Deep-Sea, 2022-02-03:
See #8033.

Koenkk, 2022-02-04:
This is breaking change on the HA side, the docs have been updated with the new config: https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#exposing-switch-as-a-light

closed 2022-02-04 by Koenkk