Koenkk/zigbee2mqtt #11212: Xiaomi Aqara wireless remote switch H1 (double rocker) (WRS-R02) No longer work

MercianWhitey, 2022-02-03:
### What happened?

After the latest update my two WRS-R02 stopped working, both still show as connected but button taps return “null”. Everything else is working fine

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

NA

### Adapter

Sonoff Universal Zigbee 3.0 USB Dongle

### Debug log

Zigbee2MQTT:error 2022-02-02 17:21:09: Failed to configure 'Kitchen KP01', attempt 1 (Error: Bind 0x54ef4410002a1fed/1 genOnOff from '0x00124b00258d405a/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))

m0wlheld, 2022-02-03:
AFAIK with HA 2022.2 you need to specify state_value_template when exposing switches as lights. 

See https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended

MercianWhitey, 2022-02-03:
I had this issue with HA 2021-12 and the H1 I'm referring to are keypads and don't control light directly

advokatden, 2022-02-04:
Да, у меня такая же проблема с этим выключателем. Уже второй раз, в прошлый раз пришлось удалять устройство и заново прописывать. И вот сегодня опять такая же проблема

TheLion, 2022-02-04:
I have the same issue, but with Domoticz

The state reported by Z2M is different from before the latest update and the state reported in Domoticz (using domoticz-zigbee2mqtt-plugin), doesn't correspond with Z2M.

In MQTT the payload is now "single_left" as it was "button_1_single" and the Domoticz plugin reports "button_1_release". And "button_2_single has changed to "single_right", but is reported in Domoticz as "button_1_hold"

Is there a way to revert to the previous version of Z2M ?

z-x-f, 2022-02-04:
This update removes the release action, because release is not provided by the device itself, it is published by z2m after pressing the device button for 5s, and the device itself does not have the release action.

TheLion, 2022-02-04:
> This update removes the release action, because release is not provided by the device itself, it is published by z2m after pressing the device button for 5s, and the device itself does not have the release action.

Ok, that explains it, somewhat.

So there should be a change to domoticz-zigbee2mqtt-plugin to accommodate this change?

MercianWhitey, 2022-02-04:
> does
Ok, so how go I get my devices working again?

TheLion, 2022-02-04:
I downgraded to Z2M v1.22.2 and all works again.

Steekvlam, 2022-02-06:
Im having the same issue.
How can I downgrade?

MercianWhitey, 2022-02-06:
I couldn't work out how to downgrade in HA but I have my H1s working again by removing them completely from Z2M and re-adding them. The issue now though is that it only exposes left-tap and right-tap, double, triple and hold have gone

z-x-f, 2022-02-06:
The hold action has been added and will be fixed in the next update.https://github.com/Koenkk/zigbee-herdsman-converters/pull/3809
Triple action needs to be in multi mode, first click the button three times, then triple action will appear in HA.

MercianWhitey, 2022-02-06:
Thanks for the update

oliver989, 2022-03-11:
Is is safe to update to 1.24.0 or is it still broken?

TheLion, 2022-03-11:
Not sure. Haven't tried. 😬

MercianWhitey, 2022-03-11:
I've updated and my H1 keypads are still working

closed 2022-03-19 by github-actions[bot]