Koenkk/zigbee2mqtt #11217: Hue light 929003055901 incorrectly exposes color_xy

Klapperman, 2022-02-04:
### What happened?

Due to my own faulty example stated in the New Device report this device exposes a color_xy option which is does not support.
https://github.com/Koenkk/zigbee2mqtt/issues/10962 for my faulty request.

The example External Converter should have been `extend: extend.light_onoff_brightness_colortemp(),`. But it erroneously contains "_color" at the end. 

### What did you expect to happen?

Not seeing color options exposed through Zigbee2MQTT for the Hue light 929003055901 model.

### How to reproduce it (minimal and precise)

Install version 1.23.0-1 containing support for this light.

### Zigbee2MQTT version

 1.23.0 commit: unknown

### Adapter firmware version

20210708

### Adapter

Electrolama zig-a-zag-ah! (zzh!)

Koenkk, 2022-02-04:
Fixed

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-04 by Koenkk

Klapperman, 2022-03-03:
> Fixed
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hi @Koenkk issue is still there in Zigbee2MQTT version 1.24.0-1. Has it not been pushed to the stable branch?

Koenkk, 2022-03-03:
Seems I forgot to push it. Done it now, change will be available in the 1 april release.