Koenkk/zigbee2mqtt #11223: SmartThings IM6001-WLP01 water leak sensor stopped working after HA update

artohi, 2022-02-04:
### What happened?

I have two SmartThings IM6001-WLP01 water leak sensors. They have worked fine for a long time. Today I updated Home Assistant to the latest version 2022.2.1 and after that the water leak sensor says "water_leak" => N/A. Same thing with battery_low and tamper. Temperature, battery and linkquality are still working.

### What did you expect to happen?

Everything should work

### How to reproduce it (minimal and precise)

Update to the latest Home Assistant (2022.2.1)

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20220103

### Adapter

SONOFF ZigBEE 3.0 USB Dongle Plus

### Debug log

Received 09:26:48
QoS: 0
Payload: battery: 80
battery_low: null
linkquality: 65
tamper: null
temperature: 23.17
voltage: 2900
water_leak: null

artohi, 2022-02-04:
[image]

Koenkk, 2022-02-04:
Try triggering it once

artohi, 2022-02-04:
> 

That worked! Both sensors are now working. Thank you for the tip!

closed 2022-02-04 by artohi