Koenkk/zigbee2mqtt #11224: Hue dimmer switch hue button

Shodan43893, 2022-02-04:
### What happened?

When binding my Hue dimmer switch ([929002398602](https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602)) the on/off button and dimmer buttons work great: But the hue button does not perform any action.

### What did you expect to happen?

Since the on/off button toggles the lights. I expected the hue button to toggle between group scenes.

Weirdly, when checking the logs after pressing the hue button gives the following payload:
 MQTT publish: topic 'zigbee2mqtt/Dining room remote/action', payload 'off_press'

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26700700

### Adapter

 ConBee2/RaspBee2

MartB, 2022-02-04:
Check this: https://github.com/Koenkk/zigbee-herdsman-converters/pull/3815 

The new remote is toggling scenes, i guess philips does some hub magic but with normal zigbee stuff you can switch between two scenes if you set up the magic group ids correctly.

MartB, 2022-02-05:
> Oh wow thanks! I think this is a useful workaround. But I wonder if it would be possible to just cycle in between scenes of a given group, bypassing the magic group ID and limit of 2 scenes only.

I would assume one could "override" this by adjusting the behavior inside the zigbee converters, but im not knowledgeable enough within the zigbee2mqtt ecosystem to pull off something like that. It would need to be configurable though.

closed 2022-03-15 by github-actions[bot]