Koenkk/zigbee2mqtt #11226: [New device support]: Yet another TS0505B/Lidl Livarno lux light bar

amnyu, 2022-02-04:
### Link

https://www.lidl.pl/p/livarno-lux-listwa-oswietleniowa-led-zigbee-smart-home-1-sztuka/p100317584001

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xb4e3f9fffee55fb4","nwkAddr":28107,"manufId":4098,"manufName":"_TZ3210_iystcadi","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_iystcadi","stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1643982491643,"defaultSendRequestWhen":"immediate"}

### Comments

I've seen that there's already a support for this one: https://www.zigbee2mqtt.io/devices/14149505L_14149506L.html, but it doesn't work with my unit :( On the device itself it's called 14149505L.
I'm on dev branch:
1.23.0-dev commit: [9fb1eb4](https://github.com/Koenkk/zigbee2mqtt/commit/9fb1eb4)

### Supported color modes

color temperature, RGB

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

amnyu, 2022-02-04:
I've added an external converter (mostly copied from lidl.js):

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    fingerprint: [
        {modelID: 'TS0505B', manufacturerName: '_TZ3210_iystcadi'}
    ],
    model: '14156505L',
    vendor: 'Lidl',
    description: 'Livarno Lux light bar RGB+CCT (black/white)',
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    meta: {applyRedFix: true, enhancedHue: false},
    configure: async (device, coordinatorEndpoint, logger) => {
        device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 25});
    },
};

module.exports = definition;
```

Brightness and color temperature works great, but RGB doesn't :(

amnyu, 2022-02-08:
I've got this JSON from Tuya api (yeah, bought Lidl gateway and tinkered a bit :D). Can someone tell me if it's helpful? (It sure does look so).
```
{
  "result": {
    "category": "dj",
    "functions": [
      {
        "code": "switch_led",
        "dp_id": 1,
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "work_mode",
        "dp_id": 2,
        "type": "Enum",
        "values": "{\"range\":[\"white\",\"colour\",\"scene\",\"music\"]}"
      },
      {
        "code": "bright_value_v2",
        "dp_id": 3,
        "type": "Integer",
        "values": "{\"min\":10,\"max\":1000,\"scale\":0,\"step\":1}"
      },
      {
        "code": "temp_value_v2",
        "dp_id": 4,
        "type": "Integer",
        "values": "{\"min\":0,\"max\":1000,\"scale\":0,\"step\":1}"
      },
      {
        "code": "colour_data_v2",
        "dp_id": 5,
        "type": "Json",
        "values": "{\"h\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":360,\"step\":1},\"s\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1},\"v\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}}"
      },
      {
        "code": "scene_data_v2",
        "dp_id": 6,
        "type": "Json",
        "values": "{\"scene_num\":{\"min\":1,\"scale\":0,\"max\":8,\"step\":1},\"scene_units\": {\"unit_change_mode\":{\"range\":[\"static\",\"jump\",\"gradient\"]},\"unit_switch_duration\":{\"min\":0,\"scale\":0,\"max\":100,\"step\":1},\"unit_gradient_duration\":{\"min\":0,\"scale\":0,\"max\":100,\"step\":1},\"bright\":{\"min\":0,\"scale\":0,\"max\":1000,\"step\":1},\"temperature\":{\"min\":0,\"scale\":0,\"max\":1000,\"step\":1},\"h\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":360,\"step\":1},\"s\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1},\"v\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}}}"
      },
      {
        "code": "countdown_1",
        "dp_id": 7,
        "type": "Integer",
        "values": "{\"min\":0,\"max\":43200,\"scale\":0,\"step\":1}"
      },
      {
        "code": "music_data",
        "dp_id": 8,
        "type": "Json",
        "values": "{\"change_mode\":{\"range\":[\"direct\",\"gradient\"]}, \"bright\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"temperature\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"h\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":360,\"step\":1},\"s\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1},\"v\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1}}"
      },
      {
        "code": "control_data",
        "dp_id": 9,
        "type": "Json",
        "values": "{\"change_mode\":{\"range\":[\"direct\",\"gradient\"]}, \"bright\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"temperature\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"h\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":360,\"step\":1},\"s\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1},\"v\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1}}"
      }
    ],
    "status": [
      {
        "code": "switch_led",
        "dp_id": 1,
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "work_mode",
[... 27 lines cut ...]
      },
      {
        "code": "countdown_1",
        "dp_id": 7,
        "type": "Integer",
        "values": "{\"min\":0,\"max\":43200,\"scale\":0,\"step\":1}"
      },
      {
        "code": "music_data",
        "dp_id": 8,
        "type": "Json",
        "values": "{\"change_mode\":{\"range\":[\"direct\",\"gradient\"]}, \"bright\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"temperature\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"h\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":360,\"step\":1},\"s\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1},\"v\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1}}"
      },
      {
        "code": "control_data",
        "dp_id": 9,
        "type": "Json",
        "values": "{\"change_mode\":{\"range\":[\"direct\",\"gradient\"]}, \"bright\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"temperature\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":1000,\"step\":1}, \"h\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":360,\"step\":1},\"s\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1},\"v\":{\"min\":0,\"scale\":0,\"unit\":\"\",\"max\":255,\"step\":1}}"
      }
    ]
  },
  "success": true,
  "t": 1644343467913
}
```

closed 2022-03-19 by github-actions[bot]

alainkovacs, 2022-03-29:
I have the same light and the same problem, even though it is listed as supported, it does not seem to be. Related to this https://github.com/Koenkk/zigbee2mqtt/discussions/5944 This is I think due to Lidl changing the internals of the light which makes it quite confusing. I had exactly the same problem with a Blitzwolf/Tuya smart socket just a few days ago.
I am unsure on how to proceed here.. Can anybody help?

vandenbogerd, 2022-06-13:
Here also the same issue... tried the lidl.js approach already, but that only results in a "Bad gateway" notice...

alainkovacs, 2022-06-13:
@vandenbogerd I forgot to add an update here. How I made it work back then was to update my coordinator firmware (not sure if that helped but I find it better always to run on latest stable) and also what was back then the "dev" branch. Currently I am running on 1.25.1 and it works.

Lynetk, 2022-06-14:
> @vandenbogerd I forgot to add an update here. How I made it work back then was to update my coordinator firmware (not sure if that helped but I find it better always to run on latest stable) and also what was back then the "dev" branch. Currently I am running on 1.25.1 and it works.

What exactly works? I'm on 1.25.2 and still have the annoying behavior with colors being washed completely out by the white LED.

alainkovacs, 2022-06-14:
@Lynetk My light works exactly as expected. Does this also happen when you change the color via the frontend? Have you tried re-pairing it? Do you have a way to test it with the LIDL gateway/app? Could it be a hardware problem?

vandenbogerd, 2022-06-14:
> @vandenbogerd I forgot to add an update here. How I made it work back then was to update my coordinator firmware (not sure if that helped but I find it better always to run on latest stable) and also what was back then the "dev" branch. Currently I am running on 1.25.1 and it works.

How do I update that coordinator? :$

vandenbogerd, 2022-06-14:
> @alainkovacs It happens no matter what I do, except if I change the color and then power it off physically and then power it on again. Then I have the nice bright color I was expecting. I've tried repairing, I've tried from Home Assistant. Not sure what else to try.
> 
> I have two of these Lidl(Livarno) LED-bar's and they have this behavior.
> 
> [image]

How did you manage to let Z2M recognize the lamp?

Lynetk, 2022-06-14:
@vandenbogerd Just by using the latest zigbee2mqtt version which is 1.25.2 (it was also recognized in 1.25.1)

vandenbogerd, 2022-06-14:
> @vandenbogerd Just by using the latest zigbee2mqtt version which is 1.25.2 (it was also recognized in 1.25.1)

AHA! I'm still on 1.18.1-1... Let me check...

vandenbogerd, 2022-06-14:
> > @vandenbogerd Just by using the latest zigbee2mqtt version which is 1.25.2 (it was also recognized in 1.25.1)
> 
> AHA! I'm still on 1.18.1-1... Let me check...

SOLVED! Apparently I had a wrong repo linked to the add-on store... :$

vandenbogerd, 2022-06-14:
> @vandenbogerd Solved in what regard? That it's recognized? or is it working as expected with the ability to change colors after having the white LEDs activated?

I'm at work right now, so I can't check wether it's working in changing colors... For me the recognition was the first step ;-)

alterschlabberanesi, 2022-11-20:
Please See #12947 

Looking for someone to sniff the traffic when controlling this device via the TuYa gateway

RunnicFusion, 2023-01-23:
@vandenbogerd did you manage to test the color changing? Have these device since last week but RGB colors are not applied correctly. Some colors are working but not as chosen from the rgb color picker inside zigbee2mqtt OR my Domoticz setup.
Looks like more Lidl devices are suffering color temp/mode changes.

vandenbogerd, 2023-01-23:
Yeah, all working as expected. However I never use the other colors, other than "warm white"...