Koenkk/zigbee2mqtt #11231: Setting boost mode on Tuya TS0601 valve changes temperature setpoint (since 1.23.0)

Plawasan, 2022-02-04:
### What happened?

(Presumably) Since 1.23.0 changing the mode to Boost also changes the mode into which it returns. Previously when using Boost the valve would only open fully for the set amount of time and then return to its original state. Now the valve opens however the temperature setpoint changes and the valve returns to "Schedule" preset instead of the mode it was originally in (Manual).

### What did you expect to happen?

Valve returns to previous mode after boost time expires

### How to reproduce it (minimal and precise)

- set valve to manual and temperature of 24
- switch to Boost mode

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20210120

### Adapter

zzh!

### Debug log

**Pre-boost state**
pi@raspberrypi:~ $ sudo journalctl -u zigbee2mqtt.service -f | grep bathroom_radiator
Feb 04 19:38:51 raspberrypi npm[2013]: Zigbee2MQTT:info  2022-02-04 19:38:51: MQTT publish: topic 'zigbee2mqtt/bathroom_radiator', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":23,"current_heating_setpoint":26,"eco_temperature":20,"force":"normal","holidays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C","linkquality":69,"local_temperature":27.5,"local_temperature_calibration":0,"max_temperature":35,"min_temperature":5,"position":0,**"preset":"manual"**,"system_mode":"auto","update":{"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C"}'

**Boost enabled**
Feb 04 19:41:18 raspberrypi npm[2013]: Zigbee2MQTT:info  2022-02-04 19:41:18: MQTT publish: topic 'zigbee2mqtt/bathroom_radiator', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":23,"current_heating_setpoint":20,"eco_temperature":20,"force":"normal","holidays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C","linkquality":81,"local_temperature":27.5,"local_temperature_calibration":0,"max_temperature":35,"min_temperature":5,"position":0,**"preset":"boost"**,"system_mode":"auto","update":{"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C"}'

**Boost expires:**
Feb 04 19:46:43 raspberrypi npm[2013]: Zigbee2MQTT:info  2022-02-04 19:46:43: MQTT publish: topic 'zigbee2mqtt/bathroom_radiator', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":23,"current_heating_setpoint":20,"eco_temperature":20,"force":"normal","holidays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C","linkquality":60,"local_temperature":27.5,"local_temperature_calibration":0,"max_temperature":35,"min_temperature":5,"position":100,**"preset":"schedule"**,"system_mode":"auto","update":{"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C"}'

freakshock88, 2022-02-05:
I'm seeing the same behavior, it breaks all my automations.
Because it's to vital for my usecaseI tried to revert back to earlier versions of Zigbee2MQTT, however I see the same behavior in many of them.
I tried all (sub) versions between 1.23.0 and 1.21.0. 
I don't understand why this doesn't fix the issue since I was running one of these versions beforehand and did not see the issue then. 
Indeed, when I set boost mode while the previous mode was manual, the thermostat is set to schedule first for a split second, and then afterwards to boost. The problem is that this changes the setpoint to the scheduled setpoint, rather than keeping the setpoint set before. 

The same happens also when moving from boost -> manual.

moskovskiy82, 2022-02-12:
I confirm the same behavior. Returns to schedule instead of manual

lyczko, 2022-02-24:
Same here. Waiting for solving this issue

Enzokot, 2022-04-07:
Keep alive

lyczko, 2022-04-07:
still the same in 1.25

moskovskiy82, 2022-04-13:
Confirm

Enzokot, 2022-05-14:
Keep alive

Enzokot, 2022-06-15:
Keep alive

Enzokot, 2022-07-16:
Keep alive

Enzokot, 2022-07-31:
Seems it works with 1.26.0. Can anybody check?

Macter4, 2022-08-02:
> Seems it works with 1.26.0. Can anybody check?

It looks like it works fine for me too

closed 2022-09-09 by github-actions[bot]