Koenkk/zigbee2mqtt #11235: Bug sur modules PTVO après mise à jour Jeedom du plugin ZigbeeLinker (zigbee2mqtt) ce midi (04/02/2022)

medline2c, 2022-02-05:
### What happened?

Salut !
J’ai fais la mise à jour ce midi du plugin ZigbeeLinker sur Jeedom (zigbee2mqtt) et depuis, tous mes modules DIY réalisés avec [PTVO](https://www.zigbee2mqtt.io/devices/ptvo.switch.html) sont hs :sob: : historiques vide, ne répondent plus … étrangement (et heureusement à la fois), cela n’a semble-t-il pas affecté les autres modules zigbee « industriels » (SonOff…).

J’ai des erreurs SQL lorsque je veux par exemple faire une modification dans le module inclus dans Jeedom, impossible d'enregistrer :
`
Une commande action portant ce nom (Action) existe déjà pour cet équipement (logicalId=action___text / action)
`
du coup je ne peux rien modifier ni ajouter ou autre...

Egalement, le service client local MQTT plante : s’arrêt et redémarre dès que je touche à un de mes modules…
`
0026|[2022-02-04 22:44:59]ERROR : ##### Le service Client MQTT "Client MQTT Local" est arrêté #####

0027|[2022-02-04 22:47:00]INFO : ##### Démarrage du service Client MQTT (Client MQTT Local) #####
0028|[2022-02-04 22:47:00]INFO : ##### Le service Client MQTT a démarré (Client MQTT Local) #####
0029|[2022-02-04 22:47:00]INFO : [Aucun][Client MQTT Local] : Client MQTT subscribe : 1
0030|[2022-02-04 22:47:00]INFO : ##### Le client MQTT Client MQTT Local a reçu un premier message de Zigbee2MQTT #####
0031|[2022-02-04 22:47:01]INFO : ##### Démarrage du service Client MQTT (Client MQTT Local) terminé avec succès #####
0032|[2022-02-04 22:47:11]INFO : ##### Le service Client MQTT "Client MQTT Local" a démarré #####
`

J’ai supprimés les modules que je pensais défaillants : ils ont alors été recréés automatiquement avec un nouvel ID mais quand je veux ajouter mes fonctions pour récupérer mes infos, j’ai le message d’erreur ci-dessus…
`
[MySQL] Error code : 23000 (1062). Duplicate entry '257-Action' for key 'unique' : INSERT INTO 'cmd' SET 'id' = :id, 'logicalId' = :logicalId' .../...
`

Lorsque je vais dans l’interface Zigbee2mqtt, je parviens à interagir avec les modules : par exemple je vois leur changement d'état mais apparemment les infos ne remontent pas dans Jeedom car quand le demande dans Jeedom l'état d'une commande, je n'ai rien en retour (vide).

Impossible de faire un « Synchroniser » car toujours des erreurs…
`[MySQL] Error code : 23000 (1062). Duplicate entry '236-Action' for key 'unique' : INSERT INTO `cmd` SET `id` = :id, `logicalId` = :logicalId, `generic_type` = :generic_type, `eqType` = :eqType, `name` = :name, `order` = :order, `type` = :type, `subType` = :subType, `eqLogic_id` = :eqLogic_id, `isHistorized` = :isHistorized, `unite` = :unite, `configuration` = :configuration, `template` = :template, `display` = :display, `value` = :value, `isVisible` = :isVisible, `alert` = :alert`

J’ai tenté de vider le cache JSon et images : pas de changement si ce n’est que maintenant il me manque des vignettes des modules… (comment je répare ça ?..).
J’ai fais « Réparer Zigbee2mqtt » mais aucun changement …
J’ai relancé les dépendances… pas de changement
J’avoue ne plus trop savoir quoi essayer … des suggestions / idées ?.. Qu’est-ce que j’ai loupé ?..

Erreur dans le Log zigbee2mqtt :
`
0000|[2022-02-04 22:38:51]ERROR : [Aucun][0x00124b0007717155] : Une commande action portant ce nom (Action) existe déjà pour cet équipement (logicalId=action___text / action)

 0001|[2022-02-04 22:39:08]ERROR : Erreur sur zigbee2mqtt::deamon_client() : Call to a member function getId() on null
 0002|[2022-02-04 22:41:22]ERROR : Erreur sur zigbee2mqtt::deamon_client() : Call to a member function getId() on null 
 0003|[2022-02-04 22:43:20]ERROR : [Aucun][zb_TESTS LASER] : Une commande action portant ce nom (Action) existe déjà pour cet équipement (logicalId=action___text / action)
 0004|[2022-02-04 22:44:55]ERROR : [Aucun][zb_Routeur Fablab] : Une commande action portant ce nom (Action) existe déjà pour cet équipement (logicalId=action___text / action)
 0005|[2022-02-04 22:44:56]ERROR : Erreur sur zigbee2mqtt::deamon_client() : Call to a member function getId() on null`

Merci

Jeedom Core : 4.2.8 - ZigbeeLinker : 2022-02-03 21:47:27
Version Zigbee2MQTT
[1.23.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.23.0) commit: [6ea61bae](https://github.com/Koenkk/zigbee2mqtt/commit/6ea61bae)
Type de Coordinateur
ConBee2/RaspBee2
Révision du Coordinateur
0x26720700
Adresse IEEE du Coordinateur
0x00212effff07711b
Version Interface
0.6.71

### What did you expect to happen?

Avant la mise à jour, les modules PTVO remontaient bien leurs infos dans Jeedom alors que maintenant les informations sont bien reçues par zigbee2mqtt mais ne remontent pas dans jeedom.

### How to reproduce it (minimal and precise)

Tout modules actuels ou nouvellement inclus souffrent du défaut : on ne peut plus les modifier sinon erreur comme décrit ci-dessus, et les informations ne remontent plus dans jeedom

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

0000|[2022-02-04 22:38:51]ERROR : [Aucun][0x00124b0007717155] : Une commande action portant ce nom (Action) existe déjà pour cet équipement (logicalId=action___text / action)
0001|[2022-02-04 22:39:08]ERROR : Erreur sur zigbee2mqtt::deamon_client() : Call to a member function getId() on null
0002|[2022-02-04 22:41:22]ERROR : Erreur sur zigbee2mqtt::deamon_client() : Call to a member function getId() on null
0003|[2022-02-04 22:43:20]ERROR : [Aucun][zb_TESTS LASER] : Une commande action portant ce nom (Action) existe déjà pour cet équipement (logicalId=action___text / action)
0004|[2022-02-04 22:44:55]ERROR : [Aucun][zb_Routeur Fablab] : Une commande action portant ce nom (Action) existe déjà pour cet équipement (logicalId=action___text / action)
0005|[2022-02-04 22:44:56]ERROR : Erreur sur zigbee2mqtt::deamon_client() : Call to a member function getId() on null
0006|[2022-02-05 11:00:08]DEBUG : [TESTS][zb_TESTS LASER] : Ping de réveil de l'équipement
0007|[2022-02-05 11:00:08]DEBUG : [TESTS][zb_TESTS LASER] : Jeedom-->MQTT, topic=zigbee2mqtt/zb_TESTS LASER/get, json={"state_l1":""}
0008|[2022-02-05 11:00:33]INFO :  XXXXX Mise à jour des informations de l'équipement, nom=zb_IKEA OnOff, id=0x14b457fffed4295c XXXXX
0009|[2022-02-05 11:00:33]DEBUG : [Noémie][zb_IKEA OnOff] Refresh widget lancé

mickadam29, 2022-02-05:
Bonjour

Je débute depuis hier sur ZigbeeLinker et donc Zigbee2mqtt
Je confirme avoir des soucis identiques. Mais aucun recul de ma part.
La suppression de l'objet avec firmware PTVO supprime l'erreur lors de la synchronisation.

[image]

medline2c, 2022-02-05:
;-)
Oui, je confirme que ce sont les modules PTVO qui posent soucis désormais, suite à cette mise à jour sauf que progressivement je remplace tous mes modules par ceux-ci, plus intéressants pour mes usages car hautement personnalisables.
J'espère trouver une solution rapidement car c'est très "dangereux" ... si tout s'écroule à chaque mise à jour, je ne suis pas certain de rester sur cette technologie... ce serait fort regrettable... J'attends avec impatience une solution car en attendant mon système est bien planté...
=> la question que je me pose est à qui s'adresser pour régler cela ?... car tout comme toi je n'ai pas assez de recul car moi aussi je découvre mqtt _(et je trouve cela plutôt bien face aux autres technologies que j'avais testé jusqu'à présent...)._

medline2c, 2022-02-05:
Dans les logs de Zigbee2Mqtt, j'ai cette erreur aussi qui remonte constamment depuis (c'est aussi un module PTVO)...

Error Publish 'get' 'state' to 'zb_Inter CHB Parents' failed: 'Error: Read 0x00124b00089fd3f1/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUP_CLUSTER_COMMAND')'

mickadam29, 2022-02-05:
@medline2c 
Pour l'image, j'ai greenshot d'installé sur mon poste. Tu captures la zone de l'écran qui t'intéresse et ensuite c'est envoyé dans le presse-papiers. Y a plus qu'à coller dans Git.

Pour le souci PTVO, comme je n'ai pas de recul, difficile d'aider. Je fabrique mes cartes électroniques avec EasyEda et les modules Zigbee CC2530. Le firmware que j'y colle provient de PTVO. Par contre je ne peux pas conserver PTVO.INFO comme nom de modèle donc je vais devoir étudier comment créer les équipements. Il faudra que l'éditeur du plugin accepte d'intégrer ces devs étant qu'il s'agit de module DIY et non commercialisés (par définition). L'équipe Abeille a intégré mes modules: Capteur de niveau de fuel d'une cuve, détecteur de panne de courant/secteur, Sirène extérieure Zigbee, des cartes 8 E/S digitales mais aussi un projet avec un firmware UART.

Actuellement donc je teste Zigbee2mqtt et ZigbeeLinker. Il faut qu'il me soit facile d'intégrer mes modules DIY. Avec différents firmware personnalisés. Expliqué précédemment. Comme c'est sur du DEV chez moi, pas d'urgence de mon côté. Si tu dis que cela fonctionnait avant, c'est qu'un bug est apparu dans une version. Attention, Jeedom est passé en 4.2.8 et améliore la sécurité et les contraintes qui vont avec. Espérons que l'on te trouve une correction rapidement.

Bon courage à toi

mickadam29, 2022-02-05:
@medline2c 

Peux-tu me dire où l'on trouve la trame pour ouvrir des issues ici ? 
What happened, ....etc

medline2c, 2022-02-05:
> L'équipe Abeille a intégré mes modules : Capteur de niveau de carburant d'une cuve, détecteur de panne de courant/secteur, Sirène extérieure Zigbee, des cartes 8 E/S digitales mais aussi un projet avec un firmware UART.
> 
> Actuellement donc je teste Zigbee2mqtt et ZigbeeLinker

Cela m'intéresse bcp de te suivre dans cette démarche, peut-être ailleurs pour ne pas encombrer ce fil dont la thématique est notre recherche de solution à ce bug PTVO avec Zigbee2mqtt... ;-) La suite PTVO en MP si tu veux ? _(sauf que je ne sais pas comment te contacter => pas vu sur ton lien avatar)_

mickadam29, 2022-02-06:
@medline2c 

OK. Pour info j'ai réussi à intégrer mon module en suivant les docs disponibles. J'ai recompilé avec le dernier firmware. Version 1.6.0.2 du 14/01 en retirant le champ Location
J'ai demandé la création du fichier js et j'ai intégré ça dans configuration.yaml. Ca semble fonctionner et je n'ai plus d'erreur. D'aucune sorte. Je pense que la suite va consister à demander l'intégration de mon js dans un PR 
Il me reste à comprendre ce toggle sur des input digitales 2/3/4/5

[image]
[image]
[image]
[image]
[image]
[image]

mickadam29, 2022-02-06:
> > @medline2c
> > Peux-tu me dire où l'on trouve la trame pour ouvrir des issues ici ? Que s'est-il passé, ... etc.
> 
> Lorsque tu ouvres un nouveau numéro dans la catégorie "problèmes", tu as des champs à remplir _(la fameuse trame dont tu me parles j'imagines)_.

C'est tout à fait ça. Merci. Comme j'ai répondu je n'étais pas passé par là. Du coup j'ai également vu comment demandé l'intégration des mes dev. Merci

medline2c, 2022-02-06:
@mickadam29 

Merci pour ta réponse... Je vois que tu es plus avancé que moi en fait. Je vais essayer de rattraper mon retard pour tester la même chose chez moi.

Pourrais-tu m'éclairer :

- PTVO / Expert
-- Manufacturer ID : 49205
--- A quoi correspond ce numéro : tu l'as créé / récupéré quelque part / déclaré .../... ?
-- Manufacturer name
--- idem
-- Model ID
--- J'imagine que tu as donné un nom représentatif de la fonction de ton module
-- Enable watchdog timer
--- J'avoue ne pas encore avoir trop compris à quoi cela correspond au juste...

- sur ta capture où on voit les lignes 32 33 et 34
-- de où vient cette capture ? configuration.yaml ? car je n'ai pas la catégorie "external_conversters"... c'est toi qui l'a créée ?

- fichier fueltanklevel.js
-- j'imagine que c'est toi qui a créé ce ficier à la main et si oui, sur quoi t'es-tu basé ? sinon, de où ça vient ?...
-- ça m'intéresse d'autant plus que lorsque l'on importe un PTVO par défaut, il importe les state_1 à state_16 ce qui est inutile, et n'importe pas ceux que tu crées genre "i2" représentant une température vanat d'un capteur HT22 par exemple ou "Humidity" etc...

Q? Annexe : toi aussi tu modélises et imprimes tes propres boitiers en 3D pour les modules que tu crées ?

Encore merci pour tes partages... j'ai revérifié sur ton Github et ne ne trouve pas comment t'envoyer un MP, ni ton mail ...

mickadam29, 2022-02-06:
Salut.

En effet, il me semble que je connais mieux le sujet. 😉 pratique le Zigbee depuIs des années. Je fais fabriquer clés circuits imprimés et soude mes composants 
Pas de soucis. Je peux t’aider 
Pour le moment, comme je suis en dev, je viens de réinitialiser mon installation, raspberry et reinit de la ZiGate usb
Tes 16 entrées sont liées au fait que tu n’as pas de converter. Donc oui, très important d’avoir l’entrée dans configuration.yaml et le fichier nom_du_module.js généré par PTVO

mickadam29, 2022-02-06:
@medline2c 

Je viens de tout reprendre à 0
Donc des bases saines. Y compris la Zigate réinitialisée.
Tu peux me joindre sur mickaeladam1090@neuf.Fr Envoies moi un mail STP

mickadam29, 2022-02-07:
Salut

En effet je pratique le « Zigbee » depuis des années ! Mes modules DIY ont plus d’un an. Ils fonctionnent très bien avec le plugin Abeille sous Jeedom. Je teste donc une bascule vers zigbee2mqqt. Oui, ce que tu relèves vient de paramètres personnalisés dans PTVO. Je fais mes propres modules électroniques avec circuit imprimé et donc firmware personnalisé. Donc j’ai eu besoin de personnaliser configuration.yaml et donc external_converters (c’est expliqué dans la doc). Le fichier fueltanklevel.hs est généré depuis le générateur firmware PTVO. Il suffit d’appuyer sur le bouton et mettre le fichier généré dans le même répertoire que configuration.yaml.

closed 2022-03-18 by github-actions[bot]