Koenkk/zigbee2mqtt #11243: Failed test HomeAssistant extension

EXP-Carthage, 2022-02-05:
### What happened?

Test failure after the commit [`0a07b41`](https://github.com/Koenkk/zigbee-herdsman-converters/commit/0a07b418b4a13e5538dfbdc99040bfca6c2d5c8e).

[image]

### How to reproduce it (minimal and precise)

`npm test`

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2

Koenkk, 2022-02-06:
I cannot reproduce this locally with `npm tst`, also the CI on github actions succeeds so this looks to be something with your setup.

EXP-Carthage, 2022-02-06:
Yes, this may be due to my system configuration.
Even though I have done a clean installation, I still have the same problem. In this case, I will close the issue.
Thanks anyway for taking the time to test ;)

closed 2022-02-06 by EXP-Carthage