Koenkk/zigbee2mqtt #11245: [New device support]:

Char-r, 2022-02-05:
### Link

https://nodon.pro/en/produits/zigbee-pro-on-off-lighting-relay-switch/

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0x9035eafffec88e8a","nwkAddr":20522,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-2-20_PRO\u0000\u0014\u0000\u0000\u0000��\u0000 �\u0002\u0001\u0000�\u0000\u0000\u0000=","epList":[242,1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,7,8,4096,64599],"outClusterList":[3,6,25],"clusters":{"genBasic":{"attributes":{"hwVersion":50}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,7,8],"outClusterList":[3,6],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":36,"stackVersion":103,"hwVersion":50,"dateCode":"2020","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1644069584399,"useImplicitCheckin":true}

### Comments

I am trying to add the "Pro" version of the NodOn micromodule, they are identical to the existing modules so I just was planning on reusing the existing converter. The strange thing is with how the name appears:
[image]

I am not sure if this is an encoding issue within the module or how Z2M reads it? As a result eventhough I have created a converted with "PRO" in the name it is not being recognised. I also unable to copy and paste those characters within the converter file.

any suggestions on the best way to troubleshoot this?

thanks

### External converter

```shell
const exposes = require('../lib/exposes');
const reporting = require('../lib/reporting');
const extend = require('../lib/extend');
const e = exposes.presets;
const tz = require('../converters/toZigbee');
const fz = require('../converters/fromZigbee');

module.exports = [
    {
        zigbeeModel: ['SIN-4-RS-20_PRO'],
        model: 'SIN-4-RS-20',
        vendor: 'NodOn',
        description: 'Roller shutter controller',
        fromZigbee: [fz.cover_position_tilt],
        toZigbee: [tz.cover_state, tz.cover_position_tilt],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.currentPositionLiftPercentage(endpoint);
            await reporting.currentPositionTiltPercentage(endpoint);
        },
        exposes: [e.cover_position()],
    },
    {
        zigbeeModel: ['SIN-4-1-20_PRO'],
        model: 'SIN-4-1-20',
        vendor: 'NodOn',
        description: 'Single LED relay',
        extend: extend.switch(),
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep = device.getEndpoint(1);
            await reporting.bind(ep, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep);
        },
    },
    {
        zigbeeModel: ['SIN-4-2-20_PRO'],
        model: 'SIN-4-2-20',
        vendor: 'NodOn',
        description: 'Double LED relay',
        exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
        extend: extend.switch(),
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep1 = device.getEndpoint(1);
            const ep2 = device.getEndpoint(2);
            await reporting.bind(ep1, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(ep2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep1);
            await reporting.onOff(ep2);
        },
    },
    {
        zigbeeModel: ['SIN-4-2-20_PRO'],
        model: 'SIN-4-2-20',
        vendor: 'NodOn',
        description: 'Double LED relay',
        exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
        extend: extend.switch(),
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep1 = device.getEndpoint(1);
            const ep2 = device.getEndpoint(2);
            await reporting.bind(ep1, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(ep2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep1);
            await reporting.onOff(ep2);
        },
    },
];
```

sjorge, 2022-02-05:
Unfortunately those null characters and what I'm guessing is some weird non printable utf8 characters are part of the model name.

@Koenkk I wonder if it might be a good idea to have some sort of check and if something outside of the standard ascii set is found to base64 encode it. It would make copying and dealing with them easier. Although that would make it harder to figure out as a human to read the model. 🤔

Char-r, 2022-02-05:
> Unfortunately those null characters and what I'm guessing is some weird non printable utf8 characters are part of the model name. So they need to somehow be copied into the external converter.
> 
> @Koenkk I wonder if it might be a good idea to have some sort of check and if something outside of the standard ascii set is found to base64 encode it. It would make copying and dealing with them easier. Although that would make it harder to figure out as a human to read the model. 🤔

Should we be asking NodOn to change the model name? I don't have the non pro version to compare so not sure if this is specific to this model.

sjorge, 2022-02-05:
> Should we be asking NodOn to change the model name? I don't have the non pro version to compare so not sure if this is specific to this model.

You could try, but I wouldn't get your hopes up. There are some other devices where the module is padded with null characters, don't think I've seen some with other non-printable stuff like this yet though.

Char-r, 2022-02-05:
This is the device signature in ZHA, as you can see we do not pickup extra characters in the name:

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4747, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0007",
        "0x0008",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0007",
        "0x0008"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0066",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "NodOn",
  "model": "SIN-4-2-20_PRO",
  "class": "zigpy.device.Device"
}`

Koenkk, 2022-02-06:
Do you maybe have a sniffer so we can see what the device send when pairing it? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

If yes, sniff the traffic while pairing the device to z2m and share the pacpng file her.

Char-r, 2022-02-06:
@Koenkk any sniffing solutions that would work with CC2652P devices? seems that Zboss only functions on CC2531 :/

Koenkk, 2022-02-06:
@Char-r wireshark + CC2652 with sniffer firmware should work: https://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/3.20.00.68/exports/docs/zigbee/html/zigbee/packet_sniffer.html

closed 2022-03-16 by github-actions[bot]

disenator, 2022-06-12:
The non-commercial version also has this problem, any solution?
[image]

Char-r, 2023-01-12:
So, I seemed to have randomly solved the naming issue, but I am not sure how this would affect the non pro version. I never was able to install the sniffing firmware (bricked a couple devices in the process)

in my custom external converter I added _PRO in the model name in zigbeeModel: and now it is fine, I guess since SIN-4-2-20 already existed it added some junk characters after it to differentiate?

        zigbeeModel: ['SIN-4-2-20_PRO'],
        model: 'SIN-4-2-20_PRO',

I will duplicate the existing code in the converter to add the PRO model

Char-r, 2023-01-23:
ok, adding on, when I test the edge version with the new Pro version code which is identical to my local converter, the error comes back, 

Zigbee2MQTT:warn  2023-01-23 13:36:00: Received message from unsupported device with Zigbee model 'SIN-4-2-20_PRO

whereas master version with local convert now displays the name correctly. I am a little dumbfounded on that one.