Koenkk/zigbee2mqtt #11251: [New device support]: Zemismart Curtain Robot motor ZM85EL-2Z

little-dgek, 2022-02-05:
### Link

https://www.zemismart.com/products/zm85el-2z

### Database entry

{ 	"id": 8, 	"type": "EndDevice", 	"ieeeAddr": "0x94deb8fffe25058b", 	"nwkAddr": 16394, 	"manufId": 4098, 	"manufName": "_TZE200_cf1sl3tj", 	"powerSource": "Battery", 	"modelId": "TS0601", 	"epList": [1], 	"endpoints": { 		"1": { 			"profId": 260, 			"epId": 1, 			"devId": 81, 			"inClusterList": [0, 4, 5, 61184], 			"outClusterList": [25, 10], 			"clusters": { 				"genBasic": { 					"attributes": { 						"modelId": "TS0601", 						"manufacturerName": "_TZE200_cf1sl3tj", 						"powerSource": 3, 						"zclVersion": 3, 						"appVersion": 72, 						"stackVersion": 0, 						"hwVersion": 1, 						"dateCode": "" 					} 				} 			}, 			"binds": [{ 				"cluster": 0, 				"type": "endpoint", 				"deviceIeeeAddress": "0x00124b0022812c80", 				"endpointID": 1 			}, { 				"cluster": 258, 				"type": "endpoint", 				"deviceIeeeAddress": "0x00124b0022812c80", 				"endpointID": 1 			}], 			"configuredReportings": [], 			"meta": {} 		} 	}, 	"appVersion": 72, 	"stackVersion": 0, 	"hwVersion": 1, 	"dateCode": "", 	"zclVersion": 3, 	"interviewCompleted": true, 	"meta": { 		"configured": -1267545107 	}, 	"lastSeen": 1644093139243, 	"defaultSendRequestWhen": "immediate" }

### Comments

Hello. Please, help me. 
I tried to follow guide, and almost win)

I used Tuya model (TS0601_cover) and I have only one warning: 

_TuYa_cover_control: Unhandled DP #13 for _TZE200_cf1sl3tj: {"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,95]}}_

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_cf1sl3tj'
        },
    ],
    model: 'ZM85EL-2Z',
    vendor: 'Zemismart',
    description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
    supports: 'curtains, blinder',

    fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
    exposes:[ e.cover_position().setAccess('position', ea.STATE_SET),
              exposes.composite('options', 'options')
                .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0)
                    .withValueMax(255)
                    .withDescription('Motor speed'))],
    meta: {multiEndpoint: true},

    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
    },
};

module.exports = definition;
```

D-side, 2022-02-21:
Hi!
I have the same device and intend to get it to work.
[Zemismart are apparently nice enough to provide protocol documentation upon request](https://github.com/Koenkk/zigbee-herdsman-converters/issues/1159#issuecomment-614659802), maybe we could try that method, it would save some time.

Here's the converter file I have so far:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_cf1sl3tj'}],
    model: 'ZM85EL-2Z',
    vendor: 'Zemismart',
    description: 'Zigbee/RF curtain motor',
    fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tz.tuya_cover_control],
    exposes: [
        exposes.cover(),
    ],
};

module.exports = definition;
```
I started from the same `TS0601_cover` template as you, but stripped the parts I wasn't certain about, including the cover position you seem to be getting an error message about, might help you as well.

**I can confirm that with this converter the device properly functions in Home Assistant with ⬆⏹⬇ buttons, which is enough for me to do some basic automation. 🎉** Though I do still observe the warning about DP#13.

Things we can also hopefully find:
- Battery state (could very well be that DP#13, but I've yet to deplete the battery significantly to observe whether the values change; so far I've seen the last byte be 80, 85 and 90, but not 95 like you are; battery voltage can fluctuate somewhat I guess)
- Limit calibration (though I recall seeing a comment that it may be exclusive to the RF remote, so hopes are low here)

D-side, 2022-03-24:
Still valid, haven't been able to get on this so far, but the converter I've provided so far has been working with no issues.
Not that there's a lot to have issues with in it, given how stripped-down it is… 😁

nchieng, 2022-03-24:
Thanks for the work on this.
Just wanting to shout out that this PR is the only thing stopping me from ordering the device.
Would love to see a working converter make it into a PR.

brunoap, 2022-04-05:
It is working, hence there are some errors on the logs, which i turned to info only. Thanks for the work on this.

christiangenco, 2022-04-16:
This was the first extension I've installed in zigbee2mqtt. It was kinda tricky for me to figure out where the converters go but I got it working!

[image]

I was able to get `OPEN`, `CLOSE`, and `STOP` buttons on the web interface by following these steps:

1. Copy @D-side's converter code above. @little-dgek's code gave me extra options for `position` and `motor speed` but I can't figure out how to get those options to do anything.
2. Paste that code in a new file called `TS0601.js` in your `data` directory (the same directory as `configuration.yaml`. For me the full path of that file is `/opt/zigbee2mqtt/data/TS0601.js`.
3. Add these two lines to `configuration.yaml`:

```
external_converters: 
  - TS0601.js
```

4. Restart zigbee2mqtt. I'm running zigbee2mqtt as a service so I restart it with `sudo systemctl restart zigbee2mqtt.service`.

Every time I send a command I'm seeing errors that look like `Warning TuYa_cover_control: Unhandled DP #13 for _TZE200_cf1sl3tj: {"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,50]}}` but at least the motor is running!

nchieng, 2022-04-19:
I would love to be able to figure out how to set the limits via MQTT. Which Datapoint and what values they are.

Does anyone have this device and a Zemismart hub, setup with Tuya?
Could anyone find the options => data point mapping?

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_8-display-device-logs

Oliviakrkk, 2022-04-22:
I will try to find it. I was able to setup developer account:
[image]

Oliviakrkk, 2022-04-22:
Current curtain position: code 3
Motor direction: code 5
Border: 16
Control 1
Curtain position setting 2
Best position 19
Mode 4
Click control 20
Motor direction 5
Work state 7
Situation set 17
Fult 12
Battery 13

@nchieng Will this info suffice?

D-side, 2022-04-22:
@Oliviakrkk oh, so 13 **is** the battery! That's very helpful to know.
In addition to the numbers of the data points I reckon we would also need the **data types** for each to be able to use them. For the battery it's obvious, others can be guessed given their purposes, but more reliable info would accelerate the process significantly.

Oliviakrkk, 2022-04-22:
I have issues with finding more info...In the request response there is a field:
```
eventDetail
```

And when it comes to battery it just shows the value...
```
{"result":{"devId":"XXX","datas":[{"devId":"XXX","eventType":"Report","sourceDetail":"","eventTime":1650626326586,"eventName":"Battery Percentage","eventTimeStr":"2022-04-22 11:18:46:586","eventDetail":"85%","requestFrom":"device itself"}],"docType":1,"currentPageStartRowKey":"XXX=="},"t":1650628858171,"success":true,"status":"ok"}
```

Is that what we need?

Oliviakrkk, 2022-04-22:
I also found some response  called status:
```
{"result":[{"code":"control","value":"open"},{"code":"percent_control","value":100},{"code":"percent_state","value":100},{"code":"mode","value":"morning"},{"code":"control_back_mode","value":"forward"},{"code":"work_state","value":"opening"},{"code":"situation_set","value":"fully_open"},{"code":"fault","value":0},{"code":"battery_percentage","value":85},{"code":"border","value":"up"},{"code":"position_best","value":0},{"code":"click_control","value":"up"}],"success":true,"t":1650629290063,"tid":"e05ac7e8c23411ec86cb4e8aeb4f13b1"}
```

This may be usefull. 

But still no field `data_type`...

D-side, 2022-04-22:
Most likely because the interface you're dealing with converts these types to more "semantically pure" (as in "without implementation details") values under the hood, without disclosing what came in originally, and zigbee2mqtt needs to do the same.

There's this bit in the original message that I observed with my device as well:

```json
{"dp":13,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,95]}}
```
Which likely means DP#13 is of type `value` (terminology [from here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/tuya.js#L6-L13)). I'm not sure if `2` beside this type stands for anything or it's internal to this project only.

Oliviakrkk, 2022-04-22:
To get this info from Toya I would need to sniff the zigbee network, right? And check payload in the wireshark...
Currently I do not have a sniffer :/

D-side, 2022-04-22:
And I only just now opened the guide above to realize this is Tuya's own interface; it doing this kind of conversion on its end makes sense. Yeah, I'm new to this. 😅

@Oliviakrkk not necessarily Wireshark, zigbee2mqtt has some built-in facilities. This would seem to be the relevant guide: https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_4-deciphering-the-data-points
You've already made a major step forward already by mapping data points to their respective **purposes**. What's left may still be guesswork, but within a much more manageable space.

I'll try to allocate some time this or next week to wrap my head around introducing mappings for more DPs, but totally wouldn't mind anyone beating me to it 😉

D-side, 2022-04-22:
> **little-dgek**'s code gave me extra options for position and motor speed but I can't figure out how to get those options to do anything.

@christiangenco I believe they were copied from another device which uses a different set of data points; but enough of it worked (motor controls, the main part) to seem like it fits overall, even if some parts don't.
I started the same way, but _just in case_ I removed the bits that didn't seem to have an effect to avoid unnecessarily randomizing the behavior of the device and misinforming in the UI while we learn what the rest of the protocol means.

rodrigogbs, 2022-05-20:
Hey guys, can you share the external converter that shows up the battery percentage?

D-side, 2022-06-21:
Alright, alright! Not stale, just very slow! Cheesus 😅

@rodrigogbs I went in expecting an easy win, but left with a slightly better understanding how the project works and yet another reminder to **check the versions**, as I was often referencing the source code of the `master` branches rather than the versions I currently use (hassio addon, ZB2MQTT 1.18.1, with converters v14.0.74).

* Battery charge is reported from DP 13, as per the warning at the start of the thread — but it took me a while to figure out that it's only reported when I push buttons on the remote, how often it reports the charge outside of that isn't clear to me
* I copied the existing `fz.tuya_cover` converter and truncated it, because the "Unhandled DP" thing apparently reports *everything* unhandled, even if it *is* actually handled in another converter (maybe I don't understand how it works?)
* In my testing `position` hasn't reported anything other than a zero, so I removed handling of that field and dependent on it `state` for now — maybe if I calibrate the motor they'll start working, but I have a hardware problem there, I'm running the motor on a very jury-rigged mounting mechanism that doesn't provide quite enough friction for it to reliably move along the rail so I don't expect any numbers there to be reliable; but if you observe meaningful values in `position` do let me know and I'll return that bit 🙂
* Still have to go through @Oliviakrkk's [list of datapoints](https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1106434392) to make sure all capabilities of the device are covered

Here's what the converter looks like now:

```js
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzDeviceSpecific = {
    cluster: 'manuSpecificTuya',
    type: ['commandSetDataResponse', 'commandGetData'],
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dp;
        const value = tuya.getDataValue(msg.data.datatype, msg.data.data);

        switch (dp) {
        case tuya.dataPoints.state: // Confirm opening/closing/stopping (triggered from Zigbee)
        case tuya.dataPoints.coverPosition: // Started moving to position (triggered from Zigbee)
        case tuya.dataPoints.coverChange: // Started moving (triggered by transmitter or pulling on curtain)
        case tuya.dataPoints.coverArrived: { // Arrived at position
            const running = dp === tuya.dataPoints.coverArrived ? false : true;
            return {running}; // Position does not seem to be working, so respective code removed
        }
        case tuya.dataPoints.config: // Returned by configuration set; ignore
            break;
        case 13: // Battery; apparently only sent when triggered via hardware remote?
            return {battery: value};
        default: // Unknown code
            meta.logger.warn(`ZM85EL-2Z: Unhandled DP #${dp}: ${JSON.stringify(msg.data)}`);
        }
    },
}

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_cf1sl3tj'}],
    model: 'ZM85EL-2Z',
    vendor: 'Zemismart',
    description: 'Zigbee/RF curtain motor',
    fromZigbee: [fzDeviceSpecific],
    toZigbee: [tz.tuya_cover_control],
    exposes: [
        exposes.cover(),
        exposes.presets.battery(),
    ],
    meta: {
        configureKey: 1,
        battery: {dontDividePercentage: true}
    },
};

module.exports = definition;
```

rodrigogbs, 2022-06-22:
> Alright, alright! Not stale, just very slow! Cheesus 😅
>
> @rodrigogbs I went in expecting an easy win, but left with a slightly better understanding how the project works and yet another reminder to [...]

Tks @D-side, amazing job!

I tried and it is working pretty well, despite the fact that the battery percentage is still not appearing... :(

The status behavior is working very well !!

About the position value (you're right, to have this is necessary to set limits with a remote control), but in this version even with limits set it is not appearing. This information is important to know if the cover is totally or partially opened, I use this in my automations, is it possible to have back?

D-side, 2022-06-22:
@rodrigogbs did the position work in my initial version above? If so, returning some of the code I removed after copying [`fz.tuya_cover` from 14.0.74](https://github.com/Koenkk/zigbee-herdsman-converters/blob/v14.0.74/converters/fromZigbee.js#L1533-L1569) should make it work. I'll get to it eventually, but it might take a while.

As I mentioned, the battery percentage right now only shows up after pushing a button on the remote, e. g. ⏹ stop. I guess to get it to show up more reliably I have to send a certain command to it during initialization, which right now is slightly beyond my level of familiarity with this codebase.

rodrigogbs, 2022-06-22:
> @rodrigogbs did the position work in my initial version above? If so, returning some of the code I removed after copying [`fz.tuya_cover` from [...]

@D-side yes, the position was working in the initial version, I will try to return the partial code... 

About the battery, even pushing some remote buttons, it didn't work...at least for me...

smarthomejunkie, 2022-06-27:
> Tks @D-side, amazing job!
>
> I tried and it is working pretty well, despite the fact that the battery percentage is still not appearing... :(
>
> The status behavior is working very well !!
>
> About the [...]

So,
I got this Curtain Motor sent to me by Zemismart and I am trying to set it up as well. First followed the instructions on the MQTT site to add a new product, but I failed (The first time for me, so lots to learn). Tried to set it up in Home Assistant, and ended up setting up Z2M on a Windows 10 machine to get more control over all the files. That runs well. Z2M detects commands when I push the buttons on the Curtain Motor.

[image]

Now I found this thread here and added the following lines to the configuration.yaml of Zigbee2MQTT:

`
external_converters: 
  - TS0601.js
`

Next to that I created a file TS0601.js with the code above and put it in the same folder as the configuration.yaml
Then I rebooted Z2M, but the device still shows that it is unsupported and it exposes nothing (I get the message: Empty exposes definition).

I am really new to this and try to understand what I'm doing wrong, but I'm in the dark here. Did I miss a step?

[image]

[image]

smarthomejunkie, 2022-06-27:
Thank you for this. I've got it to work thanks to you. In the beginning it did not work, because my Manufacturer Name is: _TZE200_nw1r9hp6

Which is different than your name. When I changed it to the Manufacturer name that I found in database.db it worked immediately. Only the battery level is n/a for some reason. I will investigate this further to see if I can find a solution for this. Is it possible to add this curtain robot to the official Z2M supported devices?

closed 2022-08-06 by github-actions[bot]

DariBer, 2022-09-02:
> Thank you for this. I've got it to work thanks to you. In the beginning it did not work, because my Manufacturer Name is: _TZE200_nw1r9hp6
>
> Which is different than your name. When I changed it to [...]

@smarthomejunkie I got my Zemismart, ZM25RX-0.8/30 _TZE200_7eue9vhc, to work when I did as you did, but I have the same problem as you, the battery level is not working. Did you get your to work?

Thanx for your help.

mrammal, 2022-09-27:
@DariBer question for you, does the ZM25RX-0.8/30 have soft stop functionality? My existing non-zigbee motors reduce speed slightly when it comes close to a stop.

DariBer, 2022-09-27:
> @DariBer question for you, does the ZM25RX-0.8/30 have soft stop functionality? My existing non-zigbee motors reduce speed slightly when it comes close to a stop.

I haven't noticed anything like that on non of my motor rollers. It use one speed and then just stop. So no reduced speed.

jumpinf00l, 2022-09-27:
Just stumbled upon this tonight. I've fiddled with this a bit with some inspiration from the built-in converters - I've gotten DP#13 working as the battery percentage. It seems like this isn't calculated very accurately or at least mine seems to vary quite wildly (as low as 0% sometimes). Maybe we can average this value out? Position and state seem to work nicely as well now.

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const fzLocal = {
    device_specific: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'commandDataResponse'],
        options: [exposes.options.invert_cover()],
        convert: (model, msg, publish, options, meta) => {
            // Protocol description
            // https://github.com/Koenkk/zigbee-herdsman-converters/issues/1159#issuecomment-614659802

            const dpValue = tuya.firstDpValue(msg, meta, 'tuya_cover');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);

            switch (dp) {
            case tuya.dataPoints.coverPosition: // Started moving to position (triggered from Zigbee)
            case tuya.dataPoints.coverArrived: { // Arrived at position
                const invert = tuya.isCoverInverted(meta.device.manufacturerName) ? !options.invert_cover : options.invert_cover;
                const position = invert ? 100 - (value & 0xFF) : (value & 0xFF);
                const running = dp !== tuya.dataPoints.coverArrived;

                // Not all covers report coverArrived, so set running to false if device doesn't report position for a few seconds
                clearTimeout(globalStore.getValue(msg.endpoint, 'running_timer'));
                if (running) {
                    const timer = setTimeout(() => publish({running: false}), 3 * 1000);
                    globalStore.putValue(msg.endpoint, 'running_timer', timer);
                }

                if (position > 0 && position <= 100) {
                    return {running, position, state: 'OPEN'};
                } else if (position == 0) { // Report fully closed
                    return {running, position, state: 'CLOSE'};
                } else {
                    return {running}; // Not calibrated yet, no position is available
                }
            }
            case tuya.dataPoints.coverSpeed: // Cover is reporting its current speed setting
                return {motor_speed: value};
            case tuya.dataPoints.state: // Ignore the cover state, it's not reliable between different covers!
            case tuya.dataPoints.coverChange: // Ignore manual cover change, it's not reliable between different covers!
                break;
            case tuya.dataPoints.config: // Returned by configuration set; ignore
                break;
            case 13: // Battery; apparently only sent when triggered via hardware remote?
                return {battery: value};
            default: // Unknown code
                meta.logger.warn(`TuYa_cover_control: Unhandled DP #${dp} for ${meta.device.manufacturerName}:
                ${JSON.stringify(dpValue)}`);
            }
        },
    },
};

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_68nvbio9'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_cf1sl3tj'},
    ],
    model: 'TS0601_cover',
    vendor: 'TuYa',
    description: 'Zigbee/RF curtain motor',
    fromZigbee: [fzLocal.device_specific, fz.ignore_basic_report],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET), e.battery()],
    meta: {
        configureKey: 1,
        battery: {dontDividePercentage: true}
    },
};

module.exports = definition;```

gbeleris, 2022-10-04:
Hey, do you think this will be integrated in the next version? 

I tried to set up an external converter, but even though it seems quite straight forward I got a bunch of errors and Z2M was not starting. I am running Z2M on a seperate host from HA. 

Thanks!

gbeleris, 2022-10-17:
@Koenkk @jumpinf00l is it possible to have this converter integrated in the next version please?

Hamper, 2022-10-18:
in https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1259521499 converter code must be line
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
```
because it used in code.

jumpinf00l, 2022-10-20:
Oops you're right @Hamper, I'd missed that line when copying+pasting. @gbeleris try this:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const fzLocal = {
    device_specific: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'commandDataResponse'],
        options: [exposes.options.invert_cover()],
        convert: (model, msg, publish, options, meta) => {

            const dpValue = tuya.firstDpValue(msg, meta, 'tuya_cover');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);

            switch (dp) {
            case tuya.dataPoints.coverPosition: // Started moving to position (triggered from Zigbee)
            case tuya.dataPoints.coverArrived: { // Arrived at position
                const invert = tuya.isCoverInverted(meta.device.manufacturerName) ? !options.invert_cover : options.invert_cover;
                const position = invert ? 100 - (value & 0xFF) : (value & 0xFF);
                const running = dp !== tuya.dataPoints.coverArrived;
                // Not all covers report coverArrived, so set running to false if device doesn't report position for a few seconds
                clearTimeout(globalStore.getValue(msg.endpoint, 'running_timer'));
                if (running) {
                    const timer = setTimeout(() => publish({running: false}), 3 * 1000);
                    globalStore.putValue(msg.endpoint, 'running_timer', timer);
                }
                if (position > 0 && position <= 100) {
                    return {running, position, state: 'OPEN'};
                } else if (position == 0) { // Report fully closed
                    return {running, position, state: 'CLOSE'};
                } else {
                    return {running}; // Not calibrated yet, no position is available
                }
            }
            case tuya.dataPoints.coverSpeed: // Cover is reporting its current speed setting
                return {motor_speed: value};
            case tuya.dataPoints.state:
            case tuya.dataPoints.coverChange: // Ignore manual cover change, it's not reliable between different covers!
                break;
            case tuya.dataPoints.config: // Returned by configuration set; ignore
                break;
            case 13: // Battery
                return {battery: value};
            default: // Unknown code
                meta.logger.warn(`TuYa_cover_control: Unhandled DP #${dp} for ${meta.device.manufacturerName}:
                ${JSON.stringify(dpValue)}`);
            }
        },
    },
};

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_68nvbio9'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_cf1sl3tj'},
    ],
    model: 'TS0601_cover',
    vendor: 'TuYa',
    description: 'Zigbee/RF curtain motor',
    fromZigbee: [fzLocal.device_specific],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET), e.battery()],
    meta: {
        configureKey: 1,
        battery: {dontDividePercentage: true}
    },
};

module.exports = definition;
```

With that in mind, as @Oliviakrkk pointed out there are several more potential cases which this converter could handle

> Current curtain position: code 3 Motor direction: code 5 Border: 16 Control 1 Curtain position setting 2 Best position 19 Mode 4 Click control 20 Motor direction 5 Work state 7 Situation set 17 Fult 12 Battery 13
> 
> nchieng Will this info suffice?

I'm just not entirely sure how to convert these without sinking a whole bunch of time into it unfortunately

simonwood0609, 2023-01-07:
> you're right @Hamper, I'd missed that line when copying+pasting. @gbeleris try this:

I've been trying to get my TZE200_pw7mji0l (0x84ba20fffe3a2afb) connected for days without success.

Zigbee2MQTT is running on a separate physical machine (Raspberry Pi) than Hass, currently have over 20 devices connected and have some experience with pairing so it should be straight forward but isn't for some reason.

Constantly seeing the same Unhandled DP #13 error, but more problematic I cannot get the device to appear in Home Assistant (and I do not know why).  Could somebody help a dummy by providing the actual list of steps?
1. I can see the device in Zigbee2MQTT - when I press buttons on the blind motor it reports to Zigbee2MQTT.
2. I cannot see the device in Home Assistant.  I've ensured that my MQTT Broker has "Enabled newly added entities." switched on (and with this enabled in past, the newly paired device instantly appears).

Kinda stuck not sure what to check next.

jumpinf00l, 2023-01-09:
@simonwood0609 here's what I did:

1. Copy the code from my [post above](https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1285207519) to config/zigbee2mqtt/zigbee_blind_motor.js
2. Duplicate line 71
3. Replace the manufacturerName from your duplicated line with your specific blind motor's manufacturerName (ie 'TZE200_pw7mji0l')
[image]
4. Open config/zigbee2mqtt/configuration.yaml
5. Locate or add the external_converters: section, then add the name (not path) of the .js file from step 1
`external_converters:`
`  - zigbee_blind_motor.js`
6. Reload Zigbee2MQTT add-on, and check in the log tab to make sure everything is happy

From @Oliviakrkk's post, we can see that DP #13 is the battery level but as I mentioned it's reported every time the blind does something and can vary quite wildly which I assume is due to the draw on the battery. The code above interprets DP #13 as the battery level which should reduce the number of 'unhandled' messages, though some do still appear (there are a few in @Oliviakrkk's post) such as when setting the top/bottom positions which would also be nice to be able to set over Zigbee rather than the RF remote. I'd prefer to poll the blind for battery rather than grab it as the blind is doing something, but honestly haven't touched this since my last post

simonwood0609, 2023-01-10:
@jumpinf00l thanks so much for the ELI5 - very helpful!

I ended up messing up my instance of Zigbee2MQTT somehow (probably trying to edit far too many js files) so I **rmdir** and started from scratch github clone.  This time the blind paired instantly - I didn't even need to create the external_converter which is strange.  I'm glad I can now control the blind via _Zigbee2MQTT_.

Unfortunately no new devices appearing in Home Assistant (in the MQTT: Mosquitto broker device/entity lists).  I have "Enable newly added entities." checked, and I can't find the device anywhere to delete it (in hope it'll re-create properly).

jumpinf00l, 2023-01-10:
@simonwood0609 glad to see you got it sorted. The good old ‘turn it off and on again’ saves the day

ouchkilljoys, 2023-03-06:
> @simonwood0609 here's what I did:
> 
> 1. Copy the code from my [post above](https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1285207519) to config/zigbee2mqtt/zigbee_blind_motor.js
> 2. Duplicate line 63
> 3. Replace the manufacturerName from your duplicated line with your specific blind motor's manufacturerName (ie 'TZE200_pw7mji0l')
>    [image]
> 4. Open config/zigbee2mqtt/configuration.yaml
> 5. Locate or add the external_converters: section, then add the name (not path) of the .js file from step 1
>    `external_converters:`
>    `  - zigbee_blind_motor.js`
> 6. Reload Zigbee2MQTT add-on, and check in the log tab to make sure everything is happy
> 
> From @Oliviakrkk's post, we can see that DP #13 is the battery level but as I mentioned it's reported every time the blind does something and can vary quite wildly which I assume is due to the draw on the battery. The code above interprets DP #13 as the battery level which should reduce the number of 'unhandled' messages, though some do still appear (there are a few in @Oliviakrkk's post) such as when setting the top/bottom positions which would also be nice to be able to set over Zigbee rather than the RF remote. I'd prefer to poll the blind for battery rather than grab it as the blind is doing something, but honestly haven't touched this since my last post

I always assumed that Z2M polled battery devices every so often, but I guess the challenge here is that the battery data needs to be discarded when the blind is in motion?

Cjkeenan, 2023-03-15:
Has anyone run into intermittent "connection" issues with this external converter? It never actually leaves the mesh or reports as offline, it just stops responding to commands though. It seems to work for a bit, then you give it some time and they stop responding to commands. After I tried “waking” them up by reading something simple like time from the dev console, they start responding again, but again after some time they stop again.

I am wondering if it is something as simple as I need to setup a reporting method to keep it alive, but as I have never messed with external converters I do not know. Their LQI's also hover around 100 which doesn't seem great, but also not terrible.

jumpinf00l, 2023-04-25:
> Has anyone got the instructions for this. Can't find any online?

Sure thing, sorry the original got a bit crumpled

[CCO_20230425_200028_0001.pdf](https://github.com/Koenkk/zigbee2mqtt/files/11322260/CCO_20230425_200028_0001.pdf)

cloudbr34k84, 2023-04-25:
> > Has anyone got the instructions for this. Can't find any online?
> 
> Sure thing, sorry the original got a bit crumpled
> 
> [CCO_20230425_200028_0001.pdf](https://github.com/Koenkk/zigbee2mqtt/files/11322260/CCO_20230425_200028_0001.pdf)
> 

Legend!  I can finally re add it lol

ianfromnyc, 2023-06-03:
Looks like there is official support for ZM85EL-2Z in 1.31.0 but not the _TZE200_nw1r9hp6 variant.  Can this be added?

Koenkk, 2023-06-03:
@ianfromnyc added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

esackbauer, 2023-09-05:
Battery status is not working for _TZE200_nw1r9hp6 model.

I used an external converter which I tweaked from another device which properly showed the battery status.
Please someone have a look at it, this was my external converter:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const fzLocal = {
    device_specific: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'commandDataResponse'],
        options: [exposes.options.invert_cover()],
        convert: (model, msg, publish, options, meta) => {

            const dpValue = tuya.firstDpValue(msg, meta, 'tuya_cover');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);

            switch (dp) {
            case tuya.dataPoints.coverPosition: // Started moving to position (triggered from Zigbee)
            case tuya.dataPoints.coverArrived: { // Arrived at position
                const invert = tuya.isCoverInverted(meta.device.manufacturerName) ? !options.invert_cover : options.invert_cover;
                const position = invert ? 100 - (value & 0xFF) : (value & 0xFF);
                const running = dp !== tuya.dataPoints.coverArrived;
                // Not all covers report coverArrived, so set running to false if device doesn't report position for a few seconds
                clearTimeout(globalStore.getValue(msg.endpoint, 'running_timer'));
                if (running) {
                    const timer = setTimeout(() => publish({running: false}), 3 * 1000);
                    globalStore.putValue(msg.endpoint, 'running_timer', timer);
                }
                if (position > 0 && position <= 100) {
                    return {running, position, state: 'OPEN'};
                } else if (position == 0) { // Report fully closed
                    return {running, position, state: 'CLOSE'};
                } else {
                    return {running}; // Not calibrated yet, no position is available
                }
            }
            case tuya.dataPoints.coverSpeed: // Cover is reporting its current speed setting
                return {motor_speed: value};
            case tuya.dataPoints.state:
            case tuya.dataPoints.coverChange: // Ignore manual cover change, it's not reliable between different covers!
                break;
            case tuya.dataPoints.config: // Returned by configuration set; ignore
                break;
            case 13: // Battery
                return {battery: value};
            default: // Unknown code
                meta.logger.warn(`TuYa_cover_control: Unhandled DP #${dp} for ${meta.device.manufacturerName}:
                ${JSON.stringify(dpValue)}`);
            }
        },
    },
};

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_68nvbio9'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_nw1r9hp6'},
    ],
    model: 'TS0601_cover',
    vendor: 'TuYa',
    description: 'Zigbee/RF curtain motor',
    fromZigbee: [fzLocal.device_specific],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET), e.battery()],
    meta: {
        configureKey: 1,
        battery: {dontDividePercentage: true}
    },
};

module.exports = definition;
`

bat86, 2023-11-22:
I'm experiencing an issue with the 'Invert Cover' setting for a Zigbee curtain controller (Zemismart model ZM85EL-2Z (Zigbee Model - TS0601. Zigbee Manufacturer- _TZE200_cf1sl3tj)). Despite setting invert_cover to true, the open and close positions of the cover are not being inverted as expected.

mrmaximas, 2025-01-25:
Hello!
I think I've found the problem.
1.  If you set the position of the curtains from the RF remote key, the subsequent report with DP 3 percent_state (30-50-70%) is ignored. 
2. If you fully open or close the curtains from RF remote key, the motor at the end of the operation reports 0 or 100%, but the OPEN/CLOSE state does not reported by motor itsef and does not change in z2m. 
it seems to me that the converter should take into these variants of motor operation. Unfortunately, I don't have the qualifications to modify.

mrmaximas, 2025-01-25:
I've compiled some information on this motor. It might be useful.
DP's

```
"mapping": {
            "1": {
                "code": "control",
                "type": "Enum",
                "values": {
                    "range": [
                        "open",
                        "stop",
                        "close",
                        "continue"
                    ]
                }
            },
            "2": {
                "code": "percent_control",
                "type": "Integer",
                "values": {
                    "unit": "%",
                    "min": 0,
                    "max": 100,
                    "scale": 0,
                    "step": 1
                }
            },
            "3": {
                "code": "percent_state",
                "type": "Integer",
                "values": {
                    "unit": "%",
                    "min": 0,
                    "max": 100,
                    "scale": 0,
                    "step": 1
                }
            },
            "4": {
                "code": "mode",
                "type": "Enum",
                "values": {
                    "range": [
                        "morning",
                        "night"
                    ]
                }
            },
            "5": {
                "code": "control_back_mode",
                "type": "Enum",
                "values": {
                    "range": [
                        "forward",
                        "back"
                    ]
                }
            },
            "7": {
                "code": "work_state",
                "type": "Enum",
                "values": {
                    "range": [
                        "opening",
                        "closing"
                    ]
                }
            },
            "11": {
                "code": "situation_set",
                "type": "Enum",
                "values": {
                    "range": [
                        "fully_open",
                        "fully_close"
                    ]
                }
            },
            "12": {
                "code": "fault",
                "type": "Bitmap",
                "values": {
                    "label": [
                        "motor_fault"
                    ]
                }
            },
            "13": {
                "code": "battery_percentage",
                "type": "Integer",
                "values": {
                    "unit": "%",
                    "min": 0,
                    "max": 100,
                    "scale": 0,
                    "step": 1
                }
            },
            "16": {
                "code": "border",
                "type": "Enum",
                "values": {
                    "range": [
                        "up",
                        "down",
                        "up_delete",
                        "down_delete",
                        "remove_top_bottom"
                    ]
                }
            },
            "19": {
                "code": "position_best",
                "type": "Integer",
                "values": {
                    "unit": "%",
                    "min": 0,
                    "max": 100,
                    "scale": 0,
                    "step": 1
                }
            },
            "20": {
                "code": "click_control",
                "type": "Enum",
                "values": {
                    "range": [
                        "up",
                        "down"
                    ]
                }
            }
        }

```
Chines origin DPs descriptions

```
abilityId	accessMode	code	extensions	name	typeSpec	description
1	rw	control	
(2 items)
控制	
(2 items)
【必选项】该DP点用于控制电机的开启、暂停、关闭。该DP点枚举值参数不允许修改，不允许增减。其中，continue功能在APP面板无呈现，用于第三方（Google home）开通使用，下发指令后，需要MCU记住上一次的状态，并执行动作，注意，无该功能，无法通过WWGA认证。
2	rw	percent_control	
(2 items)
开启百分比控制	
(6 items)
如果窗帘电机可以监控窗帘目前的所处位置，例如窗帘现在开了一半，也就是50%，那么可以通过该DP点进行窗帘位置的自定义控制。控制使用该DP点，如果选择了该DP点，必选选择“开启百分比状态显示”，该DP点数值范围值参数不允许修改，不允许增减。
3	ro	percent_state	
(2 items)
开启百分比状态	
(6 items)
该DP点与“开启百分比状态控制”配套使用。该DP点用于显示实际窗帘所处位置。该DP点数值范围值参数不允许修改，不允许增减。
4	rw	mode	
(2 items)
工作模式	
(2 items)
5	rw	control_back	
(2 items)
电机反向	
(2 items)
用于设置电机转向，顺时针，或者逆时针，forward代表顺时针，back代表逆时针
7	ro	work_state	
(2 items)
工作状态	
(2 items)
【必选项】用于显示电机的当前工作状态。该DP点枚举值参数不允许修改，不允许增减。如需要修改在APP面板中显示的内容，可以前往第四步的拓展中心，多语言管理修改代表值，新增的枚举值所代表的的内容也是在多语言中修改代表值。
11	ro	situation_set	
(2 items)
百分比含义	
(2 items)
上报该DP点枚举值，以告知APP，该窗帘，100%对应的是全开，还是全关。不上报或者不选择改DP点，则默认100%为全关。
12	ro	fault	
(2 items)
故障告警	
(3 items)
【必选】设备上报故障值，该DP点枚举值可增减，但不可修改枚举值名称，例如可以删除motor_fault，可以增加其他值，未选则不展示该功能。 如需要修改在APP面板中显示的内容，可以前往第四步的拓展中心，多语言管理修改代表值，新增的枚举值所代表的的内容也是在多语言中修改代表值。
13	ro	battery_percentage	
(2 items)
电池电量	
(6 items)
16	rw	border	
(2 items)
设置限位	
(2 items)
19	rw	position_best	
(2 items)
最佳位置	
(6 items)
20	rw	click_control	
(2 items)
点动控制	
(2 items)

```