Koenkk/zigbee2mqtt #11252: Tuya TRV TS0601 firmware update fails

guidokieboom, 2022-02-05:
### What happened?

Firmware updates for TS0601 thermostat valve fail to complete.

### What did you expect to happen?

OTA firmware update would complete successfully

### How to reproduce it (minimal and precise)

In Z2M OTA section at first check for new firmware, then click on 'update device firmware'
[image]

Tried again by force removing the TRV, then pairing again. Same result: firmware fails between 5-15% progress.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26580700

### Adapter

Conbee 2 USB stick with Z2M (no ZHA, no Deconz)

### Debug log

[image]
[image]
[image]

guidokieboom, 2022-02-05:
noticed following in the Home Assistant supervisor log:
[image]

Lyr3x, 2022-02-08:
I thought too its not working, but check the OTA troubleshooting:
https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#troubleshooting

I removed the battery of my valves for a few seconds and updated successfully afterwards.

guidokieboom, 2022-02-08:
Thanks Lyr3x for your suggestions. I tried this and unfortunately still no OTA.....Then I realized I recently changed my USB hub with an USB extension cable.... After re-using the USB hub (between Conbee2 and RPi4) my Z2M works like a charm again including successful OTA of the TRV.

guidokieboom, 2022-02-08:
use USB hub between Conbee2 stick and Raspberry PI.

closed 2022-02-08 by guidokieboom

KimHEF, 2022-10-20:
Is it possible to OTA-Update these TVRs (Tuya TS0601) with z2m and CC2531?