Koenkk/zigbee2mqtt #11255: LoraTap SS6400ZB-V2 Does not Pair

scolby33, 2022-02-06:
### What happened?

I received a LoraTap SS6400ZB-V2 four-button Zigbee remote today. I already have a LoraTap SS6400ZB remote, which paired correctly. However, they have produced a revision.

There is no longer a button inside the remote for pairing. Instead, the new procedure is to press and hold any one of the main buttons for 10s to enter pairing mode. From this point, the LED on the device flashes at a moderate rate for a minute, and then turns off.

I enabled pairing in Zigbee2MQTT, pressed and held one of the buttons on the remote, and waited for it to pair. Unfortunately, it did not. There was no log output at all. I tried entering the pairing mode with all four of the buttons on the device; the behavior was the same each time.

I upgraded my coordinator (Tube's CC2652P2 USB Coordinator) to the latest firmware from [here](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin), from 20211114 to 20211217. I followed the other steps from [the FAQ](https://www.zigbee2mqtt.io/guide/faq/#no-logging-is-shown-at-all):
- The coordinator is on a USB cable far from my RPi and other devices
- There are no other Zigbee networks
- I do not have a replacement battery to try, but it reads good on my voltmeter
- I was attempting to pair the device less than six inches from the controller

### What did you expect to happen?

I expected that the SS6400ZB-V2 would pair with Zigbee2MQTT.

### How to reproduce it (minimal and precise)

1. Enable pairing in Zigbee2MQTT
2. Press and hold any of the buttons on the SS6400ZB-V2 for 10s until the LED begins to flash
3. Wait one minute
4. The device will not have paired

### Zigbee2MQTT version

1.23.0 commit: 6ea61ba

### Adapter firmware version

20211217

### Adapter

Tube's CC2652P2 USB Coordinator

### Debug log

```
Zigbee2MQTT:info  2022-02-06 04:38:14: Succesfully changed options
Zigbee2MQTT:info  2022-02-06 04:38:14: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"sgk8i-1"}'
Zigbee2MQTT:debug 2022-02-06 04:38:18: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"sgk8i-2","value":true}'
Zigbee2MQTT:info  2022-02-06 04:38:18: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-02-06 04:38:18: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"sgk8i-2"}'
Zigbee2MQTT:debug 2022-02-06 04:41:36: Saving state to file /app/data/state.json
```

xxsokolov, 2022-03-05:
Hi, +1

skylineboy20000, 2022-03-19:
same issue here

alexh3o, 2022-04-03:
Hi, same with 1.25.0

alexh3o, 2022-04-05:
> Hi, same with 1.25.0

Well actually it works now. My problem was something else (my coordinator wouldn't pair any new devices until i changed usb port on the rpi4...)
So I have the SS6400ZB that enter pairing mode by pressing any buttons, and it was seen immediately in zigbee2mqtt recognized as 
Zigbee Model TS0044
Zigbee Manufacturer _TZ3000_ee8nrt2l

I have a Sonoff USB 3.0 Dongle Plus flashed with  Z-Stack_3.x.0 20220219 (CC1352P2_CC2652P_launchpad_coordinator_20220219.zipà

scolby33, 2022-04-11:
Unfortunately, no luck for me:
```
1.25.0 (commit #6f1460e)
zigbee-herdsman (0.14.20)
Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
```
I tried the pairing process again and had the same result as before: absolutely no response from Zigbee2MQTT that something was trying to pair at all.

scolby33, 2022-05-12:
Still an issue

scolby33, 2022-06-14:
I have once again updated my Zigbee2MQTT installation and adapter firmware, and this issue remains.

Minims, 2022-07-01:
Same issue with SonOff USB Dongle. Need to take my old CC2531 with same IEEE address to be able to pair.

nielswest65, 2022-07-10:
Hi, same problem, I tried making an external converter using the ts0044 base from tuya.js and for the -V2 and/or adding the -V2 to the whitelabel list in the tuya.js, and using this as an external converter voor tuya, nothing works.
Stating:  I am not a coder just a user +  and quite new to Zigbee, mqtt and Home Assistant. 
Loratap does state this -V2 remote only works with the Tuya hub? hoping just for marketing purposses
[SS6400ZB-V2.zip](https://github.com/Koenkk/zigbee2mqtt/files/9079032/SS6400ZB-V2.zip)

closed 2022-08-18 by github-actions[bot]