Koenkk/zigbee2mqtt #11256: Z2M in docker-compose doesent start

flayy73, 2022-02-06:
### What happened?

Hi,
I have a CONBEE2 adapter connected to my Ubuntu server.
I try to install Z2M into portainer by Docker Compose.
Here my configuration:

Stack:
```
version: 3.8
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
      - /home/zigbee2mqtt/data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8080:8080
    environment:
      - TZ=Europe/Rome
    devices:
      # Make sure this matched your adapter location
      - /dev/ttyACM0:/dev/ttyACM0
```

Here the configuration.yaml 
```
# Home Assistant integration (MQTT discovery)
homeassistant: true

# allow new devices to join
permit_join: false

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://192.168.2.34:1883'
  # MQTT server authentication, uncomment if required:
  #user: noneed
  #password: noneed

# Serial settings
serial:
  adapter: deconz
  port: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2439063-if00
```

Here the log 

```
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-06 11:36:37: Logging to console and directory: '/app/data/log/2022-02-06.11-36-36' filename: log.txt
Zigbee2MQTT:info  2022-02-06 11:36:37: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-06 11:36:37: Starting zigbee-herdsman (0.14.10)
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-02-06 11:36:37: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-02-06 11:36:37: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2022-02-06 11:36:37: Currently 0 devices are joined:
Zigbee2MQTT:warn  2022-02-06 11:36:37: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-02-06 11:36:37: Allowing new devices to join.
Zigbee2MQTT:warn  2022-02-06 11:36:37: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-02-06 11:36:37: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-02-06 11:36:38: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:error 2022-02-06 11:36:38: MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883
Zigbee2MQTT:error 2022-02-06 11:36:38: Exiting...
Zigbee2MQTT:info  2022-02-06 11:36:38: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-02-06 11:36:40: Stopped zigbee-herdsman
```

### What did you expect to happen?

I dont understand why the docker try to connect to mqtt://localhost instead that 192.168.2.34:1883 ? 
I also try with localhost without ay results.
The MQTT server is up and works (I can connect with MQTT Explorer).

any idea?

tks

### Zigbee2MQTT version

1.23.00

### Adapter firmware version

-

### Adapter

CONBEE 2

Underknowledge, 2022-02-06:
` MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883` 
Your MQTT broker refuses the connection. Try to use the container name instead of loopback and restart z2m

flayy73, 2022-02-06:
Tks UK, this works!!
```
network_mode: host
#    ports:
#      - 8888:8080 
```

now next question: How I can reach the frontend ?

Underknowledge, 2022-02-06:
see https://www.zigbee2mqtt.io/guide/configuration/frontend.html 
tldr: The default port is 8080, try to use the linux command `ss` to see opend ports for TCP, listen, not translating to services, procceses run `ss -tlnp`. This is a easy way to see whats going on on your machine. 
When youre unhappy with 8080 change it in your `configuration.yaml`
```yaml
frontend:
  port: 80
  ```
  For other general questions swing by the discord channel https://discord.gg/NyseBeK

m0wlheld, 2022-02-11:
This:

```
    devices:
      # Make sure this matched your adapter location
      - /dev/ttyACM0:/dev/ttyACM0
```

and this:

```
serial:
  adapter: deconz
  port: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2439063-if00
```

does not match.

closed 2022-03-21 by github-actions[bot]