Koenkk/zigbee2mqtt #11261: Saswell SEA802-Zigbee local temperature calibration doesn't allow floats?

smalec, 2022-02-06:
### What happened?

It seems that it's not possible to set float value for `local_temperature_calibration` parameter. I'm not sure if this is a problem of software since (if I understand it correctly) this line indicates, that it should be possible with `0.1` step:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/saswell.js#L49
I've tried to find device specification, but no luck so far :(

### What did you expect to happen?

Publishing a message to `zigbee2mqtt/FRIENDLY_NAME/set` topic with payload `{"local_temperature_calibration": 0.1}` should set this parameter to `0.1` (or any information that device doesn't support float values).

### How to reproduce it (minimal and precise)

1. Publish a message to `zigbee2mqtt/FRIENDLY_NAME/set` topic with payload `{"local_temperature_calibration": 0.1}`
2. Verify somehow that state of device isn't correct (in web gui or in console):
```
zigbee2mqtt      | Zigbee2MQTT:info  2022-02-06 16:05:56: MQTT publish: topic 'zigbee2mqtt/0x84fd27fffe335f8f', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":22,"frost_detection":"ON","linkquality":255,"local_temperature":21.6,"local_temperature_calibration":-102,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
```

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

 0x26580700

### Adapter

ConBee2/RaspBee2

brunoleon, 2022-02-06:
Hello,

are you sure 0.1 is effectively supported ? I also have SEA-802 and the step is 1 not 0.1
See https://github.com/Koenkk/zigbee-herdsman-converters/pull/3657/files

smalec, 2022-02-09:
No, I'm not sure. Probably you're right, and the step is 1. To avoid cnofusion I will create PR later.

gomble, 2022-03-04:
I have the same Problem. Therefor my automatisation in Homeassitant to sync the local calibration with a thermometer does not work. How to solve this ?

closed 2022-04-11 by github-actions[bot]