Koenkk/zigbee2mqtt #11267: Cannot started GUI

KuzinHouse, 2022-02-06:
### What happened?

Zigbee2MQTT:error 2022-02-06 21:47:51: Failed to start zigbee
Zigbee2MQTT:error 2022-02-06 21:47:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-02-06 21:47:51: Exiting...
Zigbee2MQTT:error 2022-02-06 21:47:51: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/src/controller/database.ts:19:35)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:116:34)
    at Zigbee.start (/app/lib/zigbee.ts:58:47)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-06 21:47:57: Logging to console and directory: '/app/data/log/2022-02-06.21-47-56' filename: log.txt
Zigbee2MQTT:info  2022-02-06 21:47:57: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-06 21:47:57: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:error 2022-02-06 21:47:57: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-02-06 21:47:57: Failed to start zigbee
Zigbee2MQTT:error 2022-02-06 21:47:57: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-02-06 21:47:57: Exiting...
Zigbee2MQTT:error 2022-02-06 21:47:57: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/src/controller/database.ts:19:35)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:116:34)
    at Zigbee.start (/app/lib/zigbee.ts:58:47)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-06 21:48:07: Logging to console and directory: '/app/data/log/2022-02-06.21-48-05' filename: log.txt
Zigbee2MQTT:info  2022-02-06 21:48:07: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-06 21:48:07: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:error 2022-02-06 21:48:07: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-02-06 21:48:07: Failed to start zigbee
Zigbee2MQTT:error 2022-02-06 21:48:07: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-02-06 21:48:07: Exiting...
Zigbee2MQTT:error 2022-02-06 21:48:07: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/src/controller/database.ts:19:35)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:116:34)
    at Zigbee.start (/app/lib/zigbee.ts:58:47)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-06 21:48:22: Logging to console and directory: '/app/data/log/2022-02-06.21-48-21' filename: log.txt
Zigbee2MQTT:info  2022-02-06 21:48:22: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-06 21:48:22: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:error 2022-02-06 21:48:22: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-02-06 21:48:22: Failed to start zigbee
Zigbee2MQTT:error 2022-02-06 21:48:22: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-02-06 21:48:22: Exiting...
Zigbee2MQTT:error 2022-02-06 21:48:22: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/src/controller/database.ts:19:35)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:116:34)
    at Zigbee.start (/app/lib/zigbee.ts:58:47)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-06 21:48:51: Logging to console and directory: '/app/data/log/2022-02-06.21-48-50' filename: log.txt
Zigbee2MQTT:info  2022-02-06 21:48:51: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-06 21:48:51: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:error 2022-02-06 21:48:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-02-06 21:48:51: Failed to start zigbee
Zigbee2MQTT:error 2022-02-06 21:48:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-02-06 21:48:51: Exiting...
Zigbee2MQTT:error 2022-02-06 21:48:51: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/src/controller/database.ts:19:35)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:116:34)
    at Zigbee.start (/app/lib/zigbee.ts:58:47)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-06 21:49:45: Logging to console and directory: '/app/data/log/2022-02-06.21-49-44' filename: log.txt
Zigbee2MQTT:info  2022-02-06 21:49:45: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-06 21:49:45: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:error 2022-02-06 21:49:45: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-02-06 21:49:45: Failed to start zigbee
Zigbee2MQTT:error 2022-02-06 21:49:45: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-02-06 21:49:45: Exiting...
Zigbee2MQTT:error 2022-02-06 21:49:45: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/src/controller/database.ts:19:35)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:116:34)
    at Zigbee.start (/app/lib/zigbee.ts:58:47)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-02-06 21:50:48: Logging to console and directory: '/app/data/log/2022-02-06.21-50-47' filename: log.txt
Zigbee2MQTT:info  2022-02-06 21:50:48: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-06 21:50:48: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:error 2022-02-06 21:50:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-02-06 21:50:48: Failed to start zigbee
Zigbee2MQTT:error 2022-02-06 21:50:48: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-02-06 21:50:48: Exiting...
Zigbee2MQTT:error 2022-02-06 21:50:48: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/src/controller/database.ts:19:35)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:116:34)
    at Zigbee.start (/app/lib/zigbee.ts:58:47)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)

### How to reproduce it (minimal and precise)

Cannot started GUI

### Zigbee2MQTT version

last

### Adapter firmware version

00000000

### Adapter

CC2652P1

### Debug log

[_z2m_logs (4).txt](https://github.com/Koenkk/zigbee2mqtt/files/8011311/_z2m_logs.4.txt)

pminarik72, 2022-02-07:
I had the same error. Check your database.db. In my case  }  was missing at the end of the last line.

Koenkk, 2022-02-07:
Potentially a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/11106, please check if it has already been fixed in the dev branch. 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-18 by github-actions[bot]